espressif/esp-idf #6810: Fail ESP32S2 FTM test with Android Smartphone.

TeiShim, 2021-04-01:
When I tested FTM on ESP32-S2(AP) with Android Smartphone(STA), I could see the FTM response message, but can NOT catch the results of RTT from ESP32-S2 AP as following link:
https://esp32.com/viewtopic.php?f=13&t=19919&sid=54c30292f4b42a7f805f9db8351182f7

Do you have a plan that updating successfully the FTM test with Android ? 
ref) https://source.android.com/devices/tech/connect/wifi-rtt

BRs,
Taehyoung Shim

zhangyang428, 2021-04-16:
@TeiShim  Hi, can you use the sniffer probe to capture FTM? Can you tell me the format of the captured FTM packet?
Thank you!

TeiShim, 2021-04-26:
@zhangyang428 Hi, Zhang Yang. Yes, we can capture FTM packets with Wireshark as you can see: https://www.wireshark.org/
And, you can also see the format of FTMs in the standard of IEEE 802.11-2016. 
If you need further information. I can send you the wireshark files.

zhangyang428, 2021-04-26:
@TeiShim  Thanks! I need further information for FTM frame. Can you share how you capture these data frames? My email address: 
dlutzhangyang@gmail.com

zhangyang428, 2021-05-09:
@TeiShim Thanks for your help! I also tried to capture frames of FTM, but not as informative as yours. For example, in the data frames that I've captured, there aren't Radiotap Header v0 or 802.11 radio information. How do you capture this information? Do you turn on some options before capturing a data frame?

TeiShim, 2021-06-23:
@zhangyang428 Sorry for late responding. This webpage will be help you. 
https://www.wireshark.org/docs/dfref/w/wlan.html

nachiketkukade, 2021-06-28:
> FTM on ESP32-S2(AP) with Android Smartphone(STA)

@TeiShim , as discussed on the ESP32 Forum thread, we're planning to include support for this configuration in IDF v4.3.1. I'll provide an update when this version is released.

TeiShim, 2021-06-30:
@nachiketkukade 
Thank you for notification. Could you let me know when IDF v4.3.1 will be released?
Depending on the release date, we will make a test schedule.

TeiShim, 2021-09-12:
@nachiketkukade 
Is FTM test connecting Android phone available in IDF v4.3.1? I tested, but fail again.

crackwitz, 2021-12-08:
I just tried this again with esp-idf master from today, an esp32-c2, and a Google Pixel 4a running Google's `WifiRttScan`.

Something's still not quite right, but I can't tell right now if it's the google app's fault, or ESP, or both.

Behavior: phone/app gets no measurements while disconnected, but does get measurements as soon as it's connected to *any* AP.

The phone doesn't have to connect to *that* ESP32-C2 AP. FTM sessions to the FTM-capable softAP succeed even if the phone's wifi connects to another non-FTM-capable regular ESP32 that runs a softAP.

I currently don't have FTM-capable APs at hand. I think the app managed to FTM to one of those without needing to be connected to any access point, but I might be mistaken.

nachiketkukade, 2021-12-10:
@crackwitz , I guess you mean ESP32-S2. I re-tested with the Pixel with no connection, and yes, there seems to be some issue there. Protocol wise there seems to be nothing different going on with or without the connection, but the issue is not there with the Velop AP. I also tested with Mi 11X Pro, it doesn't have this issue, could be some quirky behaviour due to different implementations. Regardless, I'll investigate further and update.

nachiketkukade, 2021-12-10:
> Is FTM test connecting Android phone available in IDF v4.3.1? I tested, but fail again.

@TeiShim , yes, the support is there in v4.3.1. Can you pls confirm which Smartphone you tested with?
Also I must warn that the accuracy will be very low.

crackwitz, 2021-12-10:
Just thought I should mention the phenomenon. I _don't_ actually need Android and ESP32 to cooperate. That's just a convenient way to check a few things.

nachiketkukade, 2024-03-04:
An update to anyone interested - we've done more tests with Pixel devices and fixed some issues. Changes are expected to merge in coming weeks.

TeiShim, 2024-04-05:
Thank you @nachiketkukade, Is this FTM function able to use in 5GHz with ESP32-C5?

nachiketkukade, 2024-04-12:
@TeiShim , support for FTM for C5 in 5GHz is planned

crackwitz, 2024-04-12:
> @TeiShim , support for FTM in 5GHz for C5 is planned

Off topic but perhaps related: are there plans for 802.11ba Wake-Up Radio ("Wake on WLAN") in the C5 or P4 or other series?