espressif/esp-idf #6811: Duplicate http response header causes the style not to be loaded

ahmedwahdan, 2021-04-01:
## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version  v4.2
  
- Build System:         idf.py

- Operating System:     Windows
- (Windows only) environment type: Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

//Detailed problem description goes here.
I'm updating from the old http-server implantation to ESP-IDF but the browser can't load style file.
I suspect that because the response header is set twice, the browser can't load the style (I can be mistaken, not sure)
[image]

I set the header type before sending the file 

```
static esp_err_t http_send_file(httpd_req_t *req,
                           const char *file_datap,
                           size_t file_data_length,
                           const char *content_typep)
//===========================================================================
{
  if (!(strcmp(content_typep, HTTP_CONTENT_TYPE_APP_JSON) == 0 ||
        strcmp(content_typep, HTTP_CONTENT_TYPE_TEXT_PLAIN) == 0))
  {
    esp_err_t err = httpd_resp_set_hdr(req, "Content-Encoding", "gzip");
    err |= httpd_resp_set_hdr(req, "Content-Type", content_typep);
    if (err != ESP_OK)
    {
      ESP_LOGI(TAG, "HTTP set header error: %d", err);
      return ESP_FAIL;
    }
  }

  // size_t response_size = file_data_length;

  // http_buffer_t http_buffer;
  // http_buffer.data_is_persistent = true;
  // http_buffer.size = response_size;
  // http_buffer.data = file_datap;

  if (httpd_resp_send(req, file_datap, file_data_length) != ESP_OK) 
  {
    ESP_LOGE(TAG, "File sending failed!");
   // /* Abort sending file */
    //httpd_resp_sendstr_chunk(req, NULL);
    /* Respond with 500 Internal Server Error */
    httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Failed to send file");
    return ESP_FAIL;
  }

  /* Respond with an empty chunk to signal HTTP response completion */
 // httpd_resp_send_chunk(req, NULL, 0);
  return ESP_OK;
}
```
regards

negativekelvin, 2021-04-01:
Use httpd_resp_set_type

ahmedwahdan, 2021-04-01:
Thanks, this fixed the issue.

closed 2021-04-01 by ahmedwahdan