espressif/esp-idf #6812: CMake can't complete configuration on project that configured fine previously

MartyMcFlyInTheSky, 2021-04-01:
## Environment
- IDF version: v4.4-dev-473-gc49d03b8e
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: Yes, VS Code
- Hardware: AMD Ryzen 7 5800H, NVIDIA Geforce RTX 3070

## Problem Description

When I run build from the VSCode add-on button OR type it manually into the command prompt (idf.py build), CMake fails to build my project with the following error:

```
-- Configuring incomplete, errors occurred!
See also "C:/dev/aurora-esp32-firmware/build/CMakeFiles/CMakeOutput.log".
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja .." terminated with exit code: 1.
```

This exact project built fine on a different laptop. Now I have a new one and installed the esp-idf Framework via VSCode add-on. I have never seen this error before, so what could cause it?

Note: CMakeOutput.log has over 2000 lines and is very non descriptive, so I put it in the attachment:

[CMakeOutput.log](https://github.com/espressif/esp-idf/files/6244797/CMakeOutput.log)

brianignacio5, 2021-04-02:
Have you tried deleting the `build` folder before building again ? Maybe there are some files in cache.

georgik, 2021-04-06:
@MartyMcFlyInTheSky That's interesting error. 
Please, could you try to run following build commands in the command prompt? Suggested command should give us an answer whether the problem is not caused by CCache and parallel build on Windows: 
```
idf.py fullclean
set IDF_CCACHE_ENABLE=
set CMAKE_BUILD_PARALLEL_LEVEL=1
idf.py build
```

MartyMcFlyInTheSky, 2021-04-12:
Sorry was afk for holidays. Thanks for the answers!

> Have you tried deleting the `build` folder before building again ? Maybe there are some files in cache.

I have deleted it and rebuilt but with the same result

> @MartyMcFlyInTheSky That's interesting error.
> Please, could you try to run following build commands in the command prompt? Suggested command should give us an answer whether the problem is not [...]

Still yields the same error:

[Terminal out.txt](https://github.com/espressif/esp-idf/files/6297539/Terminal.out.txt)

georgik, 2021-04-13:
@MartyMcFlyInTheSky Thank you for additional logs. There seem to be only two differences:
Old OS: 10.0.19042
New OS: 10.0.19041
That does not seem to be significant, it might be worth checking for updates and restart the computer.

The more significant difference is the version of Ninja
Old: 1.10.0
New: 1.10.2

The update 1.10.2 contains changes in Windows build logic: https://groups.google.com/g/ninja-build/c/oobwq_F0PpA/m/FeJC5LoRBgAJ

I would suggest the following test:
- take file `tools/tools.json` from working ESP-IDF installation and replace the file in not-working installation (make a backup before overwriting) 
- run `install.bat` which will download 1.10.0
- start new ESP-IDF shell
- check `ninja --version`, you should get 1.10.0
- rebuild the project

MartyMcFlyInTheSky, 2021-04-13:
I'm stupid I'm sorry. I downgraded ninja but I think the error was related to CMake not being able to resolve the "WiFi" component. So why did it work on the old system? I remember to have once changed the folder names on the old system from lower to uppercase. Of course I had to make the adaptions in the CMakeLists.txt to reflect the changes. However, it seems that git doesn't pass on such changes and when I cloned the repo on the new computer it just used the old lowercase names which led to CMake not being able to find certain components.

I'm following this topic now to make git case sensitive:
https://stackoverflow.com/questions/37280945/why-does-the-name-of-this-folder-in-git-not-match-up-with-how-it-exists-on-the-f

georgik, 2021-04-13:
Thank you @MartyMcFlyInTheSky for posting the solution to the problem.
CMake is case sensitive even on Windows where some directories might be created under WSL:
https://devblogs.microsoft.com/commandline/improved-per-directory-case-sensitivity-support-in-wsl

closed 2021-06-02 by Alvin1Zhang