espressif/esp-idf #6813: HTTPS Memory Leak

AndrewBHarmon, 2021-04-01:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32D
- IDF version: v4.4-dev-479-g1067b2870
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

HTTPS Server does not allow for more than 4 or 5 connections when acting as a server without having to reboot, even if other connections get closed. Monitoring heap usage shows that memory gets used for new connections but never released.

### Expected Behavior

Memory should be released back to the heap when client disconnects from Server to allow for new connections.

### Actual Behavior

Available heap memory decrements for each client session that opens, but does not increment when connection is closed. This results in no new clients being able to connect and other tasks seizing due to unavailability of memory.

### Steps to reproduce

1. Use HTTPS Server Simple example
2. Add print statement to show available heap memory 
2. Connect to HTTPS Server from multiple devices
3. Close all sessions from client side

### Code to reproduce this issue

<a href="https://github.com/espressif/esp-idf/blob/c49d03b8eb9cb2070960927cbd7194645c44a874/examples/protocols/https_server/simple/main/main.c"> HTTPS Simple Server example </a> with 
```
while(1){  
    vTaskDelay(2000 / portTICK_PERIOD_MS);  
    ESP_LOGW(TAG, "Mem free: %u", heap_caps_get_free_size(MALLOC_CAP_8BIT));  
} 
``` 
added to the very end of ```app_main()``` to show available memory

## Debug Logs

```
I (689) example_connect: Connecting to MYSSID...
I (689) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (789) wifi:mode : sta (ac:67:b2:47:70:a4)
I (789) wifi:enable tsf
I (799) example_connect: Waiting for IP(s)
I (1409) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2139) wifi:state: init -> auth (b0)
I (2149) wifi:state: auth -> assoc (0)
I (2159) wifi:state: assoc -> run (10)
I (2199) wifi:connected with MYSSID, aid = 1, channel 6, BW20, bssid = d4:6a:91:ca:e2:e0
I (2199) wifi:security: WPA2-PSK, phy: bgn, rssi: -65
I (2199) wifi:pm start, type: 1

I (2209) wifi:AP's beacon interval = 102400 us, DTIM period = 2
W (2219) wifi:<ba-add>idx:0 (ifx:0, d4:6a:91:ca:e2:e0), tid:0, ssn:0, winSize:64
W (3239) wifi:<ba-add>idx:1 (ifx:0, d4:6a:91:ca:e2:e0), tid:6, ssn:3, winSize:64
I (4089) example: Starting server
I (4089) esp_https_server: Starting server
I (4089) esp_https_server: Server listening on port 443
I (4089) example: Registering URI handlers
I (4099) esp_netif_handlers: example_connect: sta ip: 192.168.3.158, mask: 255.255.255.0, gw: 192.168.3.1
I (4109) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.3.158
I (4119) example_connect: Connected to example_connect: sta
I (4119) example_connect: - IPv4 address: 192.168.3.158
Mem free 213096
Mem free 213104
[... 236 lines cut ...]
W (195779) httpd: httpd_accept_conn: session creation failed
W (195779) httpd: httpd_server: error accepting new connection
I (195789) esp_https_server: performing session handshake
E (196109) esp-tls-mbedtls: mbedtls_ssl_handshake returned -19840
E (196109) esp_https_server: esp_tls_create_server_session failed
W (196119) httpd: httpd_accept_conn: session creation failed
W (196119) httpd: httpd_server: error accepting new connection
I (196569) esp_https_server: performing session handshake
Mem free 7564
Mem free 9124
E (200699) esp-tls-mbedtls: mbedtls_ssl_handshake returned -16
E (200699) esp_https_server: esp_tls_create_server_session failed
W (200699) httpd: httpd_accept_conn: session creation failed
W (200699) httpd: httpd_server: error accepting new connection
Mem free 37856
Mem free 37852
Mem free 37852
Mem free 37852
Mem free 37852
Mem free 37852
Mem free 37852
Mem free 37852
Mem free 38060
Mem free 38060
```

Alvin1Zhang, 2021-04-02:
Thanks for reporting and sharing the details, we will look into.

AdityaHPatwardhan, 2021-04-06:
Hi @AndrewBHarmon thank you for notifying about the leak !,
Can you  please check this patch, I think it should fix the memory leak !.
[https_server_leak.patch.zip](https://github.com/espressif/esp-idf/files/6264959/https_server_leak.patch.zip)

AndrewBHarmon, 2021-04-06:
I ran the same test again and this patch seems to have resolved the issue. Thank you!

AdityaHPatwardhan, 2021-04-06:
thanks for checking it out, I have raised appropriate MR in our internal code-base, it should be available on GitHub in a few days.

mahavirj, 2021-04-06:
Keeping issue open, it will be (auto) closed once fix reaches to github repository.

closed 2021-04-15 by espressif-bot