espressif/esp-idf #6816: Read NVS Flash not working in Timer ISR. Please suggest alternative way.

Shravani100, 2021-04-02:
Dear all,

Development Kit: ESP32S2 saola-1 Develpement Board

Kit version :v1.2
Module or chip used: ESP32-S2-WROOM
IDF version :4.4
Build System: idf.py
Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: Ubuntu 20.04 on vbox
Using an IDE?: Yes
Power Supply: USB

What I want to do exactly:

I am trying to read NVS flash using the example given in esp-idf flash and load the stored value in the NVS flash to update the counter of timer in the ISR of the timer. But when the interrupt occurs the ESP32 restarts. This is because NVS flash read is a blocking function and it is not recommended to use blocking function in ISR.  So I tried to create a queue and receive the value from queue in ISR.  Is there any other way to read the NVS flash and return the NVS flash values to the timer ISR.

0xjakob, 2021-04-06:
Hi @Shravani100!

One question, why do you need to update the timer counter? If you provide more information about your use-case, we might find a different solution for you.

As you already notice, you can't use NVS inside ISR routines. There are actually two reasons:
1. Blocking procedures must be avoided in ISRs as they would block the CPU core and no other ISRs could be served for that time, very basically.
2. Currently, we can only use code which is loaded into DRAM inside ISRs. The code for the NVS flash resides in the flash of the ESP. If an ISR tries to access that code in flash anyway, the CPU will issue an exception, as is briefly in the [Fatal Errors documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/fatal-errors.html?highlight=illegal%20instruction%20exception#cache-disabled-but-cached-memory-region-accessed).

Furthermore, because flash cannot be accessed during ISRs, you can't read the NVS data from flash, either.
The solution would be to start a FreeRTOS task (`vTaskCreate()`) which waits using a queue. The ISR will send a signal via the queue to start the NVS read. But the ISR will return *before* the task even picks up the work, which doesn't seem to be your intention. This leads us back to the question in the beginning, please tell us more what you want to do. If you don't want to disclose details here, please let us know, too. There are ways to send us information privately.

Shravani100, 2021-04-06:
I am developing a alarm clock with multiple alarms. Each alarm triggers the ISR and when the ISR is triggered I want to load next alarm value into the timer counter by reading the nvs flash. I tried to make a task to read_nvs flash and send the value to ISR via RTOS xQueueRecieveInISR() function, but as you said the ISR returns even before the task read_nvs flash starts executing.

Shravani100, 2021-04-06:
Can you please give me personal contact so that I can give more information

0xjakob, 2021-04-06:
@Shravani100 Are you using the `esp_timer` API or the timer group API? The former already provides that functionality. You can have multiple timers registered in there. Internally, it is doing the same thing as you are trying to do (once an alarm as been triggered, it is registering the next alarm, very basically). I would suggest to try the `esp_timer` API then. Another advantage is that the the `esp_timer` API already uses a task, so you can actually run code which is located in flash. You still should offload long-running or blocking tasks to another task, though.

Shravani100, 2021-04-06:
@0xjakob I am using the timer group API. Thanks for the suggestion.  I will try esp_timer API and revert you back.

PrameetP, 2021-04-06:
Hello,  I managed to initialize the timer using the timers using esp_timer API. I have an array of events of alarm time in seconds. 
Following is the code:

`

#include <stdio.h>
#include <string.h>
#include <unistd.h>
#include "esp_timer.h"
#include "esp_log.h"
#include "esp_sleep.h"
#include "sdkconfig.h"
#include "esp_types.h"
#include "stdint.h"

static void periodic_timer_callback(void* arg);
static void oneshot_timer_callback(void);
static void start_scheduling(void);

static const char* TAG = "example";

void app_main(void)
{
  start_scheduling();
}

static void periodic_timer_callback(void* arg)
{
    int64_t time_since_boot = esp_timer_get_time();
    ESP_LOGI(TAG, "Periodic timer called, time since boot: %lld us", time_since_boot);
}

static void oneshot_timer_callback(void)
{
    int64_t time_since_boot = esp_timer_get_time();
    ESP_LOGI(TAG, "One-shot timer called, time since boot: %lld us", time_since_boot);
    //esp_timer_handle_t periodic_timer_handle = (esp_timer_handle_t) arg;
    /* To start the timer which is running, need to stop it first */
    //ESP_ERROR_CHECK(esp_timer_stop(periodic_timer_handle));
   // ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer_handle, 1000000));
    time_since_boot = esp_timer_get_time();
   ESP_LOGI(TAG, "Restarted periodic timer with 1s period, time since boot: %lld us",
            time_since_boot);
}

static void start_scheduling(void)
{
    /* simulate read nvs flash */
    //read nvs flash

    int events[8] = {10,30,1,0,50,55,2,1};  //Our array of event time Format is as follows:{ontime, offtime, eventid, eventstatus}
    int i = 0;
    int ontime[2];
    int offtime[2];
    int eventstatus[2];
    int eventid[2];
    int j=0;
    /*Pick the on-time from the array*/
    while(i<8)
    {   
        ontime[j] = events[i];
        j+=1;
        i+=4;
    }
    i=1;
    j=0;
     /*Pick the off-time from the array*/
    while(i<8)
    {   
        offtime[j]=events[i];
         j+=1;
         i+=4;
    }
    i=2;
    j=0;
    /*Pick the event id from the array*/
    while(i<8)
    {   
        eventid[j]=events[i];
        j+=1;
         i+=4;
    }
    /*Pick the event status from the array*/
    i=3;
    j=0;
    while(i<8)
    {   
        eventstatus[j]=events[i];
         j+=1;
         i+=4;
    }  
    
   //  const esp_timer_create_args_t periodic_timer_args = {
     //       .callback = &periodic_timer_callback,
    //        /* name is optional, but may help identify the timer when debugging */
   //         .name = "periodic"
   // }; 

    //esp_timer_handle_t periodic_timer;
    //ESP_ERROR_CHECK(esp_timer_create(&periodic_timer_args, &periodic_timer));
    /* The timer has been created but is not running yet */

    const esp_timer_create_args_t oneshot_timer_args = {
            .callback = &oneshot_timer_callback,
            /* argument specified here will be passed to timer callback function */
            //.arg = (void*) periodic_timer,
            .name = "one-shot"
    };

    /*For each on-time create a timer*/

     for(i =0;i<2;i++)
   {  
        esp_timer_handle_t oneshot_timer[2];
    ESP_ERROR_CHECK(esp_timer_create(&oneshot_timer_args, &oneshot_timer[i]));
    ESP_ERROR_CHECK(esp_timer_start_once(oneshot_timer[i],(uint64_t)(ontime[i]*1000000)));
   

        printf("\nOntime: %d", ontime[i]);
        printf("\nOfftime: %d", offtime[i]);
        printf("\nEvent ID: %d", eventid[i]);
        printf("\nEvent Status: %d", eventstatus[i]); 
    }
    

    /* Start the timers */
    //ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, 500000));
    
   
   

    ESP_LOGI(TAG, "Started timers, time since boot: %lld us", esp_timer_get_time());

    /* Print debugging information about timers to console every 2 seconds */
    for (int i = 0; i < 5; ++i) {
        ESP_ERROR_CHECK(esp_timer_dump(stdout));
        usleep(2000000);
    }

    /* Timekeeping continues in light sleep, and timers are scheduled
     * correctly after light sleep.
     */
    ESP_LOGI(TAG, "Entering light sleep for 0.5s, time since boot: %lld us",
            esp_timer_get_time());

    ESP_ERROR_CHECK(esp_sleep_enable_timer_wakeup(500000));
    esp_light_sleep_start();

    ESP_LOGI(TAG, "Woke up from light sleep, time since boot: %lld us",
                esp_timer_get_time());

    /* Let the timer run for a little bit more */
    usleep(50000000);

    /* Clean up and finish the example */
    //ESP_ERROR_CHECK(esp_timer_stop(periodic_timer));
    //ESP_ERROR_CHECK(esp_timer_delete(periodic_timer));
     //for(i=0;i<2;i++)
    //{
    //ESP_ERROR_CHECK(esp_timer_delete(&oneshot_timer[0]));
   // ESP_ERROR_CHECK(esp_timer_delete(&oneshot_timer[1]));
    //} 
    ESP_LOGI(TAG, "Stopped and deleted timers");
   

    
}
`
It works. But I am not sure how to delete the timer events

PrameetP, 2021-04-06:
@0xjakob Hello, I managed to initialize the timer using the timers using esp_timer API. I have an array of events of alarm time in seconds.
Following is the code:

`

#include <stdio.h>
#include <string.h>
#include <unistd.h>
#include "esp_timer.h"
#include "esp_log.h"
#include "esp_sleep.h"
#include "sdkconfig.h"
#include "esp_types.h"
#include "stdint.h"

static void periodic_timer_callback(void* arg);
static void oneshot_timer_callback(void);
static void start_scheduling(void);

static const char* TAG = "example";

void app_main(void)
{
start_scheduling();
}

static void periodic_timer_callback(void* arg)
{
int64_t time_since_boot = esp_timer_get_time();
ESP_LOGI(TAG, "Periodic timer called, time since boot: %lld us", time_since_boot);
}

static void oneshot_timer_callback(void)
{
int64_t time_since_boot = esp_timer_get_time();
ESP_LOGI(TAG, "One-shot timer called, time since boot: %lld us", time_since_boot);
//esp_timer_handle_t periodic_timer_handle = (esp_timer_handle_t) arg;
/* To start the timer which is running, need to stop it first */
//ESP_ERROR_CHECK(esp_timer_stop(periodic_timer_handle));
// ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer_handle, 1000000));
time_since_boot = esp_timer_get_time();
ESP_LOGI(TAG, "Restarted periodic timer with 1s period, time since boot: %lld us",
time_since_boot);
}

static void start_scheduling(void)
{
/* simulate read nvs flash */
//read nvs flash

int events[8] = {10,30,1,0,50,55,2,1};  //Our array of event time Format is as follows:{ontime, offtime, eventid, eventstatus}
int i = 0;
int ontime[2];
int offtime[2];
int eventstatus[2];
int eventid[2];
int j=0;
/*Pick the on-time from the array*/
while(i<8)
{   
    ontime[j] = events[i];
    j+=1;
    i+=4;
}
i=1;
j=0;
 /*Pick the off-time from the array*/
while(i<8)
{   
    offtime[j]=events[i];
     j+=1;
     i+=4;
}
i=2;
j=0;
/*Pick the event id from the array*/
while(i<8)
{   
    eventid[j]=events[i];
    j+=1;
     i+=4;
}
/*Pick the event status from the array*/
i=3;
j=0;
while(i<8)
{   
    eventstatus[j]=events[i];
     j+=1;
     i+=4;
}  

// const esp_timer_create_args_t periodic_timer_args = {
// .callback = &periodic_timer_callback,
// /* name is optional, but may help identify the timer when debugging */
// .name = "periodic"
// };

//esp_timer_handle_t periodic_timer;
//ESP_ERROR_CHECK(esp_timer_create(&periodic_timer_args, &periodic_timer));
/* The timer has been created but is not running yet */

const esp_timer_create_args_t oneshot_timer_args = {
        .callback = &oneshot_timer_callback,
        /* argument specified here will be passed to timer callback function */
        //.arg = (void*) periodic_timer,
        .name = "one-shot"
};

/*For each on-time create a timer*/

 for(i =0;i<2;i++)

{
esp_timer_handle_t oneshot_timer[2];
ESP_ERROR_CHECK(esp_timer_create(&oneshot_timer_args, &oneshot_timer[i]));
ESP_ERROR_CHECK(esp_timer_start_once(oneshot_timer[i],(uint64_t)(ontime[i]*1000000)));

    printf("\nOntime: %d", ontime[i]);
    printf("\nOfftime: %d", offtime[i]);
    printf("\nEvent ID: %d", eventid[i]);
    printf("\nEvent Status: %d", eventstatus[i]); 
}

/* Start the timers */
//ESP_ERROR_CHECK(esp_timer_start_periodic(periodic_timer, 500000));

ESP_LOGI(TAG, "Started timers, time since boot: %lld us", esp_timer_get_time());

/* Print debugging information about timers to console every 2 seconds */
for (int i = 0; i < 5; ++i) {
    ESP_ERROR_CHECK(esp_timer_dump(stdout));
    usleep(2000000);
}

/* Timekeeping continues in light sleep, and timers are scheduled
 * correctly after light sleep.
 */
ESP_LOGI(TAG, "Entering light sleep for 0.5s, time since boot: %lld us",
        esp_timer_get_time());

ESP_ERROR_CHECK(esp_sleep_enable_timer_wakeup(500000));
esp_light_sleep_start();

ESP_LOGI(TAG, "Woke up from light sleep, time since boot: %lld us",
            esp_timer_get_time());

/* Let the timer run for a little bit more */
usleep(50000000);

/* Clean up and finish the example */
//ESP_ERROR_CHECK(esp_timer_stop(periodic_timer));
//ESP_ERROR_CHECK(esp_timer_delete(periodic_timer));
 //for(i=0;i<2;i++)
//{
//ESP_ERROR_CHECK(esp_timer_delete(&oneshot_timer[0]));

// ESP_ERROR_CHECK(esp_timer_delete(&oneshot_timer[1]));
//}
ESP_LOGI(TAG, "Stopped and deleted timers");

}
`
It works. But I am not sure how to delete the timer events

0xjakob, 2021-04-07:
@PrameetP 
```c
esp_timer_stop()
esp_timer_delete()
```
It's the code you commented.

PrameetP, 2021-04-07:
When there is single timer instance:

For example: `esp_timer_handle_t oneshot_timer` I can delete it by passing `oneshot_timer` structure to `esp_delete_timer() `function.

When there are multiple timers instances:
For example: 

To create the timers: 
```
int ontime[2] = {20,40} //Array of alarm times to trigger the timers 

for(i =0;i<2;i++)
   {  
        esp_timer_handle_t oneshot_timer[i]; //create a timer for each value of alarm from ontime array
    ESP_ERROR_CHECK(esp_timer_create(&oneshot_timer_args, &oneshot_timer[i]));
    ESP_ERROR_CHECK(esp_timer_start_once(oneshot_timer[i],(uint64_t)(ontime[i]*1000000)));
   
}
```
**The above code works**

To delete the timers:

for(i=0;i<2;i++)
    {
    ESP_ERROR_CHECK(esp_timer_delete(oneshot_timer[i]));
   
    } 
This gives error. If I comment the above code, the error goes. I am attaching the Error log.

One more question is there any limitation on number of maximum instance of timers I can create.
[Error_log.txt](https://github.com/espressif/esp-idf/files/6268925/Error_log.txt)

0xjakob, 2021-04-12:
@PrameetP Are in some way affiliated with @Shravani100? If not, could you please open a post on esp32.com? This is our forum for general programming questions related to ESP-IDF. This particular issue is about @Shravani100's problem with the timers. It is quite confusing for the original poster to see different questions discussed in their post. Thanks!

PrameetP, 2021-04-12:
Yes, Sharavani is my colleague.

0xjakob, 2021-04-13:
@PrameetP OK, I didn't know that :)

If you look at the build log (Error_log.txt), the compiler tells you quite exactly what's wrong. First of all, the callback type of `oneshot_timer_callback` doesn't have the correct type (the first warning). Then, `oneshot_timer` isn't declared (you commented it in the code above). 

Please keep in mind: these are general programming questions and are usually solved in our forum (esp32.com) or any other programming forum (e.g. stackoverflow.com). Github issues in the IDF project are for solving particular problems with IDF (like the original question of this issue), which are not general programming questions.

0xjakob, 2021-06-28:
@PrameetP @Shravani100 I closed this issue due to inactivity. If you have further questions regarding the original issue, feel free to re-open.

closed 2021-06-30 by Alvin1Zhang