espressif/esp-idf #6817: Failed to Disabling Flash Encryption

noyelseth, 2021-04-02:
## Environment

- Module or chip used: ESP32-S2
- IDF version (run ``git describe --tags`` to find it): v4.2
- Build System:        idf.py
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

Failed to Disabling Flash Encryption using espefuse.py 

Link 1: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#disabling-flash-encryption
Link 2: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/flash-encryption.html#disabling-flash-encryption

### Actual Behavior
[image]

[image]

Last Flashed Efuse Summary:
[efuse_summary.txt](https://github.com/espressif/esp-idf/files/6249860/efuse_summary.txt)

negativekelvin, 2021-04-02:
https://github.com/espressif/esptool/blob/b8293b240ce937ee1f30dd8d6dc1422679ff3214/espressif/efuse/esp32s2/mem_definition.py#L181

This is because class is not set to bitcount

noyelseth, 2021-04-05:
To, Disable Flash Encryption in esp32-S2 
I used to reset the efuse bit **`EFUSE_SPI_BOOT_CRYPT_CNT`**.  by uing this command 

`espefuse.py burn_efuse EFUSE_SPI_BOOT_CRYPT_CNT` 

[LINK](https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32s2/security/flash-encryption.html#disabling-flash-encryption)

Its shows **`fatal error occurred: Invalid the efuse name 'EFUSE_SPI_BOOT_CRYPT_CNT'`**

negativekelvin, 2021-04-05:
You can use `espefuse.py burn_efuse SPI_BOOT_CRYPT_CNT 3` until it is fixed

noyelseth, 2021-04-06:
Hello, @negativekelvin 

I used yours mention command, but its still show this.

```
espefuse.py v3.0
The efuses to burn:
  from BLOCK0
     - SPI_BOOT_CRYPT_CNT

Burning efuses:

    - 'SPI_BOOT_CRYPT_CNT' (Enables encryption and decryption, when an SPI boot mode is set. Enabled when 1 or 3 bits are set,disabled otherwise) 0b111 -> 0b011

A fatal error occurred:         New value contains some bits that cannot be cleared (value will be 0b111)
```

[image]

negativekelvin, 2021-04-06:
Well your efuse already has value 7 so encryption is permanently enabled

noyelseth, 2021-04-06:
hi, @negativekelvin 

Is there any way to revert this encryption?

projectgus, 2021-04-12:
Hi @noyelseth,

I'm afraid there is no way to disable flash encryption if the SPI_BOOT_CRYPT_CNT efuse is already set to the maximum value. As [noted in the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/flash-encryption.html#disabling-flash-encryption), disabling flash encryption can only be done one time per chip.

Note that if your project is configured for Flash Encryption in Development mode, you can still reflash the chip without needing to disable flash encryption first: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/flash-encryption.html#re-flashing-updated-partitions

noyelseth, 2021-04-13:
Hello, @projectgus and @negativekelvin 

Now I want to update the firmware over OTA. 

But when I run ***idf.py build***, the code compiled as ***plaintext*** for that reason OTA update failed. 

So, what would I do to resolve this?

please help.

projectgus, 2021-04-13:
Hi @noyelseth,

During an OTA update the binary is downloaded by the ESP32-S2 as plaintext and then the ESP32-S2 encrypts it automatically when writing it to flash before booting it. This encryption happens automatically if using the OTA APIs inside ESP-IDF. If you need to also protect the firmware in transit, suggest using HTTPS with some kind of secure authentication scheme.

noyelseth, 2021-04-13:
Ok, Thanks, @projectgus 

Actually, I work remotely from home.
I flash a basic code for OTA into the ***esp32s2*** from my own laptop using this cmd ***idf.py encrypted-flash monitor***

Now, I build the operation code from the Office machine and update it via the OTA with the same configuration for both side

Is there any problem, If I do this?

[image]

projectgus, 2021-04-13:
Hi @noyelseth, that should be fine. In both cases the ESP32-S2 will receive non-encrypted binary and then encrypt it internally when it is written to flash.

projectgus, 2021-05-06:
Hi @noyelseth,

The esptool version in the master branch was updated in commit 7bb5c8cc82c647316802f051cff30a61ddcc2b5e and now contains the fix for burning SPI_BOOT_CRYPT_CNT.

The same fix will be backport to release/v4.2 branch in time for the v4.2.2 release.

closed 2021-05-06 by projectgus