espressif/esp-idf #6818: Connection dropping in BLE examples

Benedito821, 2021-04-03:
Hello! I am trying to use the BLE examples that are on this repository, specifically gatt_server and gatt_server_service table. The problem is that after some seconds the connection drops and advertising is restarted. On monitor I see these messages:

W (134977) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x8
**I (134977) GATTS_TABLE_DEMO: ESP_GATTS_DISCONNECT_EVT, reason = 0x8**
I (134997) GATTS_TABLE_DEMO: advertising start successfully

Is it a particular functionality of this example or really a problem? Help,please. I am using android and esp32-wroom-32 board.

RahulEspressif, 2021-04-05:
Hi @Benedito821 ,
   Reason value 0x8 points to connection timeout. Can you please share more details  like remote device used , steps used to reproduced issue ,  and complete logs at your end when issue occurs ?  
 

Thanks

Benedito821, 2021-06-05:
Here is the full log. I use a samsung smartphone as client and the app LightBlue to see BLE profiles. I compile from ubuntu's shell using idf.py commands. On the app LightBlue after the error occurs is written "Disconnect,data is stale".

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6744
load:0x40078000,len:14236
ho 0 tail 12 room 4
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (29) boot: ESP-IDF v4.4-dev-1404-gc13afea63 2nd stage bootloader
I (29) boot: compile time 04:12:09
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=165a8h ( 91560) map
I (146) esp_image: segment 1: paddr=000265d0 vaddr=3ffbdb60 size=03670h ( 13936) load
I (151) esp_image: segment 2: paddr=00029c48 vaddr=40080000 size=063d0h ( 25552) load
I (162) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=59024h (364580) map
I (294) esp_image: segment 4: paddr=0008904c vaddr=400863d0 size=0f938h ( 63800) load
I (321) esp_image: segment 5: paddr=0009898c vaddr=50000000 size=00010h (    16) load
I (331) boot: Loaded app from partition at offset 0x10000
I (332) boot: Disabling RNG early entropy source...
I (343) cpu_start: Pro cpu up.
I (343) cpu_start: Starting app cpu, entry point is 0x40081070
0x40081070: call_start_cpu1 at /home/benedito/esp/esp-idf/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (357) cpu_start: Pro cpu start user code
I (358) cpu_start: cpu freq: 160000000
I (358) cpu_start: Application information:
I (362) cpu_start: Project name:     gatt_server_demos
I (368) cpu_start: App version:      1
I (372) cpu_start: Compile time:     Jun  5 2021 04:11:58
I (378) cpu_start: ELF file SHA256:  9251a85e0253ea6c...
I (384) cpu_start: ESP-IDF:          v4.4-dev-1404-gc13afea63
I (391) heap_init: Initializing. RAM available for dynamic allocation:
I (398) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (404) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (410) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (416) heap_init: At 3FFC4D70 len 0001B290 (108 KiB): DRAM
I (422) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (429) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (435) heap_init: At 40095D08 len 0000A2F8 (40 KiB): IRAM
I (443) spi_flash: detected chip: generic
I (446) spi_flash: flash io: dio
W (450) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (464) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (498) BTDM_INIT: BT controller compile version [d03a5d3]
I (508) system_api: Base MAC address is not set
I (508) system_api: read default base MAC address from EFUSE
I (508) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (908) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

I (918) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (918) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1

I (938) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (938) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (938) GATTS_DEMO: the gatts demo char length = 3

I (948) GATTS_DEMO: prf_char[0] =11

I (948) GATTS_DEMO: prf_char[1] =22

I (958) GATTS_DEMO: prf_char[2] =33

I (958) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44

I (968) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

I (978) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44

I (978) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44

I (988) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44

I (4358) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 76:c0:7e:ac:94:08:
I (4358) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote 76:c0:7e:ac:94:08:
I (4988) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 30,latency = 0, timeout = 400
W (49078) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x8
I (49078) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x8

expresspotato, 2021-11-14:
Same problem here, to reproduce on IDF v4.3.1 fire up the HFP example, pair (SSP or PIN, doesn't matter), then wait 10 seconds.

```
W (16462) BT_HCI: DiscCmpl evt: hdl=129, rsn=0x8
W (16462) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
E (16462) LIBERTY: APP HFP event: CONNECTION_STATE_EVT
E (16472) LIBERTY: --connection state disconnected, peer feats 0x0, chld_feats 0x0
W (16472) LIBERTY: Disconnected
```

chewhs00, 2022-05-18:
I received this as well with  IDF v4.3.1 both on Android and IOS

```
W (3425136) BT_HCI: DiscCmpl evt: hdl=0, rsn=0x8
W (3425138) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x8
```

martinberlin, 2022-08-23:
Bump! Same here only difference is that I'm using ESP32S3-WROOM1 dev-board as GATT server. Just compiled the Client demo in another ESP32-WROVER that I have in the office. 
Get exactly the same:

```
I (11189) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote a8:03:2a:e1:f2:fe:
I (11189) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote a8:03:2a:e1:f2:fe:
I (12199) GATTS_DEMO: update connection params status = 16, min_int = 16, max_int = 32,conn_int = 12,latency = 0, timeout = 600
W (15379) BT_L2CAP: le con upd: err_stat=0x8
W (15379) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8
I (15379) GATTS_DEMO: update connection params status = 1, min_int = 0, max_int = 0,conn_int = 12,latency = 0, timeout = 600
I (15389) GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x8
```

Disconnect reason 0x8: Connection timeout.

In #3551 they recommend [increasing timeout value in connection settings](https://github.com/espressif/esp-idf/issues/3551#issuecomment-496472519)