espressif/esp-idf #6819: idf.py create-project  fails

cbaurtx, 2021-04-04:
installed esp.idf on a linux system and it works great except for:
idf.py create-project myproject

I get this output:

*Executing action: create-project
CMakeLists.txt not found in project directory /home/me/Development/Source/ESP32*

So was this dropped? Is the documentation (Web site) wrong?

dobairoland, 2021-04-04:
Hi @cbaurtx. This feature is available but maybe not in the IDF version you are using. Please tell us which version are you using, ie. the output of `idf.py --version` or `git describe --tags`.

cbaurtx, 2021-04-15:
#i got it all working. The latest version of my code builds fine

cbaurtx, 2021-04-15:
Check out the newest version which is compiling and working

closed 2021-04-15 by cbaurtx

dobairoland, 2021-04-15:
Thank you for the feedback!

mrhempman69, 2023-10-18:
This error exists again, such a joke

dobairoland, 2023-10-23:
> This error exists again, such a joke

If you still experience the issue then open a new one and provide all the necessary information for reproducing it.