espressif/esp-idf #6820: E (150677082) wifi:ieee80211_input.c 1585

surfskidude, 2021-04-04:
Based on the head branch, I have a smart config example running. My WiFi network turns off at night. In the morning when WiFi turns on again automatically I had one printout and all my smart config events stopped working and network did not come back online.

I (150674859) SmartConfig: WiFi disconnected, trying to reconnect...
I (150676911) SmartConfig: WiFi disconnected, trying to reconnect...
E (150677082) wifi:ieee80211_input.c 1585
---->>>no more printouts

The system is still running, but not the WiFi part.

AshUK, 2021-04-06:
Although probably different issues (I'm guessing the different line numbers) similar issue is described [here](https://www.esp32.com/viewtopic.php?p=74170).

zhangyanjiaoesp, 2021-04-06:
@surfskidude This branch  b3612b73 include the solution of this problem, please update your branch and test a again.

zhangyanjiaoesp, 2021-04-06:
@AshUK This branch 6acebbee on master has solved your problem, please update your branch and test again. And this solution has been backported to  release/v4.3, but it doesn't synchronize to github now, you can update v4.3 later.

surfskidude, 2021-04-19:
I have had it running for 4 days now with the latest code and it seems to be working :-)

surfskidude, 2021-04-21:
I think I need to re-open this. I believe there are still issues. This morning, I got the following error:

I (425360004) wifi:new:<6,1>, old:<6,0>, ap:<255,255>, sta:<6,1>, prof:1
I (426259907) wifi:new:<6,0>, old:<6,1>, ap:<255,255>, sta:<6,0>, prof:1
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40168db8  PS      : 0x00060530  A0      : 0x801909e2  A1      : 0x3ffc7350  
A2      : 0x3ffc6520  A3      : 0xffffffff  A4      : 0x3f84c610  A5      : 0x3ffe9500  
A6      : 0x3f844dd4  A7      : 0xb33fffff  A8      : 0x80168db8  A9      : 0x3ffc7330  
A10     : 0x000000c4  A11     : 0x000000c4  A12     : 0x3ffe94d0  A13     : 0x00060d23  
A14     : 0x0000cdcd  A15     : 0xb33fffff  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000cc  LBEG    : 0x40089c55  LEND    : 0x40089c79  LCOUNT  : 0x80171318  

Backtrace:0x40168db5:0x3ffc73500x401909df:0x3ffc7370 0x3ffc739d:0x3ffc73a0  |<-CORRUPTED

ELF file SHA256: f1ccfbd86f0d525e

Rebooting...

AxelLin, 2021-04-21:
What's your  IDF version? (run ``git describe --tags`` to find it)

surfskidude, 2021-04-22:
~/esp/esp-idf$ git describe --tags
v4.4-dev-930-ga6f33a9d3

Alvin1Zhang, 2021-06-08:
Thanks for reporting, fix on release/4.3 is available at https://github.com/espressif/esp-idf/commit/f00d7ee9d03f51c3da0d9606c2fc87c99a64b224, feel free to reopen.

closed 2021-06-08 by Alvin1Zhang