espressif/esp-idf #6821: Incorrect ethernet initialization sequence. PHY: LAN8720

Sejny, 2021-04-04:
I found bug in IDF ethernet driver. PHY LAN8720 initialization sequence is not correct. PHY clock is supplied by ESP32 (ETH_CLOCK_GPIO17_OUT), tested on ESP-IDF Release v3.3.5

**ESP must deliver PHY clock approx 40 clock cycles before PHY reset go HIGH.**

**Microchip note in LAN8720 datasheet:**
[image]

**Measured on real setup:**
[image]

Iam not skilled for repair this self :(

suda-morris, 2021-04-13:
Hi @Sejny May I ask how you do the PHY hardware reset in your code? In IDF 3.3, we didn't provide a "reset GPIO" configuration in the ethernet driver.

Alvin1Zhang, 2021-06-30:
Thanks for reporting, fix is available https://github.com/espressif/esp-idf/commit/fc60e0929562fbe9e033ed998eeb8760948c0f02, feel free to reopen.

closed 2021-06-30 by Alvin1Zhang