espressif/esp-idf #6822: product icorrect output

raoshubam96, 2021-04-05:
[](url)
[image]

we are facing this problem with Esp32 wrover B after we are molding it to PCB.

your associate has already wasted our 3-4 days by solving saying she will connect technician over the phone but now she says to genrate the query, so i would like to have a solution by today anyhow.

dobairoland, 2021-04-05:
Hi @raoshubam96. I'm sorry but this is insufficient amount of information for helping you. Please fill out the information as suggested in the issue template: https://github.com/espressif/esp-idf/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

The error is the most general error message from [esptool](https://github.com/espressif/esptool). I can suggest to invoke `esptool` directly and add the `--trace` argument. You can see the other arguments in the terminal after you built your project.

> we are facing this problem with Esp32 wrover B after we are molding it to PCB.

If you are using a custom PCB then most likely it is a hardware issue and we cannot do anything about it on the ESP-IDF and Esptool level. I can suggest to seek help with hardware issues here: https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review

raoshubam96, 2021-04-06:
hey, should attach circuit schematic and pcb design so then you can figure out the problem for us ? please? or is there anything else you would require along with it?

raoshubam96, 2021-04-06:
[image]
[image]
[PCB design.pdf](https://github.com/espressif/esp-idf/files/6262526/PCB.design.pdf)

please check the PCB design and circuit schematic.

raoshubam96, 2021-04-06:
and i am not able to send the PCB design as it doesnt support the file here.
is it possible to send it through email?

dobairoland, 2021-04-06:
Hi @raoshubam96. You can send the design files through the web page I linked earlier. However, if I understood your schematics well then you don't have an automatic reset circuitry which would reset the chip into bootloader mode. Connecting the TxD and RxD pins is not enough. The DTR and RTS signals should be used to generate an EN and IO0 reset sequence. It is also possible to put the chip into download mode by manual reset. However, I don't see any switch connected to your IO0.

ESP-IDF cannot flash the chip if it cannot be put into bootloader mode.

Here is also a working example of a board with proper reset circuitry:
https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch-20180607a.pdf

You can discuss hardware related questions on the forums as well:
https://www.esp32.com/viewforum.php?f=12&sid=192e6125de9fc392432482c3bd8865e1

I'm closing this because this is not related or cased by ESP-IDF.

closed 2021-04-06 by dobairoland

dobairoland, 2021-04-06:
More information on this topic: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection

raoshubam96, 2021-04-07:
hey just to add little imformation in that. we are using CP2102 to upload to codes. and we placed switch over pin on Wrover B to by giving the connection from ground but it didn't work. now would you connect us to your tech guy so then we can solve this problem because wrover and CP2102 are both bought from you, and its taking too long to solve the problem over here. and we are in middle of making a product and this is making it delay which can't afford it anymore. sp please connect your tech guy over phone.

raoshubam96, 2021-04-07:
[image]

raoshubam96, 2021-04-07:
hey, as you said to use the DTR and RTS signals should be used to generate an EN and IO0 reset sequence and we are doing that and we are prasing the switch of reset and IO0 while uploading the code but it still says failed to connect to moduel. 
and as you shared last schematic of CP2102 IC to upload the coding at the place of it we are using the CP2102 module which already the IC in it.
if you connect to somone then we could solve this problem asap please.