espressif/esp-idf #6823: v3.3版本 WPA2 Enterprise 无法连接（PEAP）

HeFeng1947, 2021-04-05:
## 环境

- Module or chip used: ESP32-WROOM-32U
- IDF version : v3.3 - bf022060964128556b3d3205b65c5d35df9beef6
- Build System: CMake
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## 问题描述

如题，**在v3.3版本**下无法跑通example下的wap2 enterprise例程，使用PEAP的认证方式。我尝试切换到目前release/v3.3分支的最新提交（7c86027531ebffe937c9d9d1080cd433eb993f35），也还是出现一样的错误。但在mster分支下（即v4.4），例程运行正常。

测试使用的认证服务器是freeRadius3，已在PC等设备上测试确认无问题。

**在v3.3版本例程里注释以下代码**

```c
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
	// 使用 PEAP 认证
    // ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_ca_cert(ca_pem_start, ca_pem_bytes) );
    // ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_cert_key(client_crt_start, client_crt_bytes,\
    // 		client_key_start, client_key_bytes, NULL, 0) );
    ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EXAMPLE_EAP_ID, strlen(EXAMPLE_EAP_ID)) );
    if (EXAMPLE_EAP_METHOD == EAP_PEAP || EXAMPLE_EAP_METHOD == EAP_TTLS) {
        ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EXAMPLE_EAP_USERNAME, strlen(EXAMPLE_EAP_USERNAME)) );
        ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EXAMPLE_EAP_PASSWORD, strlen(EXAMPLE_EAP_PASSWORD)) );
    }
```

**esp32错误log输出（v3.3）**

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6852
load:0x40078000,len:12128
load:0x40080400,len:6692
entry 0x40080774
I (63) boot: Chip Revision: 1
I (64) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v3.3.2-323-gbf0220609-dirty 2nd stage bootloader
I (39) boot: compile time 11:40:03
I (40) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 80MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 4MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1cc10 (117776) map
I (146) esp_image: segment 1: paddr=0x0002cc38 vaddr=0x3ffb0000 size=0x033d8 ( 13272) load
I (151) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x6e1fc (451068) map
0x400d0018: _stext at ??:?

I (284) esp_image: segment 3: paddr=0x0009e21c vaddr=0x3ffb33d8 size=0x00238 (   568) load
I (284) esp_image: segment 4: paddr=0x0009e45c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/kelly/sambaShare/workspace/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (291) esp_image: segment 5: paddr=0x0009e864 vaddr=0x40080400 size=0x13d54 ( 81236) load
I (340) boot: Loaded app from partition at offset 0x10000
I (340) boot: Disabling RNG early entropy source...
[... 460 lines cut ...]
Mon Apr  5 08:49:59 2021 daemon.notice hostapd: wlan0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Mon Apr  5 08:50:00 2021 daemon.notice hostapd: wlan0: CTRL-EVENT-EAP-SUCCESS2 98:f4:ab:0b:24:84
Mon Apr  5 08:50:00 2021 daemon.info hostapd: wlan0: STA 98:f4:ab:0b:24:84 WPA: pairwise key handshake completed (RSN)
Mon Apr  5 08:50:00 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 98:f4:ab:0b:24:84
Mon Apr  5 08:50:00 2021 daemon.info hostapd: wlan0: STA 98:f4:ab:0b:24:84 RADIUS: starting accounting session 4FD18DAE5004CB60
Mon Apr  5 08:50:00 2021 daemon.info hostapd: wlan0: STA 98:f4:ab:0b:24:84 IEEE 802.1X: authenticated - EAP type: 25 (PEAP)
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPDISCOVER(br-lan) 98:f4:ab:0b:24:84
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPOFFER(br-lan) 192.168.1.194 98:f4:ab:0b:24:84
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPDISCOVER(br-lan) 98:f4:ab:0b:24:84
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPOFFER(br-lan) 192.168.1.194 98:f4:ab:0b:24:84
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPDISCOVER(br-lan) 98:f4:ab:0b:24:84
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPOFFER(br-lan) 192.168.1.194 98:f4:ab:0b:24:84
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPDISCOVER(br-lan) 98:f4:ab:0b:24:84
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPOFFER(br-lan) 192.168.1.194 98:f4:ab:0b:24:84
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPREQUEST(br-lan) 192.168.1.194 98:f4:ab:0b:24:84
Mon Apr  5 08:50:03 2021 daemon.info dnsmasq-dhcp[1440]: DHCPACK(br-lan) 192.168.1.194 98:f4:ab:0b:24:84 espressif
```

正常连接和失败的服务器端log主要区分是，连接失败没有这一条log输出：

`Mon Apr  5 08:50:00 2021 daemon.info hostapd: wlan0: STA 98:f4:ab:0b:24:84 WPA: pairwise key handshake completed (RSN)`

## 问题补充

我翻阅了近几年idf的issue，wpa2的问题出现频率很高，我尝试了其中几个解决方法，例如 https://github.com/espressif/esp-idf/issues/4347#issuecomment-559504115 ，但都没有效果。相同的设备环境，我已经在v4.4上成功运行例程，但是两个版本间IDF的区别太大，无法简单的移植组件。如果你们已经在v4.4上成功修复了这个bug，请尽快更新到v3.3版本，或提供一个可用的补丁以满足我们客户的迫切需求。

sagb2015, 2021-04-09:
@HeFeng1947 Is it possible for you to provide sniffer capture and certificates (if they are not confidential) Please also share hostapd configuration.

HeFeng1947, 2021-04-16:
> @HeFeng1947 Is it possible for you to provide sniffer capture and certificates (if they are not confidential) Please also share hostapd configuration.

Our authentication server is FreeRadius3, which is built into OpenWrt. I can just package the entire FreeRadius3 configuration and send it to you. If you have requirements, I can directly send my test router to you for testing.

HeFeng1947, 2021-04-16:
[freeradius3.zip](https://github.com/espressif/esp-idf/files/6322975/freeradius3.zip)

My wireless config：
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'platform/ar933x_wmac'
        option htmode 'HT20'
        option disabled '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid '0-radius'
        option encryption 'wpa2'
        option key 'goodlife'
        option auth_port '1812'
        option auth_server '192.168.1.1'
```

TianaESP, 2021-04-16:
Hi @HeFeng1947,

Thanks for the reporting. That would be really helpful if you could send the test router to us. Can you please contact our Shanghai office? We would ask our colleague to support with priority. 

The email address: **sales@espressif.com**
Please briefly describe the issue, including the GitHub ticket number, and tell us you already have communication with our WiFi team. 

Thanks.
Tiana

sagb2015, 2021-04-16:
@HeFeng1947 Additionally for faster debugging (until we have to device), can you help with the following 
1) Enable supplicant debugging => CONFIG_WPA_DEBUG_PRINT=y and provide the logs with v3.3 release (The flag is available on latest release/v3.3)
2)  Test with master with CONFIG_WPA_MBEDTLS_CRYPTO=n. By default this flag is enabled on master. Please also keep CONFIG_WPA_DEBUG_PRINT=y.

HeFeng1947, 2021-04-19:
> @HeFeng1947 Additionally for faster debugging (until we have to device), can you help with the following
>
> 1. Enable supplicant debugging => CONFIG_WPA_DEBUG_PRINT=y and provide the logs with v3.3 [...]

1. v3.3 Enable supplicant debugging => CONFIG_WPA_DEBUG_PRINT=y
```
Toolchain path: /home/zsw/.espressif/tools/xtensa-esp32-elf/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-97-gc752ad5
Compiler version: 5.2.0
Python requirements from /home/zsw/esp/esp-idf/requirements.txt are satisfied.
MONITOR
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6760
load:0x40078000,len:12152
load:0x40080400,len:6664
entry 0x40080774
[0;32mI (71) boot: Chip Revision: 1[0m
[0;32mI (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (39) boot: ESP-IDF v3.3.5-14-g7c8602753-dirty 2nd stage bootloader[0m
[0;32mI (39) boot: compile time 11:14:25[0m
[0;32mI (39) boot: Enabling RNG early entropy source...[0m
[0;32mI (45) boot: SPI Speed      : 40MHz[0m
[0;32mI (49) boot: SPI Mode       : DIO[0m
[0;32mI (53) boot: SPI Flash Size : 4MB[0m
[0;32mI (57) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (83) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (90) boot: End of partition table[0m
[0;32mI (95) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1b3e0 (111584) map[0m
[0;32mI (150) esp_image: segment 1: paddr=0x0002b408 vaddr=0x3ffb0000 size=0x02fd4 ( 12244) load[0m
[0;32mI (155) esp_image: segment 2: paddr=0x0002e3e4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (157) esp_image: segment 3: paddr=0x0002e7ec vaddr=0x40080400 size=0x01824 (  6180) load[0m
[0;32mI (168) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x6cb98 (445336) map[0m
[0;32mI (331) esp_image: segment 5: paddr=0x0009cbb8 vaddr=0x40081c24 size=0x11f54 ( 73556) load[0m
[0;32mI (373) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (373) boot: Disabling RNG early entropy source...[0m
[0;32mI (374) cpu_start: cpu freq: 160[0m
[0;32mI (377) cpu_start: Pro cpu up.[0m
[0;32mI (381) cpu_start: Application information:[0m
[0;32mI (386) cpu_start: Project name:     wpa2-enterprise[0m
[0;32mI (392) cpu_start: App version:      v3.3.5-14-g7c8602753-dirty[0m
[0;32mI (398) cpu_start: Compile time:     Apr 19 2021 11:14:21[0m
[0;32mI (404) cpu_start: ELF file SHA256:  6ea7a67254d457da...[0m
[0;32mI (410) cpu_start: ESP-IDF:          v3.3.5-14-g7c8602753-dirty[0m
[0;32mI (417) cpu_start: Starting app cpu, entry point is 0x40081170[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (427) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (434) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (440) heap_init: At 3FFB8720 len 000278E0 (158 KiB): DRAM[0m
[0;32mI (447) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (453) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (459) heap_init: At 40093B78 len 0000C488 (49 KiB): IRAM[0m
[0;32mI (466) cpu_start: Pro cpu start user code[0m
[0;32mI (148) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
I (221) wifi:wifi driver task: 3ffc031c, prio:23, stack:3584, core=0
[0;32mI (221) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (221) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (261) wifi:wifi firmware version: dc30037
I (261) wifi:config NVS flash: enabled
I (261) wifi:config nano formating: disabled
[... 263 lines cut ...]
[0;32mI (5073) example: MASK:0.0.0.0[0m
[0;32mI (5073) example: GW:0.0.0.0[0m
[0;32mI (5073) example: ~~~~~~~~~~~[0m
[0;32mI (5143) esp_netif_handlers: sta ip: 192.168.1.211, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (7073) example: ~~~~~~~~~~~[0m
[0;32mI (7073) example: IP:192.168.1.211[0m
[0;32mI (7073) example: MASK:255.255.255.0[0m
[0;32mI (7073) example: GW:192.168.1.1[0m
[0;32mI (7073) example: ~~~~~~~~~~~[0m
[0;32mI (9073) example: ~~~~~~~~~~~[0m
[0;32mI (9073) example: IP:192.168.1.211[0m
[0;32mI (9073) example: MASK:255.255.255.0[0m
[0;32mI (9073) example: GW:192.168.1.1[0m
[0;32mI (9073) example: ~~~~~~~~~~~[0m
W (9753) wifi:<ba-add>idx:0 (ifx:0, e4:95:6e:41:b3:dc), tid:0, ssn:1, winSize:64
[0;32mI (11073) example: ~~~~~~~~~~~[0m
[0;32mI (11073) example: IP:192.168.1.211[0m
[0;32mI (11073) example: MASK:255.255.255.0[0m
[0;32mI (11073) example: GW:192.168.1.1[0m
[0;32mI (11073) example: ~~~~~~~~~~~[0m
[0;32mI (13073) example: ~~~~~~~~~~~[0m
[0;32mI (13073) example: IP:192.168.1.211[0m
[0;32mI (13073) example: MASK:255.255.255.0[0m
```
From the log point of view, it is possible to connect to an AP whether it is turned on CONFIG_WPA_MBEDTLS_CRYPTO or not, but with CONFIG_WPA_MBEDTLS_CRYPTO= N, the connection is faster

HeFeng1947, 2021-04-19:
> Hi @HeFeng1947,
> 
> Thanks for the reporting. That would be really helpful if you could send the test router to us. Can you please contact our Shanghai office? We would ask our colleague to support with priority.
> 
> The email address: **[sales@espressif.com](mailto:sales@espressif.com)**
> Please briefly describe the issue, including the GitHub ticket number, and tell us you already have communication with our WiFi team.
> 
> Thanks.
> Tiana

hi @TianaESP , I have sent the email and I am waiting for your reply！

sagb2015, 2021-05-19:
Hi @HeFeng1947 , Please try with the following in the radius config that your shared (mods-available/eap).
```
 disable_tlsv1_2 = no
 disable_tlsv1_1 = no
 disable_tlsv1 = no
 tls_min_version = "1.0"
```

HeFeng1947, 2021-05-20:
> Hi @HeFeng1947 , please try with the following in the radius config that your shared (mods-available/eap). We do not have support for TLS v1.2 on IDF v3.3.
> 
> ```
>  disable_tlsv1_2 = no
>  disable_tlsv1_1 = no
>  disable_tlsv1 = no
>  tls_min_version = "1.0"
> ```

Why not support TLS1.2? Do you have any plans to add this feature? If we need to add this support, what files need to be modified?

hpdhabe, 2021-05-20:
Hi @HeFeng1947 , Can you please tryout the attached path that adds support for TLS1.2 and let us know if this works. The patch was build upon the latest commit of release/v3.3 [7c86027](https://github.com/espressif/esp-idf/commit/7c86027531ebffe937c9d9d1080cd433eb993f35) and is also tested to be working on the commit (https://github.com/espressif/esp-idf/commit/bf022060964128556b3d3205b65c5d35df9beef6) that you have mentioned.

[patch.zip](https://github.com/espressif/esp-idf/files/6517028/patch.zip)

sagb2015, 2021-05-20:
@HeFeng1947  We do have support for TLS1.2, but from v4.0 onwards. We do not intend to backport it onto 3.3 since the release is in maintenance only phase (Reference [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#support-periods)).

HeFeng1947, 2021-05-21:
> Hi @HeFeng1947 , Can you please tryout the attached path that adds support for TLS1.2 and let us know if this works. The patch was build upon the latest commit of release/v3.3 [7c86027](https://github.com/espressif/esp-idf/commit/7c86027531ebffe937c9d9d1080cd433eb993f35) and is also tested to be directly applicable on the commit ([bf02206](https://github.com/espressif/esp-idf/commit/bf022060964128556b3d3205b65c5d35df9beef6)) that you have mentioned.
> 
> [patch.zip](https://github.com/espressif/esp-idf/files/6517028/patch.zip)

Thank you very much! We will test it and give you feedback as soon as possible.

HeFeng1947, 2021-05-22:
I have completed the test in Demo and it works perfect!  Thank you very much!! @sagb2015 @hpdhabe @TianaESP

sagb2015, 2021-05-22:
Good to know! Feel free to reopen if you need any other help.

closed 2021-05-22 by sagb2015