espressif/esp-idf #6826: mdf mqtt example root node restart frequently by heap corrupt

rqbh, 2021-04-06:
platform : fedora linux 34
software : esp_mdf release/v1.0
hardware: node esp32-cam,  root TTGO display

root software: 

```
void root_write_task(void *arg)
{
    mdf_err_t ret = MDF_OK;
    char *data = NULL;
    size_t size = MWIFI_PAYLOAD_LEN;
    uint8_t src_addr[MWIFI_ADDR_LEN] = { 0x0 };
    mwifi_data_type_t data_type = { 0x0 };

    MDF_LOGI("Root write task is running");

    while (esp_mesh_is_root()) {
        if (!mwifi_get_root_status()) {
            vTaskDelay(500 / portTICK_RATE_MS);
            continue;
        }

        /**
         * @brief Recv data from node, and forward to mqtt server.
         */
        ret = mwifi_root_read(src_addr, &data_type, &data, &size, portMAX_DELAY);
        MDF_ERROR_GOTO(ret != MDF_OK, MEM_FREE, "<%s> mwifi_root_read", mdf_err_to_name(ret));

        switch(data_type.protocol){
            case NAV_PROTO_BYTES:
                ret = mesh_mqtt_write(src_addr, data, size, MESH_MQTT_DATA_BYTES);
                break;
            case NAV_PROTO_JSON:
                ret = mesh_mqtt_write(src_addr, data, size, MESH_MQTT_DATA_JSON);
                break;
            default:
                ret = mesh_mqtt_write(src_addr, data, size, MESH_MQTT_DATA_JSON);
                break;
        }

        MDF_ERROR_GOTO(ret != MDF_OK, MEM_FREE, "<%s> mesh_mqtt_publish", mdf_err_to_name(ret));

MEM_FREE:
        MDF_FREE(data);
    }
    MDF_LOGW("Root write task is exit");
    mesh_mqtt_stop();
    vTaskDelete(NULL);
}
```

node software:

```
mdf_err_t send_package_to_server(mwifi_data_type_t *data_type, char *data, int data_len)
{
   data_type.protocol = NAV_PROTO_BYTES;
    if (data_len < 8096)   {
        return mwifi_write(NULL, data_type, data, data_len, true);
    }
    else   {
        mdf_err_t ret = MDF_OK;
        for (int i = 0; data_len > 0; i++)      {
            int send_size = data_len > 8095 ? 8095 : data_len;
            MDF_LOGI("send pkg, size: %d", send_size);
            ret = mwifi_write(NULL, data_type, data, send_size, true);
            if (ret != MDF_OK)      {
                return ret;
            }
            vTaskDelay(30);
            data_len -= send_size;
            data += send_size;
        }
    }
    return MDF_OK;
}

```

node device is running OK.
but root device is restart frequently when data size is 20K  more or less,  the most fault message is below:
```
......
......
 (7225) [mwifi, 239]: State represents: 1
I (7228) [mqtt_examples, 256]: event_loop_cb, event: 11
W (7337) [mwifi, 1309]: Part of the packet is lost, expect_seq: 0, recv_seq: 2
I (11568) [mqtt_examples, 204]: System information, channel: 12, layer: 1, self mac: 84:cc:a8:61:03:1c, parent bssid: 00:5a:39:19:51:8e, parent rssi: -73, node num: 2, free heap: 119544
I (11575) [mqtt_examples, 207]: Child mac: 9c:9c:1f:ea:81:74
I (21568) [mqtt_examples, 204]: System information, channel: 12, layer: 1, self mac: 84:cc:a8:61:03:1c, parent bssid: 00:5a:39:19:51:8e, parent rssi: -69, node num: 2, free heap: 119548
I (21575) [mqtt_examples, 207]: Child mac: 9c:9c:1f:ea:81:74
W (29650) [mwifi, 1309]: Part of the packet is lost, expect_seq: 2, recv_seq: 0
W (29660) [mwifi, 1309]: Part of the packet is lost, expect_seq: 0, recv_seq: 2
CORRUPT HEAP: multi_heap.c:432 detected at 0x4a4a0894
abort() was called at PC 0x4008adbb on core 0
0x4008adbb: multi_heap_assert at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/heap/multi_heap.c:432

ELF file SHA256: 4a16f17e5f9c3960

Backtrace: 0x400873ac:0x3ffb7a60 0x400875dd:0x3ffb7a80 0x4008adbb:0x3ffb7aa0 0x40082701:0x3ffb7ac0 0x40082732:0x3ffb7ae0 0x4008303d:0x3ffb7b00 0x4000beaf:0x3ffb7b20 0x400814b9:0x3ffb7b40 0x40093139:0x3ffb7b60 0x40094f17:0x3ffb7b90 0x40091e2a:0x3ffb7c10 0x40087db2:0x3ffb7c40
0x400873ac: invoke_abort at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/esp32/panic.c:715

0x400875dd: abort at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/esp32/panic.c:715

0x4008adbb: multi_heap_assert at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/heap/multi_heap.c:380
 (inlined by) multi_heap_malloc_impl at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/heap/multi_heap.c:432

0x40082701: heap_caps_malloc at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/heap/heap_caps.c:232

0x40082732: heap_caps_malloc_default at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/heap/heap_caps.c:232

0x4008303d: _malloc_r at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/newlib/syscalls.c:37

0x400814b9: wifi_malloc at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/esp32/esp_adapter.c:68

0x40093139: esf_buf_alloc at ??:?

0x40094f17: wdevProcessRxSucDataAll at ??:?

0x40091e2a: ppTask at ??:?

0x40087db2: vPortTaskWrapper at /run/media/stones/9c1031e3-2561-4172-b406-ec1e161bd067/kevin/nav_light/esp-mdf-1.0/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jul 29 2019 12:21:46
......
......
```

is this a mdf error or a idf error ? mostly a idf error CORRUPT HEAP
how can debug it ?

AxelLin, 2022-02-13:
> software : esp_mdf release/v1.0
> 

You should report to esp-mdf instead.

Alvin1Zhang, 2022-11-16:
Thanks for reporting, please help report issues at issue tracker https://github.com/espressif/esp-mdf/issues. Feel free to reopen. Thanks.

closed 2022-11-16 by Alvin1Zhang