espressif/esp-idf #6829: Deep-sleep RTC time keeping broken in v4.3 and v4.4 (master)

andersjohnsen, 2021-04-06:
## Problem Description

When using the internal 150kHz crystal for RTC time keeping in deep sleep, the device will take up with a random sleep duration. Observations includes for a 10s deep-sleep loop:

- Positive **and negative** sleep durations in the range of +/- minutes and hours.
- Sensitive to light/heat (more unstable when exposed to sunlight).
- Not a problem with v4.2.
- Build up over time (e.g initially there may be no issues, but within a few hours the RTC is all over the place).

I know that time drift is to be expected when using the internal 150kHz clock, but this is order of magnitudes worse.

This is using mainly ESP32-WROOM-32D and ESP32-WROOM-32E.

Let me know what information I can otherwise provide.

igrr, 2021-04-06:
Hi @andersjohnsen, could you clarify the method you are using to measure sleep duration, that results in a negative value? Is the sleep time measured as part of the application, or externally? If it is the former, could you post a code snippet that shows this?

andersjohnsen, 2021-04-06:
Yes, measured part of the application (calling `gettimeofday` before and after).

I'll work on a snippet (we're running a fairly complex stack on top, but I should be able to slim it down).

KonstantinKondrashov, 2021-04-19:
Hi @andersjohnsen!
There was a fix on master 7ebefe2e2bc8f905ec3f9c52525a83fa85bf276d
and backport for v4.3 1e6a6ffa0dadc55fb9e86dce6d27852ddff8012a
Could you try this? Does it help with this issue?

andersjohnsen, 2021-04-19:
Hey,

Thank you, and sorry for not getting back to you. Gonna try it out
immediately! Can you confirm it's not for esp32s2-only as the change
suggests?

Cheers!

KonstantinKondrashov, 2021-04-19:
@andersjohnsen
Yes, It is not only for S2, but for ESP32 as well.

andersjohnsen, 2021-04-19:
@KonstantinKondrashov Great!

I can't find the v4.3 commit though: https://github.com/espressif/esp-idf/commit/1e6a6ffa0dadc55fb9e86dce6d27852ddff8012a

KonstantinKondrashov, 2021-04-19:
backport for v4.3 was not published yet on GitHub. Use the patch
[13046.zip](https://github.com/espressif/esp-idf/files/6334030/13046.zip)

andersjohnsen, 2021-04-20:
The patch indeed appears to fix the issue for v4.3! Thank you

closed 2021-04-23 by Alvin1Zhang