espressif/esp-idf #6830: python -m pip install failure

AxelLin, 2021-04-06:
## Environment

Module or chip used: ESP32-WROOM-32E
IDF version: v4.4-dev-744-g1cb31e50943b
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: Linux (Ubuntu 20.04.2 LTS)
Power Supply: USB

## Problem Description

I'm trying to test nimble example codes, but I got below error:
https://gist.github.com/AxelLin/64782a2ceb6bc9575fe7504bbb2a8a3a

PS. I already have dbus installed.

$ dpkg -l |grep dbus
ii  at-spi2-core                               2.36.0-2                                   amd64        Assistive Technology Service Provider Interface (dbus core)
ii  dbus                                       1.12.16-2ubuntu2.1                         amd64        simple interprocess messaging system (daemon and utilities)
ii  dbus-user-session                          1.12.16-2ubuntu2.1                         amd64        simple interprocess messaging system (systemd --user integration)
ii  dbus-x11                                   1.12.16-2ubuntu2.1                         amd64        simple interprocess messaging system (X11 deps)
ii  fcitx-module-dbus                          1:4.2.9.7-3                                amd64        Flexible Input Method Framework - D-Bus module and IPC frontend
ii  gir1.2-dbusmenu-glib-0.4:amd64             16.04.1+18.10.20180917-0ubuntu6            amd64        typelib file for libdbusmenu-glib4
ii  libdbus-1-3:amd64                          1.12.16-2ubuntu2.1                         amd64        simple interprocess messaging system (library)
ii  libdbus-glib-1-2:amd64                     0.110-5fakssync1                           amd64        deprecated library for D-Bus IPC
ii  libdbusmenu-glib4:amd64                    16.04.1+18.10.20180917-0ubuntu6            amd64        library for passing menus over DBus
ii  libdbusmenu-gtk3-4:amd64                   16.04.1+18.10.20180917-0ubuntu6            amd64        library for passing menus over DBus - GTK+ version
ii  libdbusmenu-gtk4:amd64                     16.04.1+18.10.20180917-0ubuntu6            amd64        library for passing menus over DBus - GTK+ version
ii  libdbusmenu-qt5-2:amd64                    0.9.3+16.04.20160218-2build1               amd64        Qt implementation of the DBusMenu protocol
ii  libnet-dbus-perl                           1.2.0-1                                    amd64        Perl extension for the DBus bindings
ii  libnih-dbus1:amd64                         1.0.3-6ubuntu2                             amd64        NIH D-Bus Bindings Library
ii  libqt5dbus5:amd64                          5.12.8+dfsg-0ubuntu1                       amd64        Qt 5 D-Bus module
ii  python-dbus                                1.2.16-1build1                             amd64        simple interprocess messaging system (Python interface)
ii  python3-dbus                               1.2.16-1build1                             amd64        simple interprocess messaging system (Python 3 interface)
ii  python3-dbus.mainloop.pyqt5                5.14.1+dfsg-3build1                        amd64        D-Bus Qt main loop support for Python 3
ii  xdg-dbus-proxy                             0.1.2-1                                    amd64        filtering D-Bus proxy

dobairoland, 2021-04-06:
Thank you @AxelLin. I'm linking here the original report: https://github.com/espressif/esp-idf/issues/6656#issuecomment-813986881

AxelLin, 2021-04-28:
Hi @dobairoland 

If it takes time to get the fix, is there any workaround to make it work (so I can continue my test).

Now I cannot execute the python code in the example code due to this issue:
$ python bleprph_test.py
Traceback (most recent call last):
  File "bleprph_test.py", line 30, in <module>
    import ttfw_idf
ModuleNotFoundError: No module named 'ttfw_idf'

dobairoland, 2021-04-28:
@AxelLin Please try the following:
```sh
export PYTHONPATH="$IDF_PATH/tools:$IDF_PATH/tools/ci/python_packages:$PYTHONPATH"
```

AxelLin, 2021-04-28:
```
axel@phoenix:~/esp/esp-idf/examples/bluetooth/nimble/bleprph$ export PYTHONPATH="$IDF_PATH/tools:$IDF_PATH/tools/ci/python_packages:$PYTHONPATH"
axel@phoenix:~/esp/esp-idf/examples/bluetooth/nimble/bleprph$ python bleprph_test.py
Traceback (most recent call last):
  File "bleprph_test.py", line 30, in <module>
    import ttfw_idf
  File "/home/axel/esp/esp-idf/tools/ci/python_packages/ttfw_idf/__init__.py", line 21, in <module>
    import junit_xml
ModuleNotFoundError: No module named 'junit_xml'
```

dobairoland, 2021-04-28:
```sh
python -m pip install -r $IDF_PATH/tools/ci/python_packages/ttfw_idf/requirements.txt
```

AxelLin, 2021-04-28:
After finish all above steps, now I got below error:
```
$ python bleprph_test.py 
No module named 'dbus'
Install packages `libgirepository1.0-dev gir1.2-gtk-3.0 libcairo2-dev libdbus-1-dev libdbus-glib-1-dev` for resolving the issue
Run `pip install -r $IDF_PATH/tools/ble/requirements.txt` for resolving the issue
Traceback (most recent call last):
  File "bleprph_test.py", line 31, in <module>
    from ble import lib_ble_client
  File "/home/axel/esp/esp-idf/tools/ble/lib_ble_client.py", line 27, in <module>
    import dbus
ModuleNotFoundError: No module named 'dbus'
```
Seem back to the original issue I reported.

dobairoland, 2021-04-28:
You have reported earlier that you have run:
```sh
python -m pip install -r $IDF_PATH/tools/ble/requirements.txt
```

Still isn't working?

AxelLin, 2021-04-28:
Still got below error:
https://gist.github.com/AxelLin/f73b6e32767e794a518a03fd4617cf49

chegewara, 2021-04-28:
Just idea, maybe python version is the issue?
You are still using python 2.7.

dobairoland, 2021-04-28:
`libdbus-1-dev` and `libglib2.0-dev` Ubuntu packages seem to be needed before running `python -m pip install -r $IDF_PATH/tools/ble/requirements.txt`. The `dbus-python` package depends on them.

dobairoland, 2021-04-28:
The above is also advised in your log seen in https://github.com/espressif/esp-idf/issues/6830#issuecomment-828374664:

```
Install packages `libgirepository1.0-dev gir1.2-gtk-3.0 libcairo2-dev libdbus-1-dev libdbus-glib-1-dev` for resolving the issue
```

AxelLin, 2021-04-28:
> The above is also advised in your log seen in [#6830 (comment)](https://github.com/espressif/esp-idf/issues/6830#issuecomment-828374664):
> 
> ```
> Install packages `libgirepository1.0-dev gir1.2-gtk-3.0 libcairo2-dev libdbus-1-dev libdbus-glib-1-dev` for resolving the issue
> ```

After install above package, the "python -m pip install -r $IDF_PATH/tools/ble/requirements.txt" works now.
Now I got below

```
$ python3 bleprph_test.py 
starting running test: test_example_app_ble_peripheral

Can't down device hci0: Operation not permitted (1)
Traceback (most recent call last):
  File "/home/axel/esp/esp-idf/tools/ci/python_packages/tiny_test_fw/TinyFW.py", line 206, in handle_test
    test_func(env_inst, extra_data)
  File "bleprph_test.py", line 134, in test_example_app_ble_peripheral
    subprocess.check_output(['hciconfig','hci0','reset'])
  File "/usr/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['hciconfig', 'hci0', 'reset']' returned non-zero exit status 1.
Test Fail: test_example_app_ble_peripheral

$ sudo python3 bleprph_test.py 
Traceback (most recent call last):
  File "bleprph_test.py", line 30, in <module>
    import ttfw_idf
ModuleNotFoundError: No module named 'ttfw_idf'

```

igrr, 2021-04-28:
(Not a BLE developer, just recall having this issue in the past.)
Something like this can be used to allow running hciconfig as an unprivileged user:
```
sudo setcap 'cap_net_raw,cap_net_admin+eip' `which hciconfig`
```
You may need to do the same for `hcitool`. On ubuntu, `setcap` tool is provided by `libcap2-bin` package.

shivani-tipnis, 2021-04-28:
Hi @AxelLin 

For the error `dbus-1` not found, the installation flow expected to do is - 
1. Install packages `libgirepository1.0-dev gir1.2-gtk-3.0 libcairo2-dev libdbus-1-dev libdbus-glib-1-dev` and then
2. Run `pip install -r $IDF_PATH/tools/ble/requirements.txt` 

Then, you should not get the `No package 'dbus-1' found` issue when installing dbus-python.

Since it is a ble requirement only, it is added here when running the ble tests.

chegewara, 2021-04-28:
> ModuleNotFoundError: No module named 'ttfw_idf'

https://github.com/espressif/esp-idf/issues/6830#issuecomment-828348550

AxelLin, 2021-04-29:
> (Not a BLE developer, just recall having this issue in the past.)
> Something like this can be used to allow running hciconfig as an unprivileged user:
>
> ```
> sudo setcap 'cap_net_raw,cap_net_admin+eip' [...]

This helps, thanks.

closed 2021-04-29 by AxelLin