espressif/esp-idf #6831: Secure boot 在不同分支，不同提交下的不兼容问题

gl-zhongshangwen, 2021-04-06:
# 环境：

- Module or chip used: ESP32-WROOM-32U
- Build System: CMake
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

# 问题描述：

Secure boot 在不同分支，不同提交下，基于examples/get-started/hello_world 例程 partition table 和 secure_boot_signing_key.pem 相同的情况下，flash app后设备无法启动的问题。

以下是我们测试的IDF版本和各版本的操作步骤：

- release/V3.3 - [bf022060](https://github.com/espressif/esp-idf/commit/bf022060964128556b3d3205b65c5d35df9beef6) 
- release/V3.3 - [7c8602753](https://github.com/espressif/esp-idf/commit/7c86027531ebffe937c9d9d1080cd433eb993f35) 
- master - [8bc19ba8](https://github.com/espressif/esp-idf/commit/8bc19ba893e5544d571a753d82b44a84799b94b1) 

## 一、release/V3.3 - [bf022060](https://github.com/espressif/esp-idf/commit/bf022060964128556b3d3205b65c5d35df9beef6) :

注:bootloader 已经提前在release/V3.3 - [bf022060](https://github.com/espressif/esp-idf/commit/bf022060964128556b3d3205b65c5d35df9beef6) 烧入完成

### 1.配置

```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
CONFIG_APP_EXCLUDE_PROJECT_VER_VAR=
CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR=
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE=
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_APP_ROLLBACK_ENABLE=

#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
[... 5304 lines cut ...]
I (120) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (127) boot:  2 factory          factory app      00 00 00010000 00100000
I (135) boot: End of partition table
I (139) boot_comm: chip revision: 1, min. application chip revision: 0
I (146) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x06018 ( 24600) map
I (164) esp_image: segment 1: paddr=0x00016040 vaddr=0x3ffb0000 size=0x0206c (  8300) load
I (167) esp_image: segment 2: paddr=0x000180b4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/zsw/esp/ESP_SDK/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (173) esp_image: segment 3: paddr=0x000184bc vaddr=0x40080400 size=0x07b5c ( 31580) load
I (195) esp_image: segment 4: paddr=0x00020020 vaddr=0x400d0020 size=0x134ec ( 79084) map
0x400d0020: _stext at ??:?

I (223) esp_image: segment 5: paddr=0x00033514 vaddr=0x40087f5c size=0x020cc (  8396) load
0x40087f5c: wdt_hal_is_enabled at /home/zsw/esp/ESP_SDK/esp-idf/components/hal/wdt_hal_iram.c:193

I (227) esp_image: segment 6: paddr=0x000355e8 vaddr=0x00000000 size=0x0a998 ( 43416) 
I (245) esp_image: Verifying image signature...
E (647) esp_image: Secure boot signature verification failed
I (647) esp_image: Calculating simple hash to check for corruption...
W (701) esp_image: image valid, signature bad
E (701) boot: Factory app partition is not bootable
E (701) boot: No bootable app partitions in the partition table

```

Alvin1Zhang, 2021-04-23:
@gl-zhongshangwen 

感谢提供如此具体的问题报告。我们已经在 release/3.3 分支上复现了 espsecure.py 问题（即您提到的问题二），我们已经找到修复方案，该修复将很快被合入，后续会更新此问题的状态。

您提到 master branch 的问题 （即您提到的问题三 ），我们仍在努力复现。您能帮检查并验证下用于 app build 的 "secure_boot_signing_key.pem" 和用于构建 flash 中 bootloader.bin 的"secure_boot_signing_key.pem" 是否相同？

谢谢。

gl-zhongshangwen, 2021-04-23:
Hi,Alvin1Zhang

    您能帮检查并验证下用于 app build 的 "secure_boot_signing_key.pem" 和用于构建 flash 中 bootloader.bin 的"secure_boot_signing_key.pem" 是否相同？
        =>这个问题三是毕现的我们验证过很多遍，因此我们还烧坏了很多板子，在过程中我们只有一个密钥，并未生成其他密钥。

深圳市广联智通科技有限公司
钟尚文

手机： +86 13973619466
网址：http://www.gl-inet.com
地址：深圳市宝安区石岩街道松白路创维数字大厦305、306室

 
发件人： Alvin1Zhang
发送时间： 2021-04-23 17:54
收件人： espressif/esp-idf
抄送： GL.iNet-ShangWen.Zhong; Mention
主题： Re: [espressif/esp-idf] Secure boot 在不同分支，不同提交下的不兼容问题 (IDFGH-5046) (#6831)
@gl-zhongshangwen
感谢提供如此具体的问题报告。我们已经在 release/3.3 分支上复现了 espsecure.py 问题（即您提到的问题二），我们已经找到修复方案，该修复将很快被合入，后续会更新此问题的状态。
您提到 master branch 的问题 （即您提到的问题三 ），我们仍在努力复现。您能帮检查并验证下用于 app build 的 "secure_boot_signing_key.pem" 和用于构建 flash 中 bootloader.bin 的"secure_boot_signing_key.pem" 是否相同？
谢谢。

Alvin1Zhang, 2021-05-07:
@gl-zhongshangwen 

感谢提供更多信息。按照您所提供的信息环境，我们用 Make 构建系统构建二进制文件

* bootloader 基于 commit bf0220609
* app 基于 master commit = 8bc19ba8

bf0220609 的 boot loader 可以正常运行 master branch (8bc19ba8) 上的 app，log 如下。

（我们用的是 BOOTLOADER_REFLASHABLE=y，而不是 BOOTLOADER_ONE_TIME_FLASH。）

ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:9336
load:0x40078000,len:18432
ho 0 tail 12 room 4
load:0x40080400,len:6664
entry 0x40080774
I (101) boot: Chip Revision: 3
I (102) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (47) boot: ESP-IDF v3.3.2-323-gbf02206 2nd stage bootloader
I (47) boot: compile time 17:52:48
I (48) boot: Enabling RNG early entropy source...
I (53) boot: SPI Speed : 40MHz
I (57) boot: SPI Mode : DIO
I (61) boot: SPI Flash Size : 4MB
I (65) boot: Partition Table:
I (69) boot: ## Label Usage Type ST Offset Length
I (76) boot: 0 nvs WiFi data 01 02 0000b000 00004000
I (83) boot: 1 otadata OTA data 01 00 0000f000 00002000
I (91) boot: 2 efuse_em efuse 01 05 00011000 00002000
I (98) boot: 3 phy_init RF data 01 01 00013000 00001000
I (106) boot: 4 nvs_my WiFi data 01 02 00014000 00004000
I (113) boot: 5 nvs_key1 NVS keys 01 04 00018000 00001000
I (121) boot: 6 nvs_key2 NVS keys 01 04 00019000 00001000
I (128) boot: 7 nvs_key3 NVS keys 01 04 0001a000 00001000
I (136) boot: 8 ota_0 OTA app 00 10 00020000 00080000
I (144) boot: 9 ota_1 OTA app 00 11 000a0000 00080000
I (151) boot: End of partition table
I (156) boot_comm: chip revision: 3, min. application chip revision: 0
I (163) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x0783c ( 30780) map
I (182) esp_image: segment 1: paddr=0x00027864 vaddr=0x3ffb0000 size=0x02094 ( 8340) load
I (186) esp_image: segment 2: paddr=0x00029900 vaddr=0x40080000 size=0x00400 ( 1024) load
0x40080000: _WindowOverflow4 at /home/kostia/esp/esp-idf/components/freertos/xtensa_vectors.S:1779
I (190) esp_image: segment 3: paddr=0x00029d08 vaddr=0x40080400 size=0x06310 ( 25360) load
I (209) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x1e14c (123212) map
0x400d0020: _flash_cache_start at ??:?
I (252) esp_image: segment 5: paddr=0x0004e174 vaddr=0x40086710 size=0x03be8 ( 15336) load
0x40086710: prvSwitchTimerLists at /home/kostia/esp/esp-idf/components/freertos/timers.c:867
I (259) esp_image: segment 6: paddr=0x00051d64 vaddr=0x00000000 size=0x0e21c ( 57884) 
I (280) esp_image: Verifying image signature...
I (704) boot: Loaded app from partition at offset 0x20000
I (704) secure_boot: bootloader secure boot is already enabled. No need to generate digest. continuing..
I (709) boot: Checking secure boot...
I (713) secure_boot: bootloader secure boot is already enabled, continuing..
I (721) boot: Disabling RNG early entropy source...
I (727) cpu_start: cpu freq: 160
I (731) cpu_start: Pro cpu up.
I (734) cpu_start: Starting app cpu, entry point is 0x40080f0c
0x40080f0c: start_cpu0_default at /home/kostia/esp/esp-idf/components/esp32/cpu_start.c:398
I (0) cpu_start: App cpu up.
I (756) cpu_start: Pro cpu start user code
I (756) cpu_start: Application information:
I (756) cpu_start: Project name: hello-world
I (761) cpu_start: App version: 1
I (766) cpu_start: Compile time: Apr 28 2021 17:09:47
I (772) cpu_start: ELF file SHA256: ee2ac4cb973981c1...
I (778) cpu_start: ESP-IDF: v4.3-dev-1197-g8bc19ba
I (784) heap_init: Initializing. RAM available for dynamic allocation:
I (791) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (798) heap_init: At 3FFB2900 len 0002D700 (181 KiB): DRAM
I (804) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (810) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (817) heap_init: At 4008A2F8 len 00015D08 (87 KiB): IRAM
I (824) spi_flash: detected chip: generic
I (827) spi_flash: flash io: dio
I (832) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello world!

@gl-zhongshangwen **您能帮助提供其他信息以便我们能够复现么？** 谢谢。

Alvin1Zhang, 2021-05-11:
@gl-zhongshangwen 您好，能帮助提供更多信息以便我们能够复现么？谢谢。

gl-zhongshangwen, 2021-05-11:
@Alvin1Zhang 您好，通过将配置改成BOOTLOADER_REFLASHABLE=y，刚刚复现确实可以在如下环境正常启动，但是为什么配置成BOOTLOADER_ONE_TIME_FLASH时不能使用呢？
你们的编程指南也没有说明开启后不能在不同的SDK版本下用，并且还建议我们在正式生产的时候使用One-Time Flash。
`https://docs.espressif.com/projects/esp-idf/en/release-v3.3/security/secure-boot.html#secure-boot-reflashable`
[image]

- bootloader 基于 commit bf02206
- app 基于 master commit 8bc19ba

Alvin1Zhang, 2021-05-27:
@gl-zhongshangwen 感谢反馈。

配置 BOOTLOADER_REFLASHABLE=y 并不会改变 bootloader 的二进制文件，它会触发 ESP-IDF 构建系统预先计算出一个 eFuse 中的安全启动密钥的值 (通过 "espefuse.py burn_efuse" 命令)，以及预生成bootloader 摘要文件。bootloader 摘要文件和 app 签名没有任何关系，ESP32 ROM 只在启动加载程序前检查该摘要文件。

因此，我们不希望对由于此配置项引起的错误信息 "esp_image: image valid, signature bad" 及 "No bootable app partitions in the partition table" 做任何改动。这些错误信息与 app 数字签名相关，与 bootloader 的摘要无关。

如果 eFuse 中的安全启动密钥已经被烧录，就算将 bootloader 二进制文件的 BOOTLOADER_ONE_TIME_FLASH 改为 BOOTLOADER_REFLASHABLE 也不会有任何作用，因为原始的密钥值已经被烧入 efuse。

为此，我们的工程师用 BOOTLOADER_ONE_TIME_FLASH 和相同的 bootloader/app IDF 版本进行复测，得到了预料的输出，如下：

```
rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8788
load:0x40078000,len:18392
load:0x40080400,len:5904
entry 0x40080734
I (82) boot: Chip Revision: 0
I (32) boot: ESP-IDF v3.3.2-323-gbf02206-dirty 2nd stage bootloader
I (33) boot: compile time 20:35:14
I (62) boot: Enabling RNG early entropy source...
I (63) boot: SPI Speed      : 40MHz
I (63) boot: SPI Mode       : DIO
I (64) boot: SPI Flash Size : 4MB
I (66) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 0000d000 00006000
I (75) boot:  1 phy_init         RF data          01 01 00013000 00001000
I (79) boot:  2 factory          factory app      00 00 00020000 00100000
I (82) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x05a7c ( 23164) map
I (97) esp_image: segment 1: paddr=0x00025aa4 vaddr=0x3ffb0000 size=0x0169c (  5788) load
I (100) esp_image: segment 2: paddr=0x00027148 vaddr=0x40080000 size=0x00400 (  1024) load
I (101) esp_image: segment 3: paddr=0x00027550 vaddr=0x40080400 size=0x07260 ( 29280) load
I (117) esp_image: segment 4: paddr=0x0002e7b8 vaddr=0x00000000 size=0x01858 (  6232) 
I (120) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x12734 ( 75572) map
I (147) esp_image: segment 6: paddr=0x00042754 vaddr=0x00000000 size=0x0d82c ( 55340) 
I (167) esp_image: Verifying image signature...
I (565) boot: Loaded app from partition at offset 0x20000
I (566) esp_image: segment 0: paddr=0x00001020 vaddr=0x3fff0018 size=0x00004 (     4) 
I (567) esp_image: segment 1: paddr=0x0000102c vaddr=0x3fff001c size=0x02254 (  8788) 
I (575) esp_image: segment 2: paddr=0x00003288 vaddr=0x40078000 size=0x047d8 ( 18392) 
I (582) esp_image: segment 3: paddr=0x00007a68 vaddr=0x40080400 size=0x01710 (  5904) 
I (585) secure_boot: Generating new secure boot key...
I (591) secure_boot: Generating secure boot digest...
I (648) secure_boot: Digest generation complete.
I (648) boot: Checking secure boot...
I (648) secure_boot: Read & write protecting new key...
I (656) secure_boot: blowing secure boot efuse...
I (656) secure_boot: Disable JTAG...
I (656) secure_boot: Disable ROM BASIC interpreter fallback...
I (665) secure_boot: secure boot is now enabled for bootloader image
I (665) boot: Disabling RNG early entropy source...
I (667) cpu_start: Pro cpu up.
I (668) cpu_start: Application information:
I (671) cpu_start: Project name:     hello-world
I (673) cpu_start: App version:      v3.3.2-323-gbf02206-dirty
I (677) cpu_start: Compile time:     May 19 2021 20:36:31
I (680) cpu_start: ELF file SHA256:  5c480352214f39d7...
I (683) cpu_start: ESP-IDF:          v3.3.2-323-gbf02206-dirty
I (686) cpu_start: Single core mode
```

Alvin1Zhang, 2021-06-02:
@gl-zhongshangwen 您好，请问这个问题有更新么？谢谢。

gl-zhongshangwen, 2021-06-02:
Hi Alvin1Zhang,
         我可以理解为用BOOTLOADER_ONE_TIME_FLASH方式烧入固件到ESP32后，下一次更新固件就只能用同一SDK生成的固件成功升级了，是这样吗？（在用同一加密密钥文件的情况下，更加新的分支IDF SDK生成的固件将无法使用？）

深圳市广联智通科技有限公司
钟尚文

手机： +86 13973619466
网址：http://www.gl-inet.com
地址：深圳市宝安区石岩街道松白路创维数字大厦305、306室

 
发件人： Alvin1Zhang
发送时间： 2021-05-27 13:50
收件人： espressif/esp-idf
抄送： GL.iNet-ShangWen.Zhong; Mention
主题： Re: [espressif/esp-idf] Secure boot 在不同分支，不同提交下的不兼容问题 (IDFGH-5046) (#6831)
@gl-zhongshangwen 感谢反馈。
配置 BOOTLOADER_REFLASHABLE=y 并不会改变 bootloader 的二进制文件，它会触发 ESP-IDF 构建系统预先计算出一个 eFuse 中的安全启动密钥的值 (通过 "espefuse.py burn_efuse" 命令)，以及预生成bootloader 摘要文件。bootloader 摘要文件和 app 签名没有任何关系，ESP32 ROM 只在启动加载程序前检查该摘要文件。
因此，我们不希望对由于此配置项引起的错误信息 "esp_image: image valid, signature bad" 及 "No bootable app partitions in the partition table" 做任何改动。这些错误信息与 app 数字签名相关，与 bootloader 的摘要无关。
如果 eFuse 中的安全启动密钥已经被烧录，就算将 bootloader 二进制文件的 BOOTLOADER_ONE_TIME_FLASH 改为 BOOTLOADER_REFLASHABLE 也不会有任何作用，因为原始的密钥值已经被烧入 efuse。
为此，我们的工程师用 BOOTLOADER_ONE_TIME_FLASH 和相同的 bootloader/app IDF 版本进行复测，得到了预料的输出，如下：
rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8788
load:0x40078000,len:18392
load:0x40080400,len:5904
entry 0x40080734
I (82) boot: Chip Revision: 0
I (32) boot: ESP-IDF v3.3.2-323-gbf02206-dirty 2nd stage bootloader
I (33) boot: compile time 20:35:14
I (62) boot: Enabling RNG early entropy source...
I (63) boot: SPI Speed      : 40MHz
I (63) boot: SPI Mode       : DIO
I (64) boot: SPI Flash Size : 4MB
I (66) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 0000d000 00006000
I (75) boot:  1 phy_init         RF data          01 01 00013000 00001000
I (79) boot:  2 factory          factory app      00 00 00020000 00100000
I (82) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x05a7c ( 23164) map
I (97) esp_image: segment 1: paddr=0x00025aa4 vaddr=0x3ffb0000 size=0x0169c (  5788) load
I (100) esp_image: segment 2: paddr=0x00027148 vaddr=0x40080000 size=0x00400 (  1024) load
I (101) esp_image: segment 3: paddr=0x00027550 vaddr=0x40080400 size=0x07260 ( 29280) load
I (117) esp_image: segment 4: paddr=0x0002e7b8 vaddr=0x00000000 size=0x01858 (  6232) 
I (120) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x12734 ( 75572) map
I (147) esp_image: segment 6: paddr=0x00042754 vaddr=0x00000000 size=0x0d82c ( 55340) 
I (167) esp_image: Verifying image signature...
I (565) boot: Loaded app from partition at offset 0x20000
I (566) esp_image: segment 0: paddr=0x00001020 vaddr=0x3fff0018 size=0x00004 (     4) 
I (567) esp_image: segment 1: paddr=0x0000102c vaddr=0x3fff001c size=0x02254 (  8788) 
I (575) esp_image: segment 2: paddr=0x00003288 vaddr=0x40078000 size=0x047d8 ( 18392) 
I (582) esp_image: segment 3: paddr=0x00007a68 vaddr=0x40080400 size=0x01710 (  5904) 
I (585) secure_boot: Generating new secure boot key...
I (591) secure_boot: Generating secure boot digest...
I (648) secure_boot: Digest generation complete.
I (648) boot: Checking secure boot...
I (648) secure_boot: Read & write protecting new key...
I (656) secure_boot: blowing secure boot efuse...
I (656) secure_boot: Disable JTAG...
I (656) secure_boot: Disable ROM BASIC interpreter fallback...
I (665) secure_boot: secure boot is now enabled for bootloader image
I (665) boot: Disabling RNG early entropy source...
I (667) cpu_start: Pro cpu up.
I (668) cpu_start: Application information:
I (671) cpu_start: Project name:     hello-world
I (673) cpu_start: App version:      v3.3.2-323-gbf02206-dirty
I (677) cpu_start: Compile time:     May 19 2021 20:36:31
I (680) cpu_start: ELF file SHA256:  5c480352214f39d7...
I (683) cpu_start: ESP-IDF:          v3.3.2-323-gbf02206-dirty
I (686) cpu_start: Single core mode

gl-zhongshangwen, 2021-06-02:
Hi Alvin1Zhang,
    刚刚已经回复您。

深圳市广联智通科技有限公司
钟尚文

手机： +86 13973619466
网址：http://www.gl-inet.com
地址：深圳市宝安区石岩街道松白路创维数字大厦305、306室

 
发件人： Alvin1Zhang
发送时间： 2021-06-02 10:32
收件人： espressif/esp-idf
抄送： GL.iNet-ShangWen.Zhong; Mention
主题： Re: [espressif/esp-idf] Secure boot 在不同分支，不同提交下的不兼容问题 (IDFGH-5046) (#6831)
@gl-zhongshangwen 您好，请问这个问题有更新么？谢谢。

Alvin1Zhang, 2021-06-08:
使用 BOOTLOADER_ONE_TIME_FLASH，也能够将 ESP-IDF 更新到更新的版本，并更新至新的 app binary。 必须使用相同的签名密钥对新的二进制文件进行签名，以匹配原始安装的签名密钥。 更新 bootloader 需要有 eFuse 中 secret Secure Boot 密钥的副本（与签名密钥不同），如果您有这个副本，那么您也可以通过 serial 更新 bootloader。

Alvin1Zhang, 2021-07-01:
@gl-zhongshangwen 您好，请问该问题有更新么？您的问题是否已经解决？谢谢。

gl-zhongshangwen, 2021-07-01:
你好，Alvin
    我们测试了，还是无法启动，目前我们这边有10几台ESP32已经边砖了
    我在验证过程中用的
        v3.3 bootloader + v3.3 app 刷入到ESP32，发现可以正常启动，这个时候我再用同样的密钥编译了 v4.0 app 烧入到这个ESP32中，发现就无法启动了，其中两个工程密钥一致，分区表一致，烧入方式一致，都使用的BOOTLOADER_ONE_TIME_FLASH。

深圳市广联智通科技有限公司
钟尚文

手机： +86 13973619466
网址：http://www.gl-inet.com
地址：深圳市宝安区石岩街道松白路创维数字大厦305、306室

 
发件人： Alvin1Zhang
发送时间： 2021-06-08 14:27
收件人： espressif/esp-idf
抄送： GL.iNet-ShangWen.Zhong; Mention
主题： Re: [espressif/esp-idf] Secure boot 在不同分支，不同提交下的不兼容问题 (IDFGH-5046) (#6831)
使用 BOOTLOADER_ONE_TIME_FLASH，也能够将 ESP-IDF 更新到更新的版本，并更新至新的 app binary。 必须使用相同的签名密钥对新的二进制文件进行签名，以匹配原始安装的签名密钥。 更新 bootloader 需要有 eFuse 中 secret Secure Boot 密钥的副本（与签名密钥不同），如果您有这个副本，那么您也可以通过 serial 更新 bootloader。

Alvin1Zhang, 2021-09-22:
@gl-zhongshangwen 感谢反馈，您能再尝试使用最新的 release/4.0 分支看下问题是否存在么？谢谢。

closed 2021-10-08 by Alvin1Zhang