espressif/esp-idf #6832: usleep is not defined (kinda)

DiegoActionItem, 2021-04-06:
I am trying to port my code into more stanard C code. I am changing all `vTaskDelay` into `usleep()` . The function is available in newlib: https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/newlib/time.c#L206

However - the toolchain is guarding it (`~/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/xtensa-esp32s2-elf/include/sys/unistd.h` on my setup, this is part of the gcc port). This is not available to applications, in a direct mode anyway:
``` C
#if __XSI_VISIBLE >= 500 && __POSIX_VISIBLE < 200809 || __BSD_VISIBLE
int 	usleep (useconds_t __useconds);
#endif
```

As a workaround, in my code I did define that function as external and it does seem to work:
``` C
extern int usleep(useconds_t us);
```

 But IMHO - this is a toolchain bug.

igrr, 2021-04-06:
Hi @DiegoActionItem, `__BSD_VISIBLE` should be defined in IDF by default, so `usleep` is normally visible. I have tried to build the blink example with the following modification, and it worked:
```diff
diff --git a/examples/get-started/blink/main/blink.c b/examples/get-started/blink/main/blink.c
index 94f48b2d1b..8492279b5a 100644
--- a/examples/get-started/blink/main/blink.c
+++ b/examples/get-started/blink/main/blink.c
@@ -7,6 +7,7 @@
    CONDITIONS OF ANY KIND, either express or implied.
 */
 #include <stdio.h>
+#include <unistd.h>
 #include "freertos/FreeRTOS.h"
 #include "freertos/task.h"
 #include "driver/gpio.h"
@@ -78,6 +79,6 @@ void app_main(void)
         blink_led();
         /* Toggle the LED state */
         s_led_state = !s_led_state;
-        vTaskDelay(CONFIG_BLINK_PERIOD / portTICK_PERIOD_MS);
+        usleep(CONFIG_BLINK_PERIOD * 1000);
     }
 }
```
Can you give some example where `usleep` doesn't resolved? Either the project you are working on, or some reduced version of it which still has this problem.

DiegoActionItem, 2021-04-06:
We do this in a C++ project. If you rename this file into CPP (and fix compilation issues wit GPIO) you will get this file, which does not compile:

``` C++
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#define BLINK_GPIO (gpio_num_t) CONFIG_BLINK_GPIO

extern "C" void app_main(void)
{
    gpio_pad_select_gpio(BLINK_GPIO);
    gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);
    while(1) {
	printf("Turning off the LED\n");
        gpio_set_level( (gpio_num_t) BLINK_GPIO, 0);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
	printf("Turning on the LED\n");
        gpio_set_level(BLINK_GPIO, 1);
        // here be dragons
	usleep(1000);
    }
}
```

I get:
```
✔ blink [v4.2-rc|✚ 2…1]$ make
Toolchain path: /home/diego/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2020r3
Compiler version: 8.4.0
Python requirements from /home/diego/src/esp-idf-4.2/requirements.txt are satisfied.

App "blink" version: v4.2-rc-dirty
CXX build/main/blink.o
/home/diego/src/esp-idf-4.2/examples/get-started/blink/main/blink.cpp: In function 'void app_main()':
/home/diego/src/esp-idf-4.2/examples/get-started/blink/main/blink.cpp:40:2: error: 'usleep' was not declared in this scope
  usleep(1000);
  ^~~~~~
/home/diego/src/esp-idf-4.2/examples/get-started/blink/main/blink.cpp:40:2: note: suggested alternative: 'fseek'
  usleep(1000);
  ^~~~~~
  fseek
make[1]: *** [/home/diego/src/esp-idf-4.2/make/component_wrapper.mk:292: blink.o] Error 1
make: *** [/home/diego/src/esp-idf-4.2/make/project.mk:635: component-main-build] Error 2
```

igrr, 2021-04-06:
After adding `#include <unistd.h>`, the code snippet you have posted compiles okay with both `make` and  `idf.py build`. Could you try adding this header, please?

DiegoActionItem, 2021-04-07:
I am closing this issue as... I am no longer able to reproduce this. I had a compilation bug ... and traced it, and had a fix. Now even my original issue no longer exists. I am unsure what caused the issue. :(

closed 2021-04-07 by DiegoActionItem