espressif/esp-idf #6833: Firmware upgrade failed

simeondmr, 2021-04-06:
Hello,

I'm testing the simple_ota_example.
I have followed every step in the documentation but I receiver alway these errors:

I (6696) simple_ota_example: Starting OTA example
E (6756) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (6756) esp-tls-mbedtls: Failed to verify peer certificate!
I (6756) esp-tls-mbedtls: verification info:   ! The certificate Common Name (CN) does not match with the expected CN
  ! The certificate is not
E (6776) esp-tls: Failed to open new connection
E (6776) TRANS_SSL: Failed to open a new connection
E (6786) HTTP_CLIENT: Connection failed, sock < 0
E (6786) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (6796) esp_https_ota: Failed to establish HTTP connection
E (6806) simple_ota_example: Firmware upgrade failed

I have cheched also the https server, but seems to be ok.

Please could you help me with this problem?

Thanks

Simeon

ESP-Marius, 2021-04-07:
Hi,

The HTTPS client will by default check that the certificate's common name match the address you are connecting, and this check is failing in your example.

What are you trying to connect to? For testing purposes you can by-pass this check by enabling `EXAMPLE_SKIP_COMMON_NAME_CHECK` in menuconfig, but this could make you your connection less secure and should not be used for production purposes without considering the risk first.

simeondmr, 2021-04-07:
Hello,

thanks for your reply!
I'm trying to connect to an HTTPS server: openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070
I have enabled EXAMPLE_SKIP_COMMON_NAME_CHECK, in menuconfig but I have always the same problem.

ESP-Marius, 2021-04-08:
Hmm, that's strange. You shouldn't be seeing that if you have disabled the check.

Can you please double check that it is active? E.g. add a print next to the line **config.skip_cert_common_name_check = true;** in the simple OTA example, to be 100% sure it is configured, built and flashed correctly?

simeondmr, 2021-04-08:
Hello,

I have added a printf next to the line  config.skip_cert_common_name_check = true:

#ifdef CONFIG_EXAMPLE_SKIP_COMMON_NAME_CHECK
    config.skip_cert_common_name_check = true;
    printf("SKIP\n");
#endif

The output is:

I (4196) simple_ota_example: Starting OTA example
SKIP
E (4256) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (4256) esp-tls-mbedtls: Failed to verify peer certificate!
I (4256) esp-tls-mbedtls: verification info:   ! The certificate is not correctly signed by the trusted CA

E (4266) esp-tls: Failed to open new connection
E (4276) TRANS_SSL: Failed to open a new connection
E (4286) HTTP_CLIENT: Connection failed, sock < 0
E (4286) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (4296) esp_https_ota: Failed to establish HTTP connection
E (4296) simple_ota_example: Firmware upgrade failed

Skip is printed so the firmware is flashed correctly.
If I go in the local browser: https://192.168.1.8:8070/hello-world.bin, it work correctly...

ESP-Marius, 2021-04-08:
Hmm, that's strange. I just tested the same way you did it and it works fine with the default certificates in the example at least (server cert and key copied from example_test.py). With **CONFIG_EXAMPLE_SKIP_COMMON_NAME_CHECK** disabled it fails with **verification info:   ! The certificate Common Name (CN) does not match with the expected CN** as expected, but with it active everything is OK. 

- Which version/commit of IDF are you running?
- Are you using the example certificates or are you generating your own?

simeondmr, 2021-04-08:
I think that it's a certificate problem.
If I use the certificted copied from example_test.py and with CONFIG_EXAMPLE_SKIP_COMMON_NAME_CHECK disabled it work correctly but with CONFIG_EXAMPLE_SKIP_COMMON_NAME_CHECK enabled it doesn't work.

To create my certificate I run this command: openssl req -x509 -newkey rsa:2048 -keyout ca_key.pem -out ca_cert.pem -days 365 -nodes

IDF version: 4.2

ESP-Marius, 2021-04-09:
Ah, I see now. In the second log you posted the issue is no longer the CN; **esp-tls-mbedtls: verification info: ! The certificate is not correctly signed by the trusted CA**

So now it correctly skips the common name check, but the certificate itself is not correct.

I tested again the same way you did it and it seems to work.

Did you by any chance forget to update the certificate flashed to the ESP? It should be the **ca_cert.pem** you generated and used for the server. Replaced **examples/system/ota/simple_ota_example/server_certs/ca_cert.pem** with this new **ca_cert.pem**, rebuild and reflash and you should be able to connnect.

simeondmr, 2021-04-09:
Hello,

sorry..I forgot to update the certificate flashed to the ESP.
Thanks a lot for your time.

Simeon

ESP-Marius, 2021-04-09:
No problem, closing this issue then.

closed 2021-04-09 by ESP-Marius