espressif/esp-idf #6836: v4.3 vs v4.2: reset reason after esp_restart() is ESP_RST_TASK_WDT instead of ESP_RST_SW

boborjan2, 2021-04-06:
## Environment
- Module or chip used:  [ESP32-WROOM-32E]
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-3175-g9a2d25191
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [external]

## Problem Description
With the very same software, I experience incorrect reset reason after a reboot initiated by esp_restart().

### Expected Behavior
Until now reset reason has been ESP_RST_SW (coherently to the docs).

### Actual Behavior
I get ESP_RST_TASK_WDT.

### Steps to reproduce
1. esp_restart()
2. after reboot, read reset reason via esp_reset_reason().

Thanks,
Viktor

ESP-Marius, 2021-04-07:
Hi @boborjan2 , 

Thanks for reporting this issue. I havent been able to reproduce this using the commit ID provided though. Any more information you could provide to help me would be great:

-  What is the reset reason reported by the ROM bootloader? (Should be the first line printed at startup)
- Does the same thing happen if you try to do `esp_restart` in e.g. the hello world example? Or is it something that only happens in your full application?
- Any change you could provide the sdkconfig file?

boborjan2, 2021-04-07:
Hi @ESP-Marius,

Thanks for the answer. 
- rom bootloader reports "rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)"
- I haven't tried hello world. I experience this with an application that has been in use for a long time, and this restart is a frequent use-case with no reported issues.
- I upload the sdkconfig. Please find it attached to the ticket.

Other: here is the code that performs the restart:
```
ESP_LOGW(TAG, "going to sleep");
	mdns_free();
	esp_wifi_stop();
	esp_wifi_deinit();
	esp_vfs_spiffs_unregister(NULL);

ESP_LOGW(TAG, "calling esp_restart");

	esp_restart();
```
Here is the printout on the console. esp_restart() is indeed called (I thought perhaps it is delayed and the task watchdog jumps in):
```
W (3679) main: going to sleep
W (3679) wifi:<ba-del>idx
W (3679) wifi:<ba-del>idx
W (3679) wifi:<ba-del>idx
W (3679) wifi:hmac tx: ifx0 stop, discard
W (3679) wifi: WIFI_EVENT_STA_DISCONNECTED
W (3679) wifi: WIFI_EVENT_STA_STOP
W (3719) main: calling esp_restart
```
Interestingly enough, with the printouts added,  ROM bootloader reports the same reset reason, but application reports it to 7 (ESP_RST_WDT instead of the previouly observed ESP_RST_TASK_WDT)

Regards,
Viktor

boborjan2, 2021-04-07:
I paste sdkconfig here:
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_SDK_PYTHON="python"
# CONFIG_SDK_MAKE_WARN_UNDEFINED_VARIABLES is not set
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Bootloader config
#
CONFIG_BOOTLOADER_OFFSET_IN_FLASH=0x1000
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_SPI_CUSTOM_WP_PIN is not set
CONFIG_BOOTLOADER_SPI_WP_PIN=7
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=y
CONFIG_BOOTLOADER_NUM_PIN_FACTORY_RESET=23
CONFIG_BOOTLOADER_OTA_DATA_ERASE=y
CONFIG_BOOTLOADER_DATA_FACTORY_RESET="nvs,phy_init,storage"
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_HOLD_TIME_GPIO=5
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ON_POWER_ON is not set
# CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
# end of Bootloader config

#
# Security features
#
# CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT is not set
# CONFIG_SECURE_BOOT is not set
[... 985 lines cut ...]
# CONFIG_ESP_TCP_KEEP_CONNECTION_WHEN_IP_CHANGES is not set
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
# CONFIG_SUPPORT_TERMIOS is not set
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
# End of deprecated options

```

ESP-Marius, 2021-04-08:
I can't see anything obvious that could be causing this issue.

It's strange that the bootloader reason doesn't match the reason you read out as they both should call **rtc_get_reset_reason** and give the same result, so **ESP_RST_WDT** is the expected value here when the bootloader reports **RTCWDT_RTC_RESET**

As for why you get **RTCWDT_RTC_RESET**: we activate the RTC WDT when we start **esp_restart_noos**, so if the software somehow got stuck here without being able to reset the CPUs, then it would trigger a **RTCWDT_RTC_RESET** instead of the **ESP_RST_SW**.

Still not able to reproduce it though.

Some suggestions:

- Attach the full log if possible (including the first run of the software before the reset, so I can see the whole process from running->reset->bootloader->running again)
- Try to disable the RTC WDT in **esp_restart_noos** (see [patch](https://github.com/espressif/esp-idf/files/6277297/0001-disable-RTC-WDT-in-restart.txt)). If my suspicion is correct then you should either get stuck forever in **esp_restart_noos**  somewhere, or get a **ESP_RST_SW** after a while. If any of this happens then you could try to debug and figure out where/why it gets stuck. (JTAG would be helpful here)

Thanks for being patient with providing feedback to help troubleshoot this!

boborjan2, 2021-04-08:
Disabling RTC WDT does indeed prevent restart. I have no JTAG but with GPIO toggling I found that execution reaches the call of rtc_clk_cpu_freq_set_xtal() in esp_restart_noos() but it never reaches rtc_clk_cpu_freq_set_xtal itself.
I added IRAM_ATTR to rtc_clk_cpu_freq_set_xtal() and it did the trick: I could toggle gpio in rtc_clk_cpu_freq_set_xtal() as well.
My guess is that this is some linker file mess-up. I wonder why you don't see it (maybe you have it cached in by chance?). Do you use make or cmake?
Here is a correlating issue although it's supposedly solved: https://github.com/espressif/esp-idf/issues/6297

ESP-Marius, 2021-04-09:
Yup that explains it, cache is disabled at that point and I was building with cmake...

Like you say, this is already solved (wasn't aware of the issue you linked), but the backport to v4.3 hadn't been made yet.

Sorry for the inconvenience, I'll make sure that it's backported to 4.3 today.

ESP-Marius, 2021-04-09:
Merged to release/v4.3 as 904933745b2b00ebb9fb0a3b8aec83a7ebd4205e, should be available on github with the next sync.

boborjan2, 2021-04-09:
I confirm cherry-picking 2eb563406ffdfeee372e0f4d21d38e86ea75b913 does solve the issue. Thanks for the support. More to this, it also solves this one: https://github.com/espressif/esp-idf/issues/6835. It seems RTC watchdog reset does not reset gpio state to default. This info should be propagated in docs btw.

There is another one issue I have correlating with IRAM placement in v4.3: https://github.com/espressif/esp-idf/issues/6825. This also does not happen in 4.2. This one is still there.

closed 2021-04-09 by boborjan2