espressif/esp-idf #6838: spi_master/lcd example uses double precision

zx64, 2021-04-07:
## Environment

- Development Kit:     [ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-beta2-2-g9a2d25191
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

While measuring the speed of the `spi_master/lcd` example, I noticed some calls to `sin` in the warping effect:
https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/examples/peripherals/spi_master/lcd/main/pretty_effect.c#L54
Changing those to call `sinf` instead reduced the time spent at the start of each frame from about 14ms to 3ms (when built with `-O2`/`NDEBUG`).

Additionally, marking the multiplication constants as floats takes things down to 2ms, e.g:
`        for (int x=0; x<320; x++) xofs[x]=sinf(frame*0.15f+x*0.06f)*4.0f;`

Timing was measured by toggling a GPIO at the start of each frame and before/after calls to `pretty_effect_calc_lines`.
You can see the 14ms at the start of each frame:
[image]
After making those changes, I had to zoom in to read the timing of that first-per-frame call:
[image]

KaeLL, 2021-04-07:
I mean, it's not flash news that `float` is faster than `double` due to the former's operations being performed in hardware, unlike the latter's. It warrants a note in the docs, but it's already [there](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/freertos-smp.html#floating-point-arithmetic). Does is justify changing the code of the example? I'm kind of over the fence on this.

zx64, 2021-04-07:
It's adding the letter 'f' (and some .0s) to four lines of code.

KaeLL, 2021-04-07:
Fair enough.