espressif/esp-idf #6839: menuconfig doesn't provide tinyusb HID feature at all

oswaldtoma, 2021-04-07:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [V ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [V ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ V] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit: ESP32-S2-Saola-1
- Module or chip used:  ESP32-S2WROVER
- IDF version: master
- Build System: idf.py
- Compiler version: xtensa-esp32s2-elf-gcc
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: Yes (Eclipse with ESP-IDF plugin)
- Power Supply: USB

## Problem Description

The menuconfig accessible from idf.py script, doesn't provide us with an option to turn on HID.

### Expected Behavior

There should be.

### Actual Behavior

There's not.

### Steps to reproduce

Run idf.py menuconfig ¯\_(ツ)_/¯

dobairoland, 2021-04-08:
Hi @oswaldtoma. This is under development right now and will be available in the future. That is why you cannot find it in the menuconfig.

oswaldtoma, 2021-04-08:
When can it be expected?

abdulwasey7, 2021-07-12:
any updates regarding this feature?

leeebo, 2021-07-29:
@oswaldtoma @abdulwasey7 You can temporary have a try using: https://github.com/iot-components/examples_tinyusb_esp32sx

AEFeinstein, 2021-11-08:
This weekend I forked the IDF and added an HID option to menuconfig: https://github.com/AEFeinstein/esp-idf/commit/64f4918db0a687e093ad841704f8c9cb6a98db87. Specifically I was working to get an ESP32-S2 to enumerate as a gamepad. My IDF changes are nowhere near ready to be pulled into the main one, but they're relatively simple and they do work on a ESP32-S2-DEVKITC-1R. At a minimum it still needs a way to configure keyboard, mouse, or gamepad via menuconfig. Two callbacks need to be defined in the application code too, which isn't the best design.

@dobairoland, I'm curious how similar your development is to what I did. Is it in a public branch?

AachoLoya, 2022-07-05:
I also need to create USB HID keyboard/mouse device using ESP32-S2. Any progress on this front?

AEFeinstein, 2022-07-06:
A tiny bit (pun intended). I learned that components in your ESP32 project will override components from the IDF if they have the same name. So I copied the `tinyusb` component into my `components` folder, made a few tweaks (see my prior comment), and got my USB gamepad working nicely without modifications to the IDF.

See here: https://github.com/AEFeinstein/esp32-c3-playground/tree/main/components/tinyusb

AachoLoya, 2022-07-06:
So you copied tinyusb as-is from the original git repo into the components folder? The code in additions folder, is that the IDF changes or something specific to your project?

AEFeinstein, 2022-07-06:
> So you copied tinyusb as-is from the original git repo into the components folder?

Yep

> The code in additions folder, is that the IDF changes or something specific to your project?

That came from the IDF. The changes I made are in a commit which I linked to in my first comment.

AEFeinstein, 2022-07-15:
It's in https://github.com/AEFeinstein/esp32-c3-playground, though I haven't detailed what hardware you need to run it. Just search the code for `tusb` for now.

AEFeinstein, 2022-07-15:
Also the project name is a misnomer. I've been working with an S2. I know your asked about S3, but they should be similar enough

closed 2022-07-29 by espressif-bot