espressif/esp-idf #6840: esp_http_client basic auth no comparing scheme in case-insensitive fashion

hennejg, 2021-04-07:
## Environment

- Development Kit:      none
- Module or chip used:  does not apply
- IDF version (run ``git describe --tags`` to find it): v4.4
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): not relevant
- Operating System:     not relevant
- (Windows only) environment type: not relevant
- Using an IDE?: not relevant
- Power Supply:         not relevant

## Problem Description

HTTP basic auth does not work with all servers with esp_http_client. This is due to the authentication-scheme in the `WWW-Authenticate`-header being compared in a case-sensitive fashion against the string `Basic` in `esp_http_client_add_auth()`. Most servers use the capitalization of `Basic` given in the RFC example. However, some don't and use e.g. `BASIC` or `basic` instead. And they are rightly doing som because [RFC 2617 Secion 1.2](https://tools.ietf.org/html/rfc2617#section-2) clearly states "It uses an extensible, /case-insensitive/ token to identify the authentication scheme,..." (emphasis mine).

### Expected Behavior

Basic auth working with servers using a capitalization other than `Basic`.

### Actual Behavior

Basic auth only works with servers using `Basic`.

### Steps to reproduce

1. find a server using a `WWW-Authenticate`-header with a scheme capitalization other than `Basic`.
2. try http with basic auth
3. see it fail

### Code to reproduce this issue

Any sample http client code with authentication will do.

## Debug Logs

```
[  9512][D][test.cpp] debugHTTPEventHandler(): HTTP_EVENT_ON_HEADER, key=WWW-Authenticate, value=BASIC realm="my realm"
[ 16670][E][HTTP_CLIENT]: This authentication method is not supported: BASIC realm="my realm"
```

Alvin1Zhang, 2021-04-08:
Thanks for reporting and contributing the PR, we will look into soon.

hennejg, 2021-04-08:
Thanks for the quick response! Please note that there is probably another similar issue lurking in the code. [RFC 7617 Section 2](https://tools.ietf.org/html/rfc7617) says

> Note that both scheme and parameter names are matched case-insensitively.

In [the lines following my fix](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/esp_http_client/esp_http_client.c#L1441) several scheme parameters are matched case-sensitively as well. It would be easy to add a case-insensitive version of `http_utils_get_string_between` to address this issue. I did not have a server exhibiting this behaviour at hand. If you would like me to come up with a follow-up PR I could look into the issue, though.

closed 2021-04-15 by espressif-bot

AxelLin, 2022-08-16:
@Alvin1Zhang  @ESP-YJM 
Can someone apply this fix to v4.3? ( 19024671ea23d65a4df225d3f25811ec52a58182 can be cleanly applied on v4.3)