espressif/esp-idf #6842: Request: System Event Type Deauth

SillyHats, 2021-04-08:
**Is your feature request related to a problem? Please describe.**

When an ESP32 board connected to a mesh wifi system is deauthenticated by one router node after "Class 3 frame received from nonassociated station" there is no deauthed status similar to SYSTEM_EVENT_STA_CONNECTED and SYSTEM_EVENT_STA_DISCONNECTED to trigger the system to reconnect to the new mesh node as seen in [tiltbridge issue #171](https://github.com/thorrak/tiltbridge/issues/171)

**Describe the solution you'd like**

[Add a new system event type to esp_event_legacy.h](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/esp_event/include/esp_event_legacy.h#L29-L62)

/** System event types enumeration */
typedef enum {
    SYSTEM_EVENT_WIFI_READY = 0,           /*!< ESP32 WiFi ready */
    SYSTEM_EVENT_SCAN_DONE,                /*!< ESP32 finish scanning AP */
    SYSTEM_EVENT_STA_START,                /*!< ESP32 station start */
    SYSTEM_EVENT_STA_STOP,                 /*!< ESP32 station stop */
    SYSTEM_EVENT_STA_CONNECTED,            /*!< ESP32 station connected to AP */
    SYSTEM_EVENT_STA_DISCONNECTED,         /*!< ESP32 station disconnected from AP */

**Request--->     SYSTEM_EVENT_STA_DEAUTH,         /*!< ESP32 station deauthenticated from AP */**

**Describe alternatives you've considered**

1. Using SYSTEM_EVENT_STA_DISCONNECTED to trigger the reconnection, but this does not appear to work in this instance. 
2. Automatic ESP32 reboot when unable to connect to the network. This technically works, but results in an undesirable number of reboots. Automatic reboots happen as often as every 10 minutes based on where the board is relative to wifi nodes.

Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

N/A

Add any other context or screenshots about the feature request here.

negativekelvin, 2021-04-08:
>Using SYSTEM_EVENT_STA_DISCONNECTED to trigger the reconnection, but this does not appear to work in this instance

I think this is the right way to solve this, will post a comment on other issue

MaxwellAlan, 2022-03-08:
Hi @SillyHats 
Thanks for your feature request, but we can't support this feature.
- In WiFi connect stage, the Deauth is a temporary  state, will move to Connect or Disconnect.
- Using  SYSTEM_EVENT_STA_DISCONNECTED to trigger the reconnection is the right way, if fail, you can use esp_wifi_disconnect() and then esp_wifi_connect() to reconnect, do not need reboot.

jack0c, 2022-05-20:
@MaxwellAlan there may be some issues when trigger "Class 3 frame received from nonassociated station". @SillyHats which AP do you use?

SillyHats, 2022-05-25:
@jack0c RT-AC68U using the Asus AiMesh feature

xueyunfei998, 2022-06-07:
hi @SillyHats 

When there is a problem, can you provide the sniffer for this ap?

xueyunfei998, 2022-07-28:
[image]

xueyunfei998, 2022-07-28:
hi @SillyHats 

What is your reappearance environment like.

 Need two ac68u to build the environment, one is mesh router mode and the other is mesh mode?

Then, mesh mode is added to mesh router mode, and esp32 is connected to mesh mode?

I want to confirm whether it is this environment, I want to set up this environment locally, and I want to see which stage  received the third-level frame.

xueyunfei998, 2022-07-28:
The mode that my router is turned on now is: Wireless router mode / AiMesh Router mode (Default)

Esp32 didn't find any problems when connecting to the router.

Can you share your reappearance environment?

SillyHats, 2022-07-28:
@xueyunfei998

> What is your reappearance environment like.
>
> Need two ac68u to build the environment, one is mesh router mode and the other is mesh mode?
>
> Then, mesh mode is added to mesh router mode, and esp32 is [...]

One router is configured as wireless router (default), and the others are configured as AP/AI mesh AP and then connected to the main router via the main router web interface.

Each device, including the esp32 sees one AP with the configured name and the connection is managed between the client and routers as to which AP it connects to.

[image]
(first two are ethernet backhaul, last two shown here are wireless backhaul)

xueyunfei998, 2022-07-29:
hi @SillyHats 

When there is something wrong with the wifi connection, can you help me catch a packet? We only have one ac68u router here.

Assuming that the channel of the router is 1,

use a linux laptop,You can use it to sniff the channel 1.

ifconfig wifi. down
iwconfig wifi. mode monitor
ifconfig wifi. up
iwconfig wifi. channel 1
wireshark

xueyunfei998, 2022-08-05:
hi  @SillyHats 

In the process of configuring aimesh, I configured the main router as 
[image]

Configure the other router node as aimesh node. 
[image]

Then the main route adds mesh nodes in the web interface, and the following error occurs

[image]

xueyunfei998, 2022-08-05:
What is the firmware version of your router?

This is the firmware version of my router：
[image]

Now the group mesh has been having problems. I don't know why. Can you give me some help?

SillyHats, 2022-08-05:
Firmware Version:386.2_6

For drop setup have you tried this set of steps including the reset button on the back of the node router?
https://www.asus.com/us/support/FAQ/1035087/

xueyunfei998, 2022-08-11:
[image]

SillyHats, 2022-08-12:
That looks like it is setup correctly. If there are 3 routers in the mesh as it looks like you may want to connect one of them to the main router by ethernet from one of the numbered client ports on the main router to the uplink port on the mesh node.

How can I figure out which idf commit was used in version 1.0.2 of [tiltbridge](https://github.com/thorrak/tiltbridge)?

I see one of the developers of tiltbrige linked to [this esp-idf file](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/esp_event/include/esp_event_legacy.h#L29-L62) in [the original issue thread](https://github.com/thorrak/tiltbridge/issues/171#issuecomment-812920184). [This file](https://github.com/espressif/arduino-esp32/blob/5502879a5b25e5fff84a7058f448be481c0a1f73/libraries/WiFi/src/WiFiGeneric.cpp#L829-L835) is also linked in that thread.

xueyunfei998, 2022-08-31:
[image]

xueyunfei998, 2022-08-31:
[image]

MaxwellAlan, 2024-01-11:
Hi @SillyHats 

We will close this issue. Please feel free to reopen if still have this issue.

closed 2024-01-11 by MaxwellAlan