espressif/esp-idf #6843: Custom classic Bluetooth Gamepad on BTstack got error "E(25, 0) in user_cfm_req_reply"

limithsieh, 2021-04-08:
## Environment

- Development Kit:     |ESP32-DevKitC]
- Kit version (DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version : v4.4-dev-744-g1cb31e5
- Build System:         [Make]
- Compiler version :xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:  [external 3.3V]

## Problem Description

I have made a own classic bluettooth gamepad on BTstack with esp32.
I modify the BTstack example "hid_mouse_demo" into a gamepad.
It's strange when Google pixel 4a5G connect to it, it will get frozen 30 seconds and show error "E(25, 0) in user_cfm_req_reply".
But everything is normal on other Android mobile phone like Samsung s10e, Asus zenfone5z.
The same project also implement on PC bluetooth dongle and it works great when Pixel 4a connect to it. 
I am wondering what's the meaning of "E(25, 0) in user_cfm_req_reply", "ASSERT_PARAM(25 0), in lc_task.c at line 3328"

### Expected Behavior

My own Bluetooth gamepad will connect to Pixel 4a5G correctly.

### Actual Behavior

A 30 seconds freeze about mobile phone show "Bluetooth Pair request" dialog and user tap "OK"

### Steps to reproduce

1. use gamepad hid descriptor 
2. set Name "Wireless Controller"
3. Use Pixel 4a connect to it

### Code to reproduce this issue

```cpp
int app_main(int argc, const char * argv[]){

    (void)argc;
    (void)argv;
    
    main_init();

    gap_set_class_of_device(0x2508);
    gap_set_local_name("Wireless Controller");

    // L2CAP
    l2cap_init();

    // SDP Server
    sdp_init();

    hid_create_sdp_record(hid_service_buffer,   sdp_record_handler_id_1, emulate_hid_class, 33, 0, 0, 0, 0, 0, hid_descriptor, hid_descriptor_size, hid_device_name);
    sdp_register_service(hid_service_buffer);

   // printf("hid_device_init\n");
    hid_device_init(0, hid_descriptor_size, >hid_descriptor);
    // HID Device
    // register for HCI events
    hci_event_callback_registration.callback = &packet_handler;
    hci_add_event_handler(&hci_event_callback_registration);
   // hci_register_sco_packet_handler(&packet_handler);//kj

    // register for HID
    hid_device_register_packet_handler(&packet_handler);

    //stan set extend inquiry data
    gap_set_default_link_policy_settings( LM_LINK_POLICY_ENABLE_ROLE_SWITCH | LM_LINK_POLICY_ENABLE_SNIFF_MODE );
    gap_set_allow_role_switch(true);

    gap_discoverable_control(1);
    hci_power_control(HCI_POWER_ON);

    btstack_run_loop_execute();

    return 0;
}

```
## Debug Logs

Bluetooth packet dump
https://1drv.ms/u/s!ArmTRksdUu4ajrosF_W78DMf80GQ0w?e=wLUBn6

## Other items if possible

Bluetooth packet dump
https://1drv.ms/u/s!ArmTRksdUu4ajrosF_W78DMf80GQ0w?e=wLUBn6

Wth-Esp, 2021-05-06:
Hi @limithsieh 

`ASSERT_PARAM(25 0), in lc_task.c at line 3328` means the state that ESP32 has sent LMP_Detach PUDU and is in the disconnecting procedure, but the actual ongoing procedure is IO_Cap exchange -- a part of the connecting procedure.

We will find a Google Pixel 4a and try to reproduce the issue asap.

Thanks

Wth-Esp, 2021-05-06:
Hi @limithsieh 

I found a Google Pixel 4 and run the  HID_mouse_demo in BTStack. The issue you come up with has been reproduced.
Here is the log:

`[00:04:06.345] LOG -- hid_mouse_demo.c.237: SSP User Confirmation Auto accept

[00:04:07.981] LOG -- hci.c.398: gap_store_link_key_for_bd_addr: F8:0F:F9:D1:00:E1, type 7
[00:04:07.983] LOG -- btstack_link_key_db_tlv.c.152: tag_for_addr 42544c0f, tag_for_empy 42544c0e, tag_for_lowest_seq_nr 42544c0f
[00:04:07.990] LOG -- btstack_link_key_db_tlv.c.166: store with tag 42544c0f
[00:04:07.997] LOG -- btstack_tlv_esp32.c.97: store tag 42544C0F
[00:04:08.005] LOG -- hci.c.2150: Handle 0081 key Size: 16
[00:04:08.007] LOG -- hci.c.5049: hci_emit_security_level 2 for handle 81
[00:04:08.014] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0081
[00:04:08.022] LOG -- l2cap.c.1989: create channel 0x3ffbd5c8, local_cid 0x0045
[00:04:08.028] LOG -- hci.c.5191: gap_request_security_level requested level 2, planned level 0, current level 2
[00:04:08.038] LOG -- hci.c.5049: hci_emit_security_level 2 for handle 81
[00:04:08.045] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0081
[00:04:08.052] LOG -- l2cap.c.2347: channel 0x3ffbd5c8, cid 0045 - state 5: actual 2 >= required 2?
[00:04:08.061] LOG -- l2cap.c.1084: L2CAP_EVENT_INCOMING_CONNECTION addr F8:0F:F9:D1:00:E1 handle 0x81 psm 0x11 local_cid 0x45 remote_cid 0x45
[00:04:08.074] LOG -- l2cap.c.2600: L2CAP_ACCEPT_CONNECTION local_cid 0x45
[00:04:08.080] LOG -- l2cap.c.1989: create channel 0x3ffbd63c, local_cid 0x0046
[00:04:08.087] LOG -- hci.c.5191: gap_request_security_level requested level 0, planned level 0, current level 2
[00:04:08.097] LOG -- hci.c.5049: hci_emit_security_level 2 for handle 81
[00:04:08.103] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0081
[00:04:08.111] LOG -- l2cap.c.2347: channel 0x3ffbd5c8, cid 0045 - state 17: actual 2 >= required 2?
[00:04:08.120] LOG -- l2cap.c.2347: channel 0x3ffbd63c, cid 0046 - state 5: actual 2 >= required 0?
[00:04:08.128] LOG -- l2cap.c.1084: L2CAP_EVENT_INCOMING_CONNECTION addr F8:0F:F9:D1:00:E1 handle 0x81 psm 0x1 local_cid 0x46 remote_cid 0x46
[00:04:08.141] LOG -- l2cap.c.2600: L2CAP_ACCEPT_CONNECTION local_cid 0x46
[00:04:08.148] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x45
[00:04:08.154] LOG -- l2cap.c.1251: l2cap_start_rtx for local cid 0x45
[00:04:08.161] LOG -- l2cap.c.2809: L2CAP signaling handler code 4, state 11
[00:04:08.167] LOG -- l2cap.c.2656: Remote MTU 640
[00:04:08.172] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x46
[00:04:08.177] LOG -- l2cap.c.1251: l2cap_start_rtx for local cid 0x46
[00:04:08.213] LOG -- l2cap.c.2809: L2CAP signaling handler code 5, state 11
[00:04:08.213] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x45
[00:04:08.215] LOG -- l2cap.c.2740: l2cap_signaling_handle_configure_response
[00:04:08.222] LOG -- l2cap.c.1050: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr F8:0F:F9:D1:00:E1 handle 0x81 psm 0x11 local_cid 0x45 remote_cid 0x45 local_mtu 100, remote_mtu 640, flush_timeout 0
[00:04:08.266] LOG -- l2cap.c.2809: L2CAP signaling handler code 4, state 11
[00:04:08.267] LOG -- l2cap.c.2656: Remote MTU 1024
[00:04:08.318] LOG -- l2cap.c.1989: create channel 0x3ffbd6b0, local_cid 0x0047
[00:04:08.318] LOG -- hci.c.5191: gap_request_security_level requested level 2, planned level 0, current level 2
[00:04:08.324] LOG -- hci.c.5049: hci_emit_security_level 2 for handle 81
[00:04:08.330] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0081
[00:04:08.338] LOG -- l2cap.c.2347: channel 0x3ffbd5c8, cid 0045 - state 12: actual 2 >= required 2?
[00:04:08.347] LOG -- l2cap.c.2347: channel 0x3ffbd63c, cid 0046 - state 11: actual 2 >= required 0?
[00:04:08.355] LOG -- l2cap.c.2347: channel 0x3ffbd6b0, cid 0047 - state 5: actual 2 >= required 2?
[00:04:08.364] LOG -- l2cap.c.1084: L2CAP_EVENT_INCOMING_CONNECTION addr F8:0F:F9:D1:00:E1 handle 0x81 psm 0x13 local_cid 0x47 remote_cid 0x47
[00:04:08.377] LOG -- l2cap.c.2600: L2CAP_ACCEPT_CONNECTION local_cid 0x47
[00:04:08.384] LOG -- l2cap.c.2809: L2CAP signaling handler code 5, state 11
[00:04:08.390] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x46
[00:04:08.396] LOG -- l2cap.c.2740: l2cap_signaling_handle_configure_response
[00:04:08.403] LOG -- l2cap.c.1050: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr F8:0F:F9:D1:00:E1 handle 0x81 psm 0x1 local_cid 0x46 remote_cid 0x46 local_mtu 1691, remote_mtu 1024, flush_timeout 0
[00:04:08.421] LOG -- l2cap.c.2809: L2CAP signaling handler code 4, state 11
[00:04:08.427] LOG -- l2cap.c.2656: Remote MTU 640
[00:04:08.432] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x47
[00:04:08.438] LOG -- l2cap.c.1251: l2cap_start_rtx for local cid 0x47
[00:04:08.459] LOG -- l2cap.c.2809: L2CAP signaling handler code 5, state 11
[00:04:08.459] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x47
[00:04:08.461] LOG -- l2cap.c.2740: l2cap_signaling_handle_configure_response
[00:04:08.468] LOG -- l2cap.c.1050: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr F8:0F:F9:D1:00:E1 handle 0x81 psm 0x13 local_cid 0x47 remote_cid 0x47 local_mtu 100, remote_mtu 640, flush_timeout 0
HID Connected, control mouse using 'a','s',''d', 'w' keys for movement and 'l' and 'r' for buttons...
[00:04:08.494] LOG -- l2cap.c.2809: L2CAP signaling handler code 6, state 12
[00:04:08.501] LOG -- l2cap.c.1078: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x46
[... cut ...]
[00:04:08.645] LOG -- l2cap.c.1050: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr F8:0F:F9:D1:00:E1 handle 0x81 psm 0x1 local_cid 0x48 remote_cid 0x48 local_mtu 1691, remote_mtu 1024, flush_timeout 0
[00:04:08.707] LOG -- l2cap.c.2809: L2CAP signaling handler code 6, state 12
[00:04:08.708] LOG -- l2cap.c.1078: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x48
[00:04:08.710] LOG -- l2cap.c.1995: free channel 0x3ffbd63c, local_cid 0x0048
[00:04:08.717] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x48`

Thanks

closed 2021-06-11 by Alvin1Zhang