espressif/esp-idf #6849: Build fails if CONFIG_SPI_FLASH_ROM_DRIVER_PATCH is unset

boarchuz, 2021-04-09:
## Environment

- Module or chip used:  ESP32
- IDF version: v4.4-dev-744-g1cb31e509

## Problem Description

If "Enable SPI flash ROM driver patched functions" (CONFIG_SPI_FLASH_ROM_DRIVER_PATCH) is disabled then bootloader linking fails with many undefined references to `esp_rom_spiflash_` functions.

<details><summary>LOG</summary>

> [88/89] Linking C executable bootloader.elf
> FAILED: bootloader.elf 
> : && /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address   CMakeFiles/bootloader.elf.dir/project_elf_src_esp32.c.obj  -o bootloader.elf  esp-idf/hal/libhal.a  esp-idf/soc/libsoc.a  esp-idf/micro-ecc/libmicro-ecc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/efuse/libefuse.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/log/liblog.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/matt/espressif/Projects/hello_world/build/bootloader/bootloader.map  -Wl,--gc-sections  -fno-rtti  -fno-lto  esp-idf/hal/libhal.a  esp-idf/soc/libsoc.a  esp-idf/micro-ecc/libmicro-ecc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/efuse/libefuse.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/log/liblog.a  esp-idf/hal/libhal.a  esp-idf/soc/libsoc.a  esp-idf/micro-ecc/libmicro-ecc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/efuse/libefuse.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/log/liblog.a  esp-idf/hal/libhal.a  esp-idf/soc/libsoc.a  esp-idf/micro-ecc/libmicro-ecc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/efuse/libefuse.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/log/liblog.a  esp-idf/hal/libhal.a  esp-idf/soc/libsoc.a  esp-idf/micro-ecc/libmicro-ecc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/efuse/libefuse.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/log/liblog.a  -u abort  /home/matt/espressif/esp-idf/components/xtensa/esp32/libxt_hal.a  -L /home/matt/espressif/esp-idf/components/esp32/ld  -T esp32.peripherals.ld  -L /home/matt/espressif/esp-idf/components/esp_rom/esp32/ld  -T esp32.rom.ld  -T esp32.rom.api.ld  -T esp32.rom.libgcc.ld  -T esp32.rom.newlib-funcs.ld  -L /home/matt/espressif/esp-idf/components/bootloader/subproject/main/ld/esp32  -T bootloader.ld  -T bootloader.rom.ld && :
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bootloader_support/libbootloader_support.a(bootloader_flash.c.obj):(.literal.bootloader_flash_read+0x14): undefined reference to `esp_rom_spiflash_read'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bootloader_support/libbootloader_support.a(bootloader_flash.c.obj):(.literal.bootloader_flash_write+0xc): undefined reference to `esp_rom_spiflash_write_encrypted'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bootloader_support/libbootloader_support.a(bootloader_flash.c.obj):(.literal.bootloader_flash_write+0x10): undefined reference to `esp_rom_spiflash_write'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bootloader_support/libbootloader_support.a(bootloader_flash.c.obj):(.literal.bootloader_flash_erase_sector+0x0): undefined reference to `esp_rom_spiflash_erase_sector'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bootloader_support/libbootloader_support.a(bootloader_flash.c.obj): in function `bootloader_flash_read_no_decrypt':
> /home/matt/espressif/esp-idf/components/bootloader_support/src/bootloader_flash.c:285: undefined reference to `esp_rom_spiflash_read'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bootloader_support/libbootloader_support.a(bootloader_flash.c.obj): in function `bootloader_flash_write':
> /home/matt/espressif/esp-idf/components/bootloader_support/src/bootloader_flash.c:407: undefined reference to `esp_rom_spiflash_write_encrypted'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/matt/espressif/esp-idf/components/bootloader_support/src/bootloader_flash.c:409: undefined reference to `esp_rom_spiflash_write'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bootloader_support/libbootloader_support.a(bootloader_flash.c.obj): in function `bootloader_flash_erase_sector':
> /home/matt/espressif/esp-idf/components/bootloader_support/src/bootloader_flash.c:415: undefined reference to `esp_rom_spiflash_erase_sector'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/spi_flash/libspi_flash.a(spi_flash_rom_patch.c.obj):(.literal.esp_rom_spiflash_wait_idle+0x8): undefined reference to `esp_rom_spiflash_read_status'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/spi_flash/libspi_flash.a(spi_flash_rom_patch.c.obj):(.literal.esp_rom_spiflash_unlock+0x14): undefined reference to `esp_rom_spiflash_write_status'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/spi_flash/libspi_flash.a(spi_flash_rom_patch.c.obj):(.literal.esp_rom_spiflash_unlock+0x18): undefined reference to `esp_rom_spiflash_read_statushigh'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/spi_flash/libspi_flash.a(spi_flash_rom_patch.c.obj): in function `esp_rom_spiflash_wait_idle':
> /home/matt/espressif/esp-idf/components/spi_flash/esp32/spi_flash_rom_patch.c:47: undefined reference to `esp_rom_spiflash_read_status'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/spi_flash/libspi_flash.a(spi_flash_rom_patch.c.obj): in function `esp_rom_spiflash_unlock':
> /home/matt/espressif/esp-idf/components/spi_flash/esp32/spi_flash_rom_patch.c:73: undefined reference to `esp_rom_spiflash_read_status'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/matt/espressif/esp-idf/components/spi_flash/esp32/spi_flash_rom_patch.c:104: undefined reference to `esp_rom_spiflash_write_status'
> /home/matt/espressif/Tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/matt/espressif/esp-idf/components/spi_flash/esp32/spi_flash_rom_patch.c:109: undefined reference to `esp_rom_spiflash_read_statushigh'
> collect2: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.
> [245/983] Building C object esp-idf/vfs/CMakeFiles/__idf_vfs.dir/vfs.c.obj
> ninja: build stopped: subcommand failed.
> The terminal process "/bin/bash '-c', 'cmake --build .'" terminated with exit code: 1.
</details>

## Fix
```
if(NOT CONFIG_SPI_FLASH_ROM_DRIVER_PATCH)
    rom_linker_script("spiflash")
endif()
```
^ This needs to be added to the bootloader section here (but I'm not sure for which targets)
https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/esp_rom/CMakeLists.txt#L19-L35

Reduces bootloader size by about 1.5kb when disabled, for those interested.

Icarus113, 2021-09-15:
Hi @boarchuz , thanks for look at this.     
If you don't build those patch functions, and use all of them by ROM version instead. I'm afraid that not all of them can work. For these patch functions, each of them has their own reason to be patched. Maybe some chips don't have the ROM implementation of these functions, or some ROM implementations have bug.

boarchuz, 2021-09-15:
@Icarus113 The config option has some info explaining the cases where the patched versions need to be used (D2WD, non-standard SPI config such as with PICO packages, ISSI flash).

Is this inaccurate? I think the config option should be removed if the ROM driver should never be used regardless of hardware configuration.

Is this still true for ECO V3? Was the driver not fixed for the latest ROM version?

For what it's worth, the ROM driver builds and worked fine testing briefly with a PICO-V3-02 (Boya flash).

Icarus113, 2021-09-16:
Hi @boarchuz , I have double checked the `esp32.rom.spiflash.ld` file and our rom code, these functions are correctly implemented there. I will deal with your PR soon. Thanks for the contribution!

For your question about ECO V3. I think this one didn't change the rom code.

closed 2021-09-24 by espressif-bot