espressif/esp-idf #6850: esp_http_client can't POST multipart requests.

SergeySlyusarEZlo, 2021-04-09:
I'm using  esp_http_client for upload files to server. But now we shoul to send big files by multipart requests with Content-Type like: content-type\":\"multipart/form-data; boundary=-------.

I do not found this possibility using esp_http_client. How can i solve this problem?

lhespress, 2021-04-13:
@SergeySlyusarEZlo Could you share the detailed information, such as  the commit ID of IDF, which example I can run and so on.

SergeySlyusarEZlo, 2021-04-14:
ID: 1cb31e50943bb757966ca91ed7f4852692a5b0ed.

SergeySlyusarEZlo, 2021-04-14:
The question about support  multipart post requests using IDF. As i see now multipart requests we can create only manually using http_client_open and http_client_write methods. Are there any ready-made working examples of (multipart POST) uploading using IDF. Examples of esp-forum that i found work incorrectly. For example https://www.esp32.com/viewtopic.php?t=7867 once with corrected esp_http_client_open length fix not worked. I Tested with  https://httpbin.org/post or https://ptsv2.com/.

lhespress, 2021-04-14:
@SergeySlyusarEZlo What do you mean it doesn't work just you do it as the example code https://www.esp32.com/viewtopic.php?t=7867? and you also want to call the other api which use http_client_open  and http_client_write?

SergeySlyusarEZlo, 2021-04-14:
Have you got any worked example for post multipart data to upload server (using IDF)?

closed 2021-04-16 by SergeySlyusarEZlo