espressif/esp-idf #6854: ESP32-s2 OTA Failed when AWS s3 bucket use as public HTTPS server

Darsh-Dev, 2021-04-10:
Hi,

I have tried to perform OTA by uploading the application binary on AWS s3 Bucket and use that URL to download firmware using native OTA sample code and I have used [AmazonRootCA1.pem](https://www.amazontrust.com/repository/AmazonRootCA1.pem) as a server certificate.
It fails to verify peer certificate!. below is the log.

> E (6615) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (6615) esp-tls-mbedtls: Failed to verify peer certificate!
I (6615) esp-tls-mbedtls: verification info:   ! The certificate is not correctly signed by the trusted CAE (6625) esp-tls: Failed to open new connection

I also tried the below command to get the server certificate. but it shows a chain of certificates and I confused about how can I store it in one file.

> openssl s_client -showcerts -connect s3.console.aws.amazon.com:443 </dev/null

Can anyone help me with which server certificate I am able to resolve the above error?

This is a [link](https://ota-seed-bucket.s3.ap-south-1.amazonaws.com/hello-world.bin) for my sample binary uploaded on the AWS s3 bucket and I give the public permission to it so anyone can access.

Thanks

chegewara, 2021-04-10:
Please check your S3 certificate:
```
Issued by:
Common Name (CN)	DigiCert Baltimore CA-2 G2
Organization (O)	DigiCert Inc
Organizational Unit (OU)	www.digicert.com
```

Darsh-Dev, 2021-04-12:
Hi @chegewara,

Thank you so much above certificate works fine for me.
But I am curious about know that how can I identify that this certificate is for so and so public server?
Is there any command or utility help to find out server certificate?

Please tell me how can you identify that this is the correct server certificate for AWS S3?

Thanks

mahavirj, 2021-04-12:
@Darsh-Dev 

Please see below comment:

https://github.com/espressif/esp-idf/issues/3021#issuecomment-460917118

closed 2021-04-13 by Darsh-Dev

GKFernandes, 2022-05-31:
Hello @mahavirj ,

I'm facing the same issue as mentioned by @Darsh-Dev, even though I'm using openssl s_client -showcerts -connect s3.amazonaws.com:443 < /dev/null and copying the last certificate, I can't get past the CA verification.

My code is advanced_http_ota example by the way.

Any tips?

Thank you!

GKFernandes, 2022-05-31:
Hello @mahavirj ,

Really grateful for your prompt reply.

Unfortunately my ESP32 is already almost full and it won't fit the certificate bundle.

Just as a heads up, OTA was working before using your explanation over https://github.com/espressif/esp-idf/issues/3021#issuecomment-460917118

mahavirj, 2022-05-31:
@GKFernandes 

Please refer to https://aws.amazon.com/blogs/storage/reminder-amazon-s3-and-amazon-cloudfront-migrating-service-certificates-to-amazon-trust-services-starting-march-23-2021/. My earlier comment was prior to this migration.

Please use root certificates as listed at https://www.amazontrust.com/repository/ and that should work for S3 endpoint.

GKFernandes, 2022-05-31:
@mahavirj 

It still won't work :(

I put all 4 root CA certificates in the same file, I'll leave it attached.

I'll leave a link to my public bucket/object here: https://komeco-solar-bucket.s3.amazonaws.com/pre_release6_4.bin

Should I use a different one?

shubhamkulkarni97, 2022-06-01:
Hi @GKFernandes,

I tried advanced_https_ota example using the certificates and S3 link that you shared in above comment and didn't face any issues with certificate verification.

I am attaching the patch with the change in advanced_https_ota example: [test_ota.patch.zip](https://github.com/espressif/esp-idf/files/8813001/test_ota.patch.zip)

Also sharing my console logs: [ota.log](https://github.com/espressif/esp-idf/files/8813016/ota.log)

Are you referring the ESP-IDF OTA examples for embedding the certificates in the binary? Or you are using a different approach to embed the certificates?

GKFernandes, 2022-06-01:
Hello people,

I managed to solve the problem at least partially, let me share with you what was happening and what I did to work it around.

I'm not sure why this is, but I noticed as soon as I placed the github ca_cert.pem nothing changed at all, actually, I managed to perform a random OTA using Amazon S3 bucket, which doesn't make any sense at all. At that moment I noticed that changing the certificate wasn't providing any effect on the behaviour of the ESP32.

What I noticed then is that I had this ca_cert.pem in 2 places: 1 was located in server_certs folder and the other one in data folder (a folder I use to host ESP's file system in this specific project), I was only changing the ca_cert.pem located in the data folder, as soon as I changed both (and recompiling the project as a consequence) I managed to make it work using github 😄 

So my next step is to try and make it work with AWS S3 bucket reliably, I'll edit this post as soon as I make any progress.

Thanks!

hbelser, 2023-03-04:
Hi folks, been following your experiments and think I am close, but getting a file not found error:

I (5680) advanced_https_ota_example: Starting Advanced OTA example
I (5690) wifi:<ba-add>idx:1 (ifx:0, 3c:82:c0:fa:a7:96), tid:0, ssn:1, winSize:64
E (7060) esp_https_ota: File not found(403)
E (7060) esp_https_ota: Failed to establish HTTP connection
E (7070) advanced_https_ota_example: ESP HTTPS OTA Begin failed

- I am using ESP IDF 4.4
- Tried both Amazon Root CA1 cert and the one from Starfield Services: same error
- The AWS S3 bucket is public, I am pointing the ESP32 OTA code to this URL:
 https://hubertsbucket174919-dev.s3.amazonaws.com/public/hello_world.bin

Any ideas?

Another thought: does a file in an S3 bucket get "served' using https automatically when you point to its URL? Some other posts seem to imply that you need to set up Cloudfront to serve S3 files ...

Thanks! Hubert

hbelser, 2023-03-04:
Never mind, just got it working! Even though the bucket was public, you need to give the bin file public permissions (Everyone (public access)). That is what triggered the File Not Found error. Alternatively, I think you could share the file with a pre-signed URL (Object Actions -> Share with a resigned URL).