espressif/esp-idf #6855: How to use the freemodbus lib with callbacks?

0xFEEDC0DE64, 2021-04-10:
I am looking for some documentation on how to use the freemodbus lib with callbacks called whenever the master tries to read my holding registers.

I found the needed function pointers in mb_slave_interface_t, which is not public API.

I would like to override slave_reg_cb_holding with my own handler and not use the linked list of memory regions as I cannot hold all modbus registers in memory at the same time, also I do not know when to read them.

Is there a way to not have 2 seperate threads spawn for the modbus slave (tcp and slave task)? I would like to poll in my main thread for incoming modbus messages

negativekelvin, 2021-04-10:
https://github.com/espressif/esp-idf/issues/6700

alisitsyn, 2021-04-12:
The user callback functions are not officially supported by API and it is not planned to open the interface to override them. It is hard to support this functionality due to deterministic aspects of the protocol. However there is a way to do it in your application. See the https://github.com/espressif/esp-idf/issues/6700#issuecomment-800180752 for more information on how to do this.

closed 2021-06-16 by Alvin1Zhang