espressif/esp-idf #6856: problem with uploding codes.

raoshubam96, 2021-04-11:
we are facing this problem with Esp32 wrover B after we are molding it to PCB we are not able to upload the codes and we are also using CP2102.
as our last enquiry they said to use the DTR and RTS signals should be used to generate an EN and IO0 reset sequence and we are doing that and we are prasing the switch of reset and IO0 while uploading the code but it still says failed to connect to moduel.
and as you shared last schematic of CP2102 IC to upload the coding at the place of it we are using the CP2102 module which already the IC in it.
if you connect to somone then we could solve this problem asap please.

dobairoland, 2021-04-12:
I'm linking here https://github.com/espressif/esp-idf/issues/6822 as previous issue opened regarding this. 

So if you have buttons connected to EN & GPIO0 then you can manually put the chip into download mode (Hold GPIO0, reset by EN, release GPIO0) . 1. Connect a terminal program to the serial port (baud 115200, 8 data bits, 1 stop bit, without parity, without flow control). 2. Put the chip into download mode manually by the buttons. 3. Post here the result what can be observed on the serial port. 

We will advise you further after completing these steps.

raoshubam96, 2021-04-14:
we are trying it but that's where we are stucked even after pressing the button it isn't downloading and showing the same on screen 
[image]
[image]

raoshubam96, 2021-04-14:
there is one more pin at no.4 where there is RTS GPIOo and we placed switch at both, still not working.

raoshubam96, 2021-04-14:
[image]
[image]

raoshubam96, 2021-04-14:
please check if there is any problem with the wrover chart or tools?

dobairoland, 2021-04-14:
> we are trying it but that's where we are stucked even after pressing the button it isn't downloading and showing the same on screen

That is why I suggest to check if the chip can be put into download mode. We are not trying to download anything, only see if the first step of downloading is possible. Please do the following with a simple terminal, e.g. [putty](https://www.chiark.greenend.org.uk/~sgtatham/putty/), not Arduino:

> So if you have buttons connected to EN & GPIO0 then you can manually put the chip into download mode (Hold GPIO0, reset by EN, release GPIO0) . 1. Connect a terminal program to the serial port (baud [...]

I've linked earlier a page where you can upload your schematics for checking: https://github.com/espressif/esp-idf/issues/6822#issuecomment-813308945

> our last enquiry they said to use the DTR and RTS signals should be used to generate an EN and IO0 reset sequence and we are doing that

I don't see that in your schematics. The DTR and RTS signals of CP2102 are not connected to ESP32.

raoshubam96, 2021-04-14:
we can only see 5 pins in CP2102 module
1 - 3.3wlt 
2- rx 
3- tx 
4 - gnd
5- vcc

how to connect DTR and RTS signal to ESP32?

ozgurbostan, 2021-04-14:
Hi @raoshubam96 ,

The related schematic has been shared by@dobairoland with the indication of implementation auto-boot and reset circuitry. 

This circuit includes two transistors used for bootloader mode entrance sequence with the help of CP2102 and esptool by USB interface. 

Please observe the shared schematic careful enough and there you will find your solution. 

https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch-20180607a.pdf

raoshubam96, 2021-04-14:
listen, we have made power supply according to your schematic only and also the reset circuit according to your schematic. we have placed a switch with ground connect with GPIO0 which is 25th number pin of ESP 32, now where should we give RTS or DTS connection? where we are using CP2102 module not CP2102 IC.

ozgurbostan, 2021-04-14:
So it's not possible to implement the auto-boot and reset circuitry with a CP2102 module if your module doesn't have these pins as outputs. If it does, then you should use a breadboard and related components to create your own circuit.

raoshubam96, 2021-04-14:
we have bought it from you and you are saying that there should be a pins for output but there is not, so what should we do ? 
should we buy new?

ozgurbostan, 2021-04-15:
I am not a worker of Espressif, I am just a user like you. 

And you did not understand what you read. You said that you are using CP2102 module; there is no point anyone else can do sth, it's your choice and under your responsibility. I am just trying to provide you a solution as a community member. But, you insist on your faults and not trying to understand what's happening.

raoshubam96, 2021-04-15:
i am sorry i didnt mean to come over you but its just that because of this our product has stucked and we can't move. 
the only thing i am saying is you saying to connect connect DTS and RTS but we unable to see output pins for these, do you have any picture of CP2102 showing DTS RTS pins?

negativekelvin, 2021-04-15:
Offical Espressif external programmer is ESP-PROG.  If you are trying to use CP2102 module as a power supply, it is too weak.

ozgurbostan, 2021-04-15:
I did just say; "if" you have DTR and RTS pins on your module, then you can use. Otherwise, it's not possible. I do not know which CP2102 module you are using, whether it is a custom designed or anything else. 

Some modules have only RTS pins just like below;
[image]

Some have only RTS just like below.
[image]

So, it's so simple; if you have you can use, if not you cannot.