espressif/esp-idf #6858: esptool.py failed with exit code 1

jonathanrjpereira, 2021-04-11:
## Environment

- Development Kit:  ESP32-DevKitC-32E
- Module or chip used:  ESP32-WROOM-32E
- IDF version (run ``git describe --tags`` to find it):
    // v4.1
- Build System:   Eclipse
- Operating System:   Windows
- (Windows only) environment type: Eclipse with IDF tools setup.
- Using an IDE?: Yes - Eclipse
- Power Supply:  USB

## Problem Description

I'm trying to run the Hello World Example using the Eclipse IDE with IDF tools installed and configured as described in the documentation. But when I run the code I get the following error: 

```
[899/901] Building C object CMakeFiles/hello-world.elf.dir/project_elf_src.c.obj
[900/901] Linking CXX executable hello-world.elf
[901/901] Generating binary image from built executable
esptool.py v2.9-dev
Generated C:/Users/Jonathan/eclipse-workspace/blink/build/hello-world.bin
esptool.py -p COM6 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
esptool.py v2.9-dev
Serial port COM6
Traceback (most recent call last):
  File "C:\Users\Jonathan\esp\components/esptool_py/esptool/esptool.py", line 3422, in <module>
    _main()
  File "C:\Users\Jonathan\esp\components/esptool_py/esptool/esptool.py", line 3415, in _main
    main()
  File "C:\Users\Jonathan\esp\components/esptool_py/esptool/esptool.py", line 3079, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:\Users\Jonathan\esp\components/esptool_py/esptool/esptool.py", line 261, in __init__
    self._port = serial.serial_for_url(port)
  File "C:\Users\Jonathan\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "C:\Users\Jonathan\.espressif\python_env\idf4.1_py3.7_env\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM6': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
Adding "flash"'s dependency "all" to list of commands with default set of options.
Executing action: all (aliases: build)
Running cmake in directory c:\users\jonathan\eclipse-workspace\blink\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=1 c:\users\jonathan\eclipse-workspace\blink"...
Running ninja in directory c:\users\jonathan\eclipse-workspace\blink\build
Executing "ninja all"...
Executing action: flash
Running esptool.py in directory c:\users\jonathan\eclipse-workspace\blink\build
Executing "C:\Users\Jonathan\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe C:\Users\Jonathan\esp\components/esptool_py/esptool/esptool.py -p COM6 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash @flash_project_args"...
esptool.py failed with exit code 1
```

chegewara, 2021-04-11:
Are you sure it is on port `COM6` and you do not have opened any terminal like putty on that port?

```
could not open port 'COM6': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
```

jonathanrjpereira, 2021-04-11:
The target that I had selected was configured to COM6 instead of COM3.

closed 2021-04-11 by jonathanrjpereira