espressif/esp-idf #6860: Calling ESP_LOGI from a critical section of a task function

norventa, 2021-04-12:
## Environment

- Development Kit:      [none]
- Module or chip used:  ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    3.40200.210118
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    PACKAGES:
 - framework-espidf 3.40200.210118 (4.2.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - tool-openocd-esp32 2.1000.20201202 (10.0)
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
 - 
- Operating System:     [Windows]
- (Windows only) environment type: [Platformio].
- Using an IDE?: [No Platformio]
- Power Supply:         [external 3.3V]

## Problem Description

// From a task function:

portMUX_TYPE mutex = portMUX_INITIALIZER_UNLOCKED;

void taskfn(void *)
{
    do
    {
        portENTER_CRITICAL(&mutex);
        ESP_LOGI("test", "Hello");
        portEXIT_CRITICAL(&mutex);
    } while (1);
}

### Expected Behavior

Output log.

### Actual Behavior

locks.c throws a wobbly (look for HERE in code below):

When xPortCanYield() is called it is used to determine if in an interrupt context or in a task context. It mistakenly believes it to be an interrupt context.

```
// From locks.c
/* Acquire the mutex semaphore for lock. wait up to delay ticks.
   mutex_type is queueQUEUE_TYPE_RECURSIVE_MUTEX or queueQUEUE_TYPE_MUTEX
*/
static int IRAM_ATTR lock_acquire_generic(_lock_t *lock, uint32_t delay, uint8_t mutex_type) {
    xSemaphoreHandle h = (xSemaphoreHandle)(*lock);
    if (!h) {
        if (xTaskGetSchedulerState() == taskSCHEDULER_NOT_STARTED) {
            return 0; /* locking is a no-op before scheduler is up, so this "succeeds" */
        }
        /* lazy initialise lock - might have had a static initializer in newlib (that we don't use),
           or _lock_init might have been called before the scheduler was running... */
        lock_init_generic(lock, mutex_type);
        h = (xSemaphoreHandle)(*lock);
        configASSERT(h != NULL);
    }

    BaseType_t success;
    if (!xPortCanYield()) {
[... 459 lines cut ...]
#define CONFIG_TASK_WDT CONFIG_ESP_TASK_WDT
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU1
#define CONFIG_TASK_WDT_TIMEOUT_S CONFIG_ESP_TASK_WDT_TIMEOUT_S
#define CONFIG_TCPIP_RECVMBOX_SIZE CONFIG_LWIP_TCPIP_RECVMBOX_SIZE
#define CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY
#define CONFIG_TCPIP_TASK_STACK_SIZE CONFIG_LWIP_TCPIP_TASK_STACK_SIZE
#define CONFIG_TCP_MAXRTX CONFIG_LWIP_TCP_MAXRTX
#define CONFIG_TCP_MSL CONFIG_LWIP_TCP_MSL
#define CONFIG_TCP_MSS CONFIG_LWIP_TCP_MSS
#define CONFIG_TCP_OVERSIZE_MSS CONFIG_LWIP_TCP_OVERSIZE_MSS
#define CONFIG_TCP_QUEUE_OOSEQ CONFIG_LWIP_TCP_QUEUE_OOSEQ
#define CONFIG_TCP_RECVMBOX_SIZE CONFIG_LWIP_TCP_RECVMBOX_SIZE
#define CONFIG_TCP_SND_BUF_DEFAULT CONFIG_LWIP_TCP_SND_BUF_DEFAULT
#define CONFIG_TCP_SYNMAXRTX CONFIG_LWIP_TCP_SYNMAXRTX
#define CONFIG_TCP_WND_DEFAULT CONFIG_LWIP_TCP_WND_DEFAULT
#define CONFIG_TIMER_QUEUE_LENGTH CONFIG_FREERTOS_TIMER_QUEUE_LENGTH
#define CONFIG_TIMER_TASK_PRIORITY CONFIG_FREERTOS_TIMER_TASK_PRIORITY
#define CONFIG_TIMER_TASK_STACK_DEPTH CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH
#define CONFIG_TIMER_TASK_STACK_SIZE CONFIG_ESP_TIMER_TASK_STACK_SIZE
#define CONFIG_TOOLPREFIX CONFIG_SDK_TOOLPREFIX
#define CONFIG_UDP_RECVMBOX_SIZE CONFIG_LWIP_UDP_RECVMBOX_SIZE

- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

igrr, 2021-04-13:
@norventa This looks like expected behavior, as blocking operations (such as logging) are not allowed in critical sections. The abort which happens is a few lines below, next to the comment:
```
/* Tried to block on mutex from ISR, couldn't... rewrite your program to avoid libc interactions in ISRs! */
```

The comments in the code related to ISRs are incorrect. In this code, `xPortCanYield` checks whether task switching is allowed, i.e. that a blocking operation can be attempted. ISR context is one of the cases when blocking operations are not allowed, and a critical section (with interrupts disabled) is the other one. We need to fix the comments to mention the case of critical sections as well.

More generally, even if logging without any blocking operations was possible, it wouldn't be recommended. Since logging is slow, interrupts would get delayed for an amount of time up to a few milliseconds, and this can lead to misbehavior of some of the drivers and Wi-Fi/BT stacks. I'd recommend restructuring your code to avoid logging from critical sections.

norventa, 2021-04-13:
Thank you for clarifying that.

closed 2021-04-13 by norventa