espressif/esp-idf #6861: Provisioning Apply config status not updating

CSHaze, 2021-04-13:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version DevKitC: v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
    // v4.4-dev-479-g1067b2870
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) ESP Command Prompt.
- Using an IDE?: Yes VSCode with Espressif IDF extension
- Power Supply: USB

## Problem Description

When querying for the apply config status during provisioning I keep getting the same result (connecting) regardless of if the connection is disconnected or connected.

### Expected Behavior

I should get a status showing the device is disconnected or connected in line with what is actually happening on the esp32.

### Actual Behavior

I get connecting
`{"status":0,"staState":1,"failReason":0}` regardless of the actual status on the esp32

### Steps to reproduce

1. Start provisioning and apply a wifi config with an invalid password.
2. Verify the esp32 shows it has disconnected because of an auth error.
    `E (101227) wifi_prov_mgr: STA Disconnected
     E (101237) wifi_prov_mgr: Disconnect reason : 15
     E (101237) wifi_prov_mgr: STA Auth Error`
3. Continue polling the apply config status every 5 seconds during this process.
4. Notice that the config status always comes back with connecting.

### Code to reproduce this issue

```
private buildWifiConfigStatusPayload(): Uint8Array {
    const cmdGetStatus = new CmdGetStatus();

    const wifiConfigPayload = new WiFiConfigPayload();
    wifiConfigPayload.setCmdGetStatus(cmdGetStatus);
    wifiConfigPayload.setMsg(WiFiConfigMsgType.TYPECMDGETSTATUS);

    return wifiConfigPayload.serializeBinary();
  }

this.write(
          this.provisioningDescriptors.descriptorDict[ProvisioningDescriptors.PROV_CONFIG],
          this.ble.bytesToString(this.buildWifiConfigStatusPayload())
        ).pipe(
          map((result: OperationResult) => {
            console.log(
              `apply config status: ${JSON.stringify(
                WiFiConfigPayload.deserializeBinary(this.ble.encodedStringToBytes(result.value))
                  .getRespGetStatus()
                  .toObject()
              )}`
            );
            const staState: number = WiFiConfigPayload.deserializeBinary(this.ble.encodedStringToBytes(result.value))
              .getRespGetStatus()
              .getStaState();
            if (staState === WifiStationState.CONNECTING) {
              throw 'still connecting';
            }
            return result;
[... 109 lines cut ...]
D (104317) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffddc98
D (104327) esp_netif_lwip: if0x3ffddc98 start ip lost tmr: no need start because netif=0x3ffddd18 interval=120 ip=0
D (104337) esp_netif_lwip: check: local, if=0x3ffddc98 fn=0x401485b4
0x401485b4: esp_netif_update_default_netif_lwip at C:/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:174

D (104347) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffddc98
D (104357) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (104357) event: running post WIFI_EVENT:5 with handler 0x400e0fec and context 0x3ffb6378 on loop 0x3ffdcc88
0x400e0fec: wifi_prov_mgr_event_handler_internal at C:/esp/esp-idf/components/wifi_provisioning/src/manager.c:805

E (104367) wifi_prov_mgr: STA Disconnected
E (104377) wifi_prov_mgr: Disconnect reason : 205
D (105667) protocomm_ble: Inside read w/ session - 0 on param 48 0
D (105667) protocomm_ble: Reading attr value first time
D (110757) protocomm_ble: Inside read w/ session - 0 on param 48 0
D (110767) protocomm_ble: Reading attr value first time
D (115937) protocomm_ble: Inside read w/ session - 0 on param 48 0
D (115937) protocomm_ble: Reading attr value first time
D (121077) protocomm_ble: Inside read w/ session - 0 on param 48 0
D (121077) protocomm_ble: Reading attr value first time
D (126177) protocomm_ble: Inside read w/ session - 0 on param 48 0
D (126177) protocomm_ble: Reading attr value first time
D (131347) protocomm_ble: Inside read w/ session - 0 on param 48 0
D (131347) protocomm_ble: Reading attr value first time
```

yuanjianmin, 2021-04-22:
@CSHaze Could you please tell me which platform(IOS/Android) and version APP you used to test it. Also please tell me which example you test. I couldn't reproduce the issue you met. Thanks.

CSHaze, 2021-04-22:
I am using Ionic which is Javascript. What is the APP version? I have written my own application and every other command seems to work. I thought it could be some kind of caching issue but I see each request hitting the ESP32.

yuanjianmin, 2021-04-22:
@CSHaze Sorry, I thought you were using our official [APP](https://github.com/espressif/esp-idf-provisioning-android). You can use this APP to test whether exist your issue.

CSHaze, 2021-04-22:
@yuanjianmin ah, gotcha. I will try that and let you know if I am still seeing the issue.

yuanjianmin, 2021-04-23:
@CSHaze Any update?

CSHaze, 2021-04-25:
@yuanjianmin sorry for the delay. I think I have resolved the issue. The problem was my polling code. There seemed to be some issue with caching or the read being run repeatedly without the write. Using your app helped confirm it was an issue with my polling code.

I appreciate all your help on this. I pretty much have provisioning completely working for now except the encryption piece, I was close at one point but can't figure out how to get the decryption to track the offset correctly using JavaScript libraries. A problem for another day. Again, thank you for all the help.

Alvin1Zhang, 2021-04-26:
@CSHaze Good to hear the issue resolved. Feel free to reopen.

closed 2021-04-26 by Alvin1Zhang