espressif/esp-idf #6862: BLE Mesh Provisioner AuthValue does not match Mesh Specification

dastarling, 2021-04-13:
## Environment
- Development Kit:     ESP32-C3-DevKitM-1
- Kit version [v1.0]
- Module or chip used: ESP32-C3-Mini-1
- IDF version (run ``git describe --tags`` to find it):
    v4.4-dev-744-g1cb31e509
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    riscv32-esp-elf-gcc (crosstool-NG 1.24.0.123_64eb9ff) 8.4.0
- Operating System:     Linux
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Yes
- Power Supply:         USB

## Problem Description
The AuthValue does not match between ESP and Nordic so Authentication will not succeed.  One value or the other is incorrect, please see output:

**NORDIC Node Device**
00> D: ConfirmationKey 19a40d5be0334b3d07593d423f19e01e
00> D: RandomDevice acca9e3dd988f4c89c5876ad3b23dfea
00> D: AuthValue 00000000000000000000000000000003

The Provisioner on an **ESP32** shows:
D (14353) BLE_MESH: AuthValue 00000000000000000000000003000000

### Expected Behavior
AuthValue 00000000000000000000000000000003

Mesh Spec 1.0.1 Provisioning Authentication

For example, if the Authentication with Output OOB method is used with Output OOB Action as Blink and
value outputted is 5, then the AuthValue shall be 0x00000000000000000000000000000005. The
AuthValue is then encoded as defined in Section 5.1 in order to compute confirmation values, resulting in
an array consisting of [0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
0x00, 0x00,0x05].

### Actual Behavior
D (14353) BLE_MESH: AuthValue 00000000000000000000000003000000

### Steps to reproduce

1. Setup ESP as Provisioner on BLE Mesh
2. Setup Nordic board such as NRF52840 as Node on BLE Mesh
3. Attempt to Provision via Blink Output Authentication.
4. Enter Number of Blinks into ESP system

## Debug Logs
See above

InfiniteYuan, 2021-04-19:
Please follow the steps below to modify `components/bt/esp_ble_mesh/mesh_core/provisioner_prov.c`:

[image]

dastarling, 2021-04-20:
Thanks, It appears that patch fixes the issue, what version will it appear in?

InfiniteYuan, 2021-04-20:
We will push to GitHub as soon as possible.

closed 2021-04-27 by espressif-bot