espressif/esp-idf #6865: xtensa-esp32-elf/bin/ld: cannot find -lrtc

MRxing1, 2021-04-13:
I'm compiling the firmware of micropython for esp32 with esp-idf 4.0. I encountered the following problems and asked for assistance；

xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$ make
MicroPython v1.13-30-g1d344f446-dirty
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
LINK build-GENERIC_SPIRAM/bootloader.elf
/home/xkp/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lrtc
collect2: error: ld returned 1 exit status
Makefile:979: recipe for target 'build-GENERIC_SPIRAM/bootloader.elf' failed
make: *** [build-GENERIC_SPIRAM/bootloader.elf] Error 1

xkp@xkp-VirtualBox:~/esp32/micropython/ports/esp32$ make V=1
MicroPython v1.13-30-g1d344f446-dirty
Building with ESP IDF v4
LINK build-GENERIC_SPIRAM/bootloader.elf
xtensa-esp32-elf-gcc -nostdlib -L/home/xkp/esp32/esp-idf/lib -L/home/xkp/esp32/esp-idf/ld -u call_user_start_cpu0 -Wl,--gc-sections -static -Wl,-EL -Wl,-Map=build-GENERIC_SPIRAM/bootloader.map -Wl,--cref -T /home/xkp/esp32/esp-idf/components/bootloader/subproject/main/esp32.bootloader.ld -T /home/xkp/esp32/esp-idf/components/bootloader/subproject/main/esp32.bootloader.rom.ld -T /home/xkp/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld -T /home/xkp/esp32/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld -T /home/xkp/esp32/esp-idf/components/esp32/ld/esp32.peripherals.ld -o build-GENERIC_SPIRAM/bootloader.elf -Wl,--start-group -Lbuild-GENERIC_SPIRAM/bootloader -lbootloader_support -llog -lspi_flash -lsoc -lmain -L/home/xkp/esp32/esp-idf/components/esp_wifi/lib_esp32 -lrtc -L/home/xkp/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/esp32-psram/ -lgcc -Wl,--end-group
/home/xkp/esp32/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lrtc
collect2: error: ld returned 1 exit status
Makefile:979: recipe for target 'build-GENERIC_SPIRAM/bootloader.elf' failed
make: *** [build-GENERIC_SPIRAM/bootloader.elf] Error 1

Hope to get your help, thank you very much

igrr, 2021-04-13:
Please check if librtc.a is found in /home/xkp/esp32/esp-idf/components/esp_wifi/lib_esp32, the file is normally in that directory. If the directory is empty, it means that the submodules haven't been initialized (`git submodule update --init` in ESP-IDF directory).
One additional point, it looks like you are trying to build with ESP-IDF 4.0 and GCC 5.2.0 based Xtensa toolchain. This is not supported, please use GCC 8.x based Xtensa toolchain for IDF 4.x releases: https://docs.espressif.com/projects/esp-idf/en/release-v4.0/get-started/index.html#step-3-set-up-the-tools.

MRxing1, 2021-04-14:
OK, thank you very much for your reply. Now is OK!  How did I solve it?

I'll just download clone again from GitHub and the warehouse. It was originally from the warehouse clone that I transferred from gitee, and the sub module was imported by myself. Maybe the sub module that I imported from gitee is incomplete.

MRxing1, 2021-04-14:
> Please check if librtc.a is found in /home/xkp/esp32/esp-idf/components/esp_wifi/lib_esp32, the file is normally in that directory. If the directory is empty, it means that the submodules haven't [...]

OK, thank you very much for your reply.
After careful examination, my own ESP IDF repository on gitee is really short of lib_ Esp32, as you said, thank you very much.

But I need to confirm why I import gitee's warehouse from GitHub's warehouse. I need to think about why this file has not been imported together.

Thank you again for your guidance! Very useful!

Best wishes

igrr, 2021-04-14:
No problem, good to know the problem is solved!

closed 2021-04-14 by igrr