espressif/esp-idf #6866: invalid RTC_XTAL_FREQ_REG value: 0x00000000

markus-seidl, 2021-04-13:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-S2-WROOVER]
- IDF version (run ``git describe --tags`` to find it): ESP-IDF v4.3-beta2-2-g9a2d25191
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 3.3V

## Problem Description

Sometimes after waking up from deep sleep a HTTP request fails with a watchdog timeout (1 in 50 times).
The serial log shows then the message:

rtc_clk: invalid RTC_XTAL_FREQ_REG value 0x00000000

### Expected Behavior

### Actual Behavior

### Steps to reproduce

Unfortunately it's very hard to reproduce and happens every now and then (I guess it depends on how much time the server needs and/or how busy the WiFi network is.)

### Code to reproduce this issue

```cpp
void app_main()
{
// THIS IS PSEUDO CODE, the error happens while calling the function:

    // ESP_LOGI(TAG, "3");
    esp_http_client_config_t config = {
        .host = "192.168.1.121",
        .port = 8888,
        .path = "/",
        .query = query,
        //.event_handler = _http_event_handler,
        //.user_data = local_response_buffer,        // Pass address of local buffer to get response
        .disable_auto_redirect = true,
        .keep_alive_enable = false,
        .timeout_ms = 5000
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
15:54:10.041 -> [0;32mI (904) http: Sending data to server: t=26.006714&h=35.538261&mac=00:36:f1&uart=0&stat=-62,3138,0,0,0,1,1434[0m
15:54:10.041 -> [0;33mW (1194) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000[0m
15:54:10.041 -> [0;33mW (1494) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000[0m
15:54:10.041 -> [0;33mW (1764) rtc_clk: invalid RTC_XTAL_FREQ_REG value: 0x00000000[0m
15:54:10.041 -> ESP-ROM:esp32s2-rc4-20191025
15:54:10.041 -> Build:Oct 25 2019
15:54:10.041 -> rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
15:54:10.041 -> SPIWP:0xee
15:54:10.041 -> mode:DIO, clock div:1
15:54:10.041 -> load:0x3ffe6100,len:0x1b2c
15:54:10.041 -> load:0x4004c000,len:0xc98
15:54:10.041 -> load:0x40050000,len:0x3498
15:54:10.041 -> entry 0x4004c220
```

## Other items if possible

- [X] sdkconfig file (attach the sdkconfig file from your project folder) https://gist.github.com/markus-seidl/faa5ca6da36870a98a9a9443e3f57b3e 
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

markus-seidl, 2021-04-13:
I've uploaded the whole repository here: https://github.com/markus-seidl/temp 
The code that's executed last while the device resets is here: https://github.com/markus-seidl/temp/blob/b64610e5cf4b0fe9a9daff9c9da15df72a28cf23/main/http.c#L52

esp-lis, 2022-03-24:
> I've uploaded the whole repository here: https://github.com/markus-seidl/temp The code that's executed last while the device resets is here: https://github.com/markus-seidl/temp/blob/b64610e5cf4b0fe9a9daff9c9da15df72a28cf23/main/http.c#L52

I guess this may be caused by RTC bus isolate, you can use the following code to force on the RTC_PERIPH power domain before sleeping for testing.
`ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, `ESP_PD_OPTION_ON));`

jack0c, 2022-05-17:
@markus-seidl  any update?

markus-seidl, 2022-05-17:
@jack0c I've abandoned the project due to time constraints. I'm sorry, I can't provide any updates.

closed 2022-05-17 by Alvin1Zhang