espressif/esp-idf #6867: Ethernet Issue - Failed to get IP in Crossover Ethernet setup

nikhilavrio, 2021-04-13:
## Environment

- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Module or chip used:  ESP32-WROVER-IE
- IDF version: v4.2-351-gce2c615bb
- Build System:        idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     macOS
- Using an IDE?: VSCode
- Power Supply:        external 3.3V

## Problem Description

I am trying to setup Ethernet connection between SBC (Allwinner H3 running Linux) and ESP32 in Crossover manner.

I've programmed ESP32 Ethernet Iperf example (after modifying the LAN8720 pins).

It is failing to accept the IP Address provided by DHCP server running on the SBC. ESP32 is requesting for IP again and agian (infinite loop):

[image]

It does not look like the issue of HW because I've tested the connections in Static IP setup where both SBC and ESP32 have static IP. The connection was perfect:

[image]

I've attached the debug logs of Ethernet Iperf Example. Have a look.

Can you please let me know how to get DHCP working ?
[debug-logs.txt](https://github.com/espressif/esp-idf/files/6304444/debug-logs.txt)

Also, you might ask why I am using Dynamic IP instead of static ? 
I configured static IP and disabled DHCP client in ESP32 but after sometimes DHCP client turns on automatically loosing the ethernet connection. In our case, we want the ethernet connection to be alive 24x7.

### Expected Behavior

Ethernet Iperf example should get IP provided by DHCP and it should work.

### Actual Behavior

ESP32 is requesting DHCP for IP again and again in infinite loop.

### Steps to reproduce

1. Take SBC (Raspberry PI or any other) and connect with ESP32 via Ethernet (LAN8720) in Crossover manner. (Not straight through)
2. Run DHCP server on SBC
3. Program Ethernet Iperf code (after configuring the PHY chip).
4. Check the DHCP logs. ESP32 is asking for IP again and agian.

[image]

suda-morris, 2021-04-13:
the driver will detect the link status by reading PHY register, and one of the PHY register can tell if the link is up or down. If it's down->up, then start dhcp client. From your log, I can see, esp32 detects the link goes up and after a while, it goes down. Does this issue still exist if you connect your esp32 to a router instead of SBC?

nikhilavrio, 2021-04-15:
@suda-morris 

Yes, issue still exist if you connect your esp32 to a router instead of SBC
I've modified the LAN pins where ESP32 connects to Router instead of SBC (in Straight Through Mode instead of Crossover).

Problem remains the same. esp_netif_down_api is printed in loop, DHCP starts again.

PFA debug logs.
[debug-logs-router.txt](https://github.com/espressif/esp-idf/files/6315329/debug-logs-router.txt)

nikhilavrio, 2021-04-15:
I've compiled the binary for IDF v4.3-beta2 and master. The program crashes. 

[image]

[debug-logs-ethernet-iperf.txt](https://github.com/espressif/esp-idf/files/6315479/debug-logs-ethernet-iperf.txt)

suda-morris, 2021-04-15:
1. LAN8720 can handle the Straight Through and Crossover cables well, either cable type can work.
2. An unstable RMII clock can lead to such "link up and down" loop issue.

nikhilavrio, 2021-04-15:
Yes! 

Found out the issue. We are using external RMII clock (from 50Mhz oscillator) and GPIO0 was set to 1 as I was using SBC to program ESP32 as well. 

I've released the GPIO0 pin after reset code and added a startup delay in ESP32 Iperf example. 
Error is gone!

closed 2021-04-22 by nikhilavrio