espressif/esp-idf #6868: Support running clang-tidy with ESP-IDF

galah92, 2021-04-14:
I'd like to intergrate clang-tidy as part of my CI process for an application using ESP-IDF.
I'm using the IDF with a "vanilla" CMake, like the `idf_as_lib` example.
I've mutated that example `CMakeLists.txt` to be:
```cmake
cmake_minimum_required(VERSION 3.5)

project(idf_as_lib C)

set(CMAKE_CXX_CLANG_TIDY clang-tidy -checks=-*,readability-*)

include($ENV{IDF_PATH}/tools/cmake/idf.cmake)
idf_build_process(esp32
                COMPONENTS esp32 freertos esptool_py
                SDKCONFIG ${CMAKE_BINARY_DIR}/sdkconfig
                BUILD_DIR ${CMAKE_BINARY_DIR})

set(elf_file ${CMAKE_PROJECT_NAME}.elf)
add_executable(${elf_file} main.c)
target_link_libraries(${elf_file} idf::esp32 idf::freertos idf::spi_flash)
idf_build_executable(${elf_file})

set(CMAKE_EXPORT_COMPILE_COMMANDS 1)
```

Specifically, setting `CMAKE_CXX_CLANG_TIDY` integrate `clang-tidy` as part of the build process. I'm getting multiple errors, some can be handled by defining the IDF headers as system headers (`-isystem`), but the error I can't handle is the following:
> /home/galah92/esp-idf-v4.2-rc/components/newlib/platform_include/endian.h:24:10: error: 'machine/endian.h' file not found [clang-diagnostic-error]
#include <machine/endian.h>

Can this be fixed?

Attached the full log of errors for a specific file diagnosed by clang-tidy:
```
[207/581] Building CXX object esp-idf/cxx/CMakeFiles/__idf_cxx.dir/cxx_guards.cpp.obj
FAILED: esp-idf/cxx/CMakeFiles/__idf_cxx.dir/cxx_guards.cpp.obj 
/usr/bin/cmake -E __run_co_compile --tidy="clang-tidy;-checks=-*,readability-*" --source=/home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp -- /home/galah92/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++   -Iconfig -I/home/galah92/esp-idf-v4.2-rc/components/newlib/platform_include -I/home/galah92/esp-idf-v4.2-rc/components/freertos/include -I/home/galah92/esp-idf-v4.2-rc/components/freertos/xtensa/include -I/home/galah92/esp-idf-v4.2-rc/components/heap/include -I/home/galah92/esp-idf-v4.2-rc/components/log/include -I/home/galah92/esp-idf-v4.2-rc/components/lwip/include/apps -I/home/galah92/esp-idf-v4.2-rc/components/lwip/include/apps/sntp -I/home/galah92/esp-idf-v4.2-rc/components/lwip/lwip/src/include -I/home/galah92/esp-idf-v4.2-rc/components/lwip/port/esp32/include -I/home/galah92/esp-idf-v4.2-rc/components/lwip/port/esp32/include/arch -I/home/galah92/esp-idf-v4.2-rc/components/soc/src/esp32/. -I/home/galah92/esp-idf-v4.2-rc/components/soc/src/esp32/include -I/home/galah92/esp-idf-v4.2-rc/components/soc/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_rom/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_common/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_system/include -I/home/galah92/esp-idf-v4.2-rc/components/xtensa/include -I/home/galah92/esp-idf-v4.2-rc/components/xtensa/esp32/include -I/home/galah92/esp-idf-v4.2-rc/components/esp32/include -I/home/galah92/esp-idf-v4.2-rc/components/driver/include -I/home/galah92/esp-idf-v4.2-rc/components/driver/esp32/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_ringbuf/include -I/home/galah92/esp-idf-v4.2-rc/components/efuse/include -I/home/galah92/esp-idf-v4.2-rc/components/efuse/esp32/include -I/home/galah92/esp-idf-v4.2-rc/components/espcoredump/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_timer/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_ipc/include -I/home/galah92/esp-idf-v4.2-rc/components/soc/soc/esp32/include -I/home/galah92/esp-idf-v4.2-rc/components/soc/soc/esp32/../include -I/home/galah92/esp-idf-v4.2-rc/components/soc/soc/esp32/private_include -I/home/galah92/esp-idf-v4.2-rc/components/vfs/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_wifi/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_wifi/esp32/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_event/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_netif/include -I/home/galah92/esp-idf-v4.2-rc/components/esp_eth/include -I/home/galah92/esp-idf-v4.2-rc/components/tcpip_adapter/include -I/home/galah92/esp-idf-v4.2-rc/components/app_trace/include -I/home/galah92/esp-idf-v4.2-rc/components/pthread/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu++11 -fno-exceptions -fno-rtti -D_GNU_SOURCE -DIDF_VER=\"v4.2-rc\" -DESP_PLATFORM -MD -MT esp-idf/cxx/CMakeFiles/__idf_cxx.dir/cxx_guards.cpp.obj -MF esp-idf/cxx/CMakeFiles/__idf_cxx.dir/cxx_guards.cpp.obj.d -o esp-idf/cxx/CMakeFiles/__idf_cxx.dir/cxx_guards.cpp.obj -c /home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp
error: unknown argument '-mlongcalls'; did you mean '-mlong-calls'? [clang-diagnostic-error]
error: unknown argument: '-fstrict-volatile-bitfields' [clang-diagnostic-error]
/home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp:36:16: warning: redundant '__cxa_guard_acquire' declaration [readability-redundant-declaration]
extern "C" int __cxa_guard_acquire(__guard* pg);
           ~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/cxxabi.h:118:3: note: previously declared here
  __cxa_guard_acquire(__guard*);
  ^
/home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp:37:17: warning: redundant '__cxa_guard_release' declaration [readability-redundant-declaration]
extern "C" void __cxa_guard_release(__guard* pg);
           ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/cxxabi.h:121:3: note: previously declared here
  __cxa_guard_release(__guard*) _GLIBCXX_NOTHROW;
  ^
/home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp:38:17: warning: redundant '__cxa_guard_abort' declaration [readability-redundant-declaration]
extern "C" void __cxa_guard_abort(__guard* pg);
           ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/cxxabi.h:124:3: note: previously declared here
  __cxa_guard_abort(__guard*) _GLIBCXX_NOTHROW;
  ^
/home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp:101:13: warning: implicit conversion 'uint8_t' (aka 'unsigned char') -> bool [readability-implicit-bool-conversion]
    } while(g->pending);
            ^
                       != 0u
/home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp:115:12: warning: implicit conversion 'unsigned long' -> bool [readability-implicit-bool-conversion]
    while (count--) {
           ^
           (      ) != 0u
/home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp:125:13: warning: implicit conversion 'uint8_t' (aka 'unsigned char') -> bool [readability-implicit-bool-conversion]
        if (g->pending) {
            ^
                       != 0u
/home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp:143:13: warning: implicit conversion 'uint8_t' (aka 'unsigned char') -> bool [readability-implicit-bool-conversion]
        if (g->pending) {
            ^
                       != 0u
/home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp:160:9: warning: implicit conversion 'uint8_t' (aka 'unsigned char') -> bool [readability-implicit-bool-conversion]
    if (g->ready) {
        ^
                 != 0u
/home/galah92/esp-idf-v4.2-rc/components/newlib/platform_include/endian.h:24:10: error: 'machine/endian.h' file not found [clang-diagnostic-error]
#include <machine/endian.h>
         ^
588 warnings and 3 errors generated.
Error while processing /home/galah92/esp-idf-v4.2-rc/components/cxx/cxx_guards.cpp.
Suppressed 580 warnings (578 in non-user code, 2 with check filters).
Use -header-filter=.* to display errors from all non-system headers. Use -system-headers to display errors from system headers as well.
Found compiler error(s)
```

igrr, 2021-04-14:
Hi @galah92, this issue happens because clang-tidy is using host header files rather than the header files provided by xtensa-esp32-elf GCC based toolchain. The host header files on your system indeed don't contain machine/endian.h, while Newlib C library header files included in xtensa-esp32-elf toolchain do.
You may try to add yet more `-isystem` arguments to include xtensa-esp32-elf GCC toolchain's headers.

(Also linking https://github.com/espressif/esp-idf/issues/6213 which is also a request for clang-tidy integration, but at a different level (idf.py rather than pure CMake) — not for the purposes of your report, just for us to remember to update one when another is done)

igrr, 2022-03-22:
I'm sorry we forgot to update this issue in time. With the latest ESP-IDF `master` branch, clang-tidy integration works with the CMakeLists.txt file shown in the issue description.

The steps to get it working are:

1. Install Clang-based Xtensa toolchain: `$IDF_PATH/tools/idf_tools.py install xtensa-clang`
2. Run `. $IDF_PATH/export.sh` again
3. Configure the project:
    ```
    cmake -B build -DCMAKE_TOOLCHAIN_FILE=$IDF_PATH/tools/cmake/toolchain-clang-esp32.cmake -DTARGET=esp32 -GNinja .
    ```
    (note toolchain-clang-esp32.cmake instead of toolchain-esp32.cmake)
4. Build the project:
    ```
    cmake --build build --target app
    ```

Currently clang fails to parse some inline assembly instructions, which leads to a compilation error in esp_efuse_fields.c. This will be fixed in the next release of Clang-based Xtensa toolchain.

closed 2022-03-22 by igrr