espressif/esp-idf #6870: When using a switch, the ethernet fails to connect, until reseted

silardg, 2021-04-14:
- Module or chip used: ESP32-WROOM-32D
- IDF version: platformio 3.1.1
- Build System:      platformio
- Operating System:     macOS
- Using an IDE?: VSCode with PlatformIO
- Power Supply: External 12V, powering with 3.3V LDO

## Problem Description

I'm using LAN8720 for interfacing with ethernet. I'n connecting my ESP to my local network through a switch. When I power it up, it first fails to connect and I have to reset it for it to connect. Hardware is good, I used the layout and schematic from the official eval board.

I have seen a post where the flow control is disabled, but I can't find it in my version.
Post: https://www.esp32.com/viewtopic.php?t=6948

### Expected Behavior
Connects to the network
### Actual Behavior
Fails to connect
### Steps to reproduce
1. Use a switch with LAN8720
2. Connect

## Debug Logs

```
https://imgur.com/a/ReLVPp5
```

suda-morris, 2021-04-14:
Flow control on esp-idf v4.x is disabled by default. How is your ESP connected with LAN8720? (can post a schematic) How you configured the driver?

silardg, 2021-04-14:
`
void Network::connect()
{
    s_network_event_group = xEventGroupCreate();                                // create FreeRTOS event group
    ESP_ERROR_CHECK(esp_netif_init());                                          // init the TCP/IP stack
    ESP_ERROR_CHECK(esp_event_loop_create_default());                           // create a default event loop
    
    // if ethernet is configured
    if (config.ethernet)
    {
        esp_netif_config_t eth_cfg = ESP_NETIF_DEFAULT_ETH();                   // using default ethernet config
        esp_netif_t *eth_netif = esp_netif_new(&eth_cfg);                       // setup netif object with the prev config

        ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));               // Set default handlers to process TCP/IP stuffs

        // set the IP to static
        if(config.IPstatic)
        {
            info(TAG, "STATIC IP");

            esp_netif_dhcpc_stop(eth_netif);                                    // stop the dhcp
            
            esp_netif_set_ip_info(eth_netif, &this->ip_info);                   // setup the current ip

            // set the main and backup dnf info
            esp_netif_set_dns_info(eth_netif, ESP_NETIF_DNS_MAIN, &this->dns_info);
            esp_netif_set_dns_info(eth_netif, ESP_NETIF_DNS_BACKUP, &this->backup_dns_info);
        }

        // Register user defined event handlers
        ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &event_handler, NULL));
        ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &event_handler, NULL));

        eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();                 // use default mac config
        eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();                 // use default physical layer config
        
        phy_config.phy_addr             = 1;                                    // physical layer address set to hard 1 (problem when power is given it doesn't detect it)
        phy_config.reset_gpio_num       = -1;                                   // no hardware reset used
        phy_config.reset_timeout_ms     = 5000;                                 // reset timeout
        phy_config.autonego_timeout_ms  = 5000;                                 // 

        mac_config.smi_mdc_gpio_num     = ETH_MDC_PIN;                          // SMI MDC gpio pin
        mac_config.smi_mdio_gpio_num    = ETH_MDIO_PIN;                         // SMI MDIO gpio pin
        mac_config.sw_reset_timeout_ms  = 5000;                                 // reset timeout

        esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config);                // setup the mac config for the esp32
        esp_eth_phy_t *phy = esp_eth_phy_new_lan8720(&phy_config);              // setup the physical layer config for the lan8720 driver
        
        esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);                 // setup config for driver based on mac and phy config
        esp_eth_handle_t eth_handle = NULL;                                     // empty object
        
        ESP_ERROR_CHECK(esp_eth_driver_install(&config, &eth_handle));          // install the ethernet "driver" for the esp

        // attach Ethernet driver to TCP/IP stack 
        ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
        ESP_ERROR_CHECK(esp_eth_start(eth_handle));                             // ladies and gentelman, start up your ethernet engines!

        // wait for NETWORK_CONNECTED_BIT or NETWORK_FAIL_BIT to be set from the event handler
        EventBits_t bits = xEventGroupWaitBits(s_network_event_group,
                NETWORK_CONNECTED_BIT | NETWORK_FAIL_BIT,
                pdFALSE,
                pdFALSE,
                portMAX_DELAY);

        // depending on which one was set, act up on it
        if (bits & NETWORK_CONNECTED_BIT) {
            info(TAG, "Connected to network using Ethernet");
            if(valid(Network::networkReady)){ Network::networkReady(); }       // network is ready, mqtt starting
        } else if (bits & NETWORK_FAIL_BIT) {
            info(TAG, "Failed to connect to network using Ethernet");
[... 36 lines cut ...]

        // wait for NETWORK_CONNECTED_BIT or NETWORK_FAIL_BIT to be set from the event handler
        EventBits_t bits = xEventGroupWaitBits(s_network_event_group,
                NETWORK_CONNECTED_BIT | NETWORK_FAIL_BIT,
                pdFALSE,
                pdFALSE,
                portMAX_DELAY);

        // depending on which one was set, act up on it
        if (bits & NETWORK_CONNECTED_BIT) {
            info(TAG, "Connected to network using WiFi");
            if(valid(Network::networkReady)){ Network::networkReady(); }       // network is ready, mqtt starting
        } else if (bits & NETWORK_FAIL_BIT) {
            info(TAG, "Failed to connect to network using WiFi");
            if(valid(Network::networkError)){ Network::networkError(); }       // network has an error!

        } else {
            error(TAG, "UNEXPECTED EVENT");  
        }
    }
}
`

Schematic:
https://imgur.com/a/tmzWuxW

silardg, 2021-04-14:
`
void Network::event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    // WiFi events
    if(event_base == WIFI_EVENT) {
        switch(event_id) {
            case WIFI_EVENT_STA_START:
                // start connecting the WiFi
                esp_wifi_connect();
                break;
            case WIFI_EVENT_STA_DISCONNECTED:
                esp_wifi_connect();
                connected = false;
                info(TAG,"connect to the AP fail");
                break;
            default:
                info(TAG, "WIFI_EVENT - got event id %d", (int)event_id);
                break;
        }
    } 
    // IP Events
    else if(event_base == IP_EVENT) {
        switch(event_id) {
            case IP_EVENT_STA_GOT_IP: {
                ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;         // get the event object
                info(TAG, "got ip:" IPSTR, IP2STR(&event->ip_info.ip));             // get the ip info
                s_retry_num = 0;                                                    // reset counter
                xEventGroupSetBits(s_network_event_group, NETWORK_CONNECTED_BIT);   // network connected bit set
                break;
            }
            case IP_EVENT_ETH_GOT_IP: {
                ip_event_got_ip_t *event = (ip_event_got_ip_t *) event_data;        // get the event object
                const esp_netif_ip_info_t *ip_info = &event->ip_info;               // get the ip info

                // debug ethernet ip address
                info(TAG, "Ethernet Got IP Address");
                info(TAG, "~~~~~~~~~~~");
                info(TAG, "ETHIP:" IPSTR, IP2STR(&ip_info->ip));
                info(TAG, "ETHMASK:" IPSTR, IP2STR(&ip_info->netmask));
                info(TAG, "ETHGW:" IPSTR, IP2STR(&ip_info->gw));
                info(TAG, "~~~~~~~~~~~");

                xEventGroupSetBits(s_network_event_group, NETWORK_CONNECTED_BIT);   // network connected bit set
                break;
            }
            default:
                // some default event?!
                info(TAG, "IP_EVENT - got event id %d", (int)event_id);         
                break;
        }
    } else if(event_base == ETH_EVENT) {
        // placeholder for mac address
        uint8_t mac_addr[6] = {0};

        // we can get the ethernet driver handle from event data 
        esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;

        // handle the event_id
        switch (event_id) {
            case ETHERNET_EVENT_CONNECTED:
                connected = true;
                esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);                                    // get the ethernet mac address
                info(TAG, "Ethernet Link Up");       
                // logout the address
                info(TAG, "Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                        mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
                break;
            case ETHERNET_EVENT_DISCONNECTED:
                info(TAG, "Ethernet Link Down");
                connected = false;
                xEventGroupSetBits(s_network_event_group, NETWORK_FAIL_BIT);                                // set the fail bit
                break;
            case ETHERNET_EVENT_START:
                info(TAG, "Ethernet Started");
                break;
            case ETHERNET_EVENT_STOP:
                info(TAG, "Ethernet Stopped");
                connected = false;
                break;
            default:
                break;
        }
    } 
    else {
        info(TAG, "Network event not computed");
    }
}`

Handler

suda-morris, 2021-04-14:
OK, I think you have configured the Ethernet RMII clock to generated from GPIO17 of ESP32. But I can't tell exactly why your connection will be up and down in a loop... 
But I can somehow reproduce the same result by manually making the connection between GPIO17 and LAN8720 on and off. So I guess it's could be your hardware issue.

silardg, 2021-04-14:
This only shows up with one of my switches. What should I look for in the hardware?

suda-morris, 2021-04-15:
Sorry, I don't have experience in Switch, but if the switch detects any issue, it will close that port. Can you log in to that switch to see any logs?

Alvin1Zhang, 2021-06-02:
@silardg Thanks for reporting, would you please help share if any updates for the issue? Thanks.

silardg, 2021-06-02:
Nothing specific, I added it to detect only one device and not search on automatic mode for a phy device. 
Closing...

closed 2021-06-02 by silardg