espressif/esp-idf #6871: Required clarification about the Ble mesh scene server and scene server model

ankit-thealchemist, 2021-04-14:
- Kit version DevKitCv4
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-dev-2399-gea9063197
- Build System:         [Make|CMake|idf.py]
- Operating System:     Windows|

## Problem Description

I was implementing the scene server and the scene setup server model. I have 3 elements on the node. I have implement the scene server and the setup server on the primary element. Every thing seems fine . But when I implement the models on the other elements including the primary element, I got the warning that "Ble mesh scene server not found on the primary element" and "ble mesh scene setup server not found on the primary element". My question is that, is the implementation of the scene server and scene setup server is restricted to the primary element only?

InfiniteYuan, 2021-04-16:
[image]

ankit-thealchemist, 2021-04-16:
So the conclusion is, implementation of the scene server and setup server  has  only one instance in the node at the primary element only. 

Thanks for support!

closed 2021-04-16 by ankit-thealchemist