espressif/esp-idf #6873: Signed OTA updates fail when debugger is attached

RJPoelstra, 2021-04-14:
When the debugger is attached verification of signed OTA update images fails.
This is because the verification logic (in `image_load()`) skips the checksum check, but the checksum check also updates the image length attribute of `esp_image_metadata_t`.
When the signature of the images is then verified, the wrong position is used, as the `image_len` field is not updated. Furthermore, the corruption check (using a 'simple' hash) also fails due to the wrong end position being used.

The checksum check is skipped because when the debugger is attached software breakpoints might invalidate the checksum. This seems only relevant for running images, not for fresh images coming in via OTA.

I've attached a proposed fix, which extends the image loading enum with an `IGNORE_JTAG` option. This is then used in the OTA functions, knowing that a running debugger will not place software breakpoints in OTA images when they are checked after download.

[ota_ignore_jtag.patch.txt](https://github.com/espressif/esp-idf/files/6310564/ota_ignore_jtag.patch.txt)

KonstantinKondrashov, 2021-04-29:
Hi @RJPoelstra!
Thank you for the issue report. I have done a refactor of `image_load()` and fix the image_len. The length of the image is calculated correctly even if some checks are needed.
[12795.zip](https://github.com/espressif/esp-idf/files/6396337/12795.zip).
Could you help to check this patch to confirm that it helps?
Thanks.

RJPoelstra, 2021-04-29:
Hi,

Thanks for the update, but I'm sorry, it doesn't fully work.
It reports that the image is valid, but the signature bad.
I've tried with `espsecure.py verify_signature --version 1 --keyfile priv.pem build/S000030.bin`, which reports that the signature is valid.
Output from the ESP32:
```
I (597594) esp_image: Verifying image signature...
E (599134) esp_image: Secure boot signature verification failed
I (599134) esp_image: Calculating simple hash to check for corruption...
W (599424) esp_image: image valid, signature bad
```
After applying your patch and recompiling, I re-flashed the bootloader, partition table and application.
Do you want me to dig into something?

KonstantinKondrashov, 2021-05-04:
Hi @RJPoelstra! Thanks for testing. Yes, there was an error. Please if you have time try this one. [12795 (1).zip](https://github.com/espressif/esp-idf/files/6420672/12795.1.zip)
Thanks.

RJPoelstra, 2021-05-04:
@KonstantinKondrashov Thanks for this patch. I can confirm that it works for me.

closed 2021-05-17 by espressif-bot