espressif/esp-idf #6875: [C3 BLE] extended advertising

chegewara, 2021-04-15:
I dont know if this is issue or WIP, but i thought i will report it. 
I wanted to test extended advertising on C3 and nimble stack. Here is procedure:
- current master branch, commit 1cb31e50943bb757966ca91ed7f4852692a5b0ed
- esp32 C3 rev 2,
- OS: ubuntu 20.04,
- bleprph example, modified by enabling extended advertising in menuconfig,
- error when start advertising with not edited code and the same error when starting ext advertising:
```
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1640
load:0x403ce000,len:0x8ac
load:0x403d0000,len:0x2c00
entry 0x403ce000
I (46) boot: ESP-IDF v4.4-dev-744-g1cb31e509-dirty 2nd stage bootloader
I (46) boot: compile time 07:21:25
I (46) boot: chip revision: 2
I (50) boot.esp32c3: SPI Speed      : 80MHz
I (55) boot.esp32c3: SPI Mode       : DIO
I (60) boot.esp32c3: SPI Flash Size : 4MB
I (64) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  2 factory          factory app      00 00 00010000 00100000
I (103) boot: End of partition table
I (107) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=0f83ch ( 63548) map
I (125) esp_image: segment 1: paddr=0001f864 vaddr=3fc8e600 size=007b4h (  1972) load
I (126) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=5fc98h (392344) map
I (191) esp_image: segment 3: paddr=0007fcc0 vaddr=3fc8edb4 size=023e4h (  9188) load
I (193) esp_image: segment 4: paddr=000820ac vaddr=40380000 size=0e4ech ( 58604) load
I (208) esp_image: segment 5: paddr=000905a0 vaddr=50000000 size=00010h (    16) load
I (213) boot: Loaded app from partition at offset 0x10000
I (213) boot: Disabling RNG early entropy source...
I (229) cpu_start: Pro cpu up.
I (262) cpu_start: Pro cpu start user code
I (262) cpu_start: cpu freq: 160000000
I (262) cpu_start: Application information:
I (265) cpu_start: Project name:     bleprph
I (270) cpu_start: App version:      v4.4-dev-744-g1cb31e509-dirty
I (277) cpu_start: Compile time:     Apr 15 2021 07:28:38
I (283) cpu_start: ELF file SHA256:  36dd2fc124155ad0...
I (289) cpu_start: ESP-IDF:          v4.4-dev-744-g1cb31e509-dirty
I (296) heap_init: Initializing. RAM available for dynamic allocation:
I (303) heap_init: At 3FC93AB0 len 0002C550 (177 KiB): DRAM
I (309) heap_init: At 3FCC0000 len 0001F260 (124 KiB): STACK/DRAM
I (316) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (322) spi_flash: detected chip: generic
I (327) spi_flash: flash io: dio
I (331) sleep: Configure to isolate all GPIO pins in sleep state
I (337) sleep: Enable automatic switching of GPIO sleep configuration
I (345) cpu_start: Starting scheduler.
W (360) BTDM_INIT: esp_bt_controller_mem_release not implemented, return OK
I (362) BTDM_INIT: BT controller compile version [47235b6]
I (368) coexist: coexist rom version 8459080
I (373) phy_init: phy_version 400,4ba9e42,Mar 23 2021,17:02:06
I (492) system_api: Base MAC address is not set
I (493) system_api: read default base MAC address from EFUSE
I (493) BTDM_INIT: Bluetooth MAC: 7c:df:a1:43:35:92

I (502) NimBLE_BLE_PRPH: BLE Host Task Started
Device Address: 7c:df:a1:43:35:92
error setting advertisement data; rc=8
I (511) uart: queue free spaces: 8
```

Code changes:
```
    const struct ble_gap_ext_adv_params params = {
        .connectable = 1,
        .scannable = 1,
        .directed = 0,

        // unsigned int legacy_pdu
        // If use legacy PDUs for advertising.

        .anonymous = 1,
        // unsigned int include_tx_power
        // If include TX power in advertising PDU.

        .scan_req_notif = 1,
        // If enable scan request notification

        // uint32_t itvl_min
        // Minimum advertising interval in 0.625ms units, if 0 stack use sane defaults.

        // uint32_t itvl_max
        // Maximum advertising interval in 0.625ms units, if 0 stack use sane defaults.

        // uint8_t channel_map
        // Advertising channel map , if 0 stack use sane defaults.

        // uint8_t own_addr_type
        // Own address type to be used by advertising instance.

        // uint8_t filter_policy
        // Advertising Filter policy.

        .primary_phy = BLE_HCI_LE_PHY_1M,
        // Primary advertising PHY to use , can be one of following constants:
        // BLE_HCI_LE_PHY_1M
        // BLE_HCI_LE_PHY_CODED

        .secondary_phy = BLE_HCI_LE_PHY_2M,
        // Secondary advertising PHY to use, can be one of following constants:
        // BLE_HCI_LE_PHY_1M
        // LE_HCI_LE_PHY_2M
        // BLE_HCI_LE_PHY_CODED

        // int8_t tx_power
        // Preferred advertiser transmit power.

        // uint8_t sid
        // Advertising Set ID.
    };
    // (void)params;
    // http://mynewt.apache.org/latest/network/ble_hs/ble_gap.html#c.ble_gap_ext_adv_configure
    rc = ble_gap_ext_adv_configure(0, &params, NULL, bleprph_gap_event, NULL);
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "error setting EXT advertisement; rc=%d\n", rc);
    }
    rc = ble_gap_ext_adv_start(0, BLE_HS_FOREVER, 0);
    // rc = ble_gap_adv_start(own_addr_type, NULL, BLE_HS_FOREVER,
    //                        &adv_params, bleprph_gap_event, NULL);
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "error enabling advertisement; rc=%d\n", rc);
        return;
    }
```

Thanks

chegewara, 2021-04-16:
After reading nimble example i can see that configuration of ext adv is incorrect.
Sorry for report

closed 2021-04-16 by chegewara

prasad-alatkar, 2021-04-16:
@chegewara Thank you for reporting the issue. I see you have figured out the solution on your own :) However, I would like to put few comments on this issue which may help other members. Please feel free to add additional info and comments.

>error setting advertisement data; rc=8

When we enable `BLE_EXT_ADV` through menuconfig, the legacy APIs to configure and start advertisement (`ble_gap_adv_set_fields`, `ble_gap_adv_start`) are not supported by NimBLE stack hence `BLE_HS_ENOTSUP` error code.

Now for `ble_gap_ext_adv_params`, we can't set `scannable` and `connectable` adv types together as per the spec , ref: BLE 5.1 | Vol 2, Part E, 7.8.53 (`The advertisement shall not be both connectable and scannable`).

chegewara, 2021-04-16:
@prasad-alatkar Thanks for extended explanation. As i was not working with nimble before on esp32 it was not easy to find example for extended advertising, especially that i assumed that examples in esp-idf should be enough to get started.
It is not required of course, but would be handful to have BLE5.0 examples also for nimble, like other nimble examples.
Here is link to example i found, which helped me with this issue:
https://github.com/espressif/esp-nimble/tree/nimble-1.3.0-idf/apps/ext_advertiser

Again, thanks for support