espressif/esp-idf #6876: Build error when trying to access notification data with Nimble

afcec, 2021-04-15:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version: v4
- Module or chip used:  ESP32-WROOM-32
- IDF version: Reproduced with both, v4.0 and v4.1
    
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

If you execute Nimble example 'blecent' everything works fine but if you try to access the notification data using 'event->notify_rx.attr_data' from struct 'ble_gap_event' there is a compilation error:

In file included from ../../../../../components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs_log.h:23,
                 from ../../../../../components/bt/host/nimble/nimble/nimble/host/include/host/ble_hs.h:38,
                 from ../main/main.c:26:
../main/main.c: In function 'blecent_gap_event':
../main/main.c:456:53: **error: 'struct <anonymous>' has no member named 'attr_data'**; did you mean 'attr_handle'?
   MODLOG_DFLT(INFO, "received %s:",event->notify_rx.attr_data);

### Expected Behavior

Being ble to access notification data

### Actual Behavior

Notification data not be accesible 

### Steps to reproduce

Just try to build 'blecent' example and in callback 'blecent_gap_event' try to access 'event->notify_rx.attr_data' after receiving 'BLE_GAP_EVENT_NOTIFY_RX' event

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
case BLE_GAP_EVENT_NOTIFY_RX:
        /* Peer sent us a notification or indication. */
        MODLOG_DFLT(INFO, "received %s; conn_handle=%d attr_handle=%d "
                    "attr_len=%d\n",
                    event->notify_rx.indication ?
                    "indication" :
                    "notification",
                    event->notify_rx.conn_handle,
                    event->notify_rx.attr_handle,
                    OS_MBUF_PKTLEN(event->notify_rx.om));

        /* Attribute data is contained in event->notify_rx.attr_data. */
		MODLOG_DFLT(INFO, "received %s:",event->notify_rx.attr_data);
        return 0;

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

dhrishi, 2021-04-15:
@afcec Thank you for reporting the issue. The comment in the example seems to be incorrect as the related code is deprecated long back. We will get the comment updated.
The contents of the notification are in the os_mbuf that is present in the `notify_rx` structure. You can retrieve the data using something like:
```
        uint8_t notif_data[100]; /* Size depending on the actual size of the notification data you have */
        uint16_t notif_len;
        int offset = 0;
        notif_len = OS_MBUF_PKTLEN(event->notify_rx.om);
        os_mbuf_copydata(event->notify_rx.om, offset, notif_len, notif_data);

```
`notif_len` will have the length of notification data and the actual contents would be in `notif_data`.
Can you please give it a try and let me know if it helps?

afcec, 2021-04-15:
This works just great! thanks. By the way, Nimble documentation is very poor, at least when compared with Bluedroid. At some point it would be good to improve it somehow...

closed 2021-04-15 by afcec

dhrishi, 2021-04-15:
@afcec NimBLE being an open source stack, the documentation is not duplicated in ESP-IDF as it is already present here: https://mynewt.apache.org/latest/network/index.html. Host specific here: http://mynewt.apache.org/latest/network/ble_hs/ble_hs.html

Although, the reference to it is provided in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/nimble/index.html

We have retained all the APIs of NimBLE and only added a few new ones for host-controller init/deinit in our port. These are a part of ESP-IDF and hence their documentation can be found here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/nimble/index.html#api-reference