espressif/esp-idf #6877: How to disable efuse the public digest while enable secure boot V2

Darsh-Dev, 2021-04-15:
Hi Team,

I have only one esp32-s2-saola-1 kit available and I want to test the secure boot v2 feature without efuse the public digest. so that I can flash unsigned binary on the same board after disabling the secure boot v2 feature.

I have referred [Restrictions after Secure Boot is enabled](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/secure-boot-v2.html?highlight=secure%20boot#restrictions-after-secure-boot-is-enabled) and there is CONFIG_SECURE_BOOT_INSECURE option which I am not sure is helpful for me or not?

Please help me as I have only one kit available and I don't want to burn the efuse and want to check the secure boot v2 feature for our product.

Thanks

negativekelvin, 2021-04-16:
I think you can set CONFIG_EFUSE_VIRTUAL and try without burning efuses

projectgus, 2021-05-06:
Hi @Darsh-Dev,

CONFIG_EFUSE_VIRTUAL probably doesn't work with Secure Boot at the moment. We are working on some tools to make it easier to do internal test of Secure Boot configurations without burning efuses, but they're not available yet.

The closest feature for enabling signature checks while not enabling hardware secure boot (i.e. no hardware root of trust) is to enable signature checking on OTA update, as discussed in #6212 this is available already.

The other option is to actually enable Secure Boot. Provided you keep the signing key safe, you should be able to re-sign and re-flash the firmware as many times as you wish.

Do either of these solve the problem for you?

Alvin1Zhang, 2021-06-02:
@Darsh-Dev Thanks for reporting, would you please help share if any updates for this issue? Thanks.

KonstantinKondrashov, 2021-07-30:
Hi @Darsh-Dev !
We added support CONFIG_EFUSE_VIRTUAL with Secure Boot and Flash Encryption. This option is only for testing!!! You can enable these options below and define a partition in the partition table `emul_efuse, data, efuse, , 0x2000`. It redirects all eFuse operations to the `emul_efuse` partition, no real burn operations. The written eFuses are kept in flash. You can erase the flash to start the first boot again if needed.
CONFIG_EFUSE_VIRTUAL=y
CONFIG_EFUSE_VIRTUAL_KEEP_IN_FLASH=y

See the example https://github.com/espressif/esp-idf/blob/master/examples/system/efuse/sdkconfig.ci.virt_sb_v2_and_fe

if CONFIG_EFUSE_VIRTUAL  +  Flash Encryption then there is no encryption occurs with spi_flash, because FE is managed by hardware.

closed 2021-07-30 by Alvin1Zhang