espressif/esp-idf #6878: Crash when resolving .local hostname when no DNS servers are present

RJPoelstra, 2021-04-15:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  [ESP32-SOLO-1]
- IDF version (run ``git describe --tags`` to find it): v4.4-dev-744-g1cb31e509
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     macOS
- (Windows only) environment type: [].
- Using an IDE?: Yes, eclipse with ESP-IDF plugin
- Power Supply:         external 3.3V

## Problem Description

When lwip doesn't have DNS server (as is the case when it has a self-assigned IP) and a `hostname.local` address is resolved, `abort()` is called because it tries to use a DNS server that is out-of-bounds for the `dns_servers` array.

The problem is actually created in dns.c:1093. There it seems to loop over the `dns_servers` increasing `server_idx` looking for a valid DNS server. As there are no DNS servers it doesn't find one, but it leaves `server_idx` at `DNS_MAX_SERVERS`. A few lines lower (1097) it calls `dns_send()` with the now invalid dns entry (`dns_send()` checks that `server_idx` < `DNS_MAX_SERVERS`).

This can be fixed in dns.c:1092 by replacing 
```
      while((entry->server_idx  < DNS_MAX_SERVERS) && ip_addr_isany_val(dns_servers[entry->server_idx])) {
```
with 
```
      while((entry->server_idx + 1 < DNS_MAX_SERVERS) && ip_addr_isany_val(dns_servers[entry->server_idx])) {
```
(Note the + 1)
This prevents `server_idx` from becoming larger than `DNS_MAX_SERVERS`.
### Expected Behavior

The hostname is resolved using mDNS.

### Actual Behavior

`abort()` is called

### Steps to reproduce

1. Setup a OTA example making sure that:
- The network you connect to doesn't have a DHCP server (so no DNS servers are set in lwip), e.g. use the AutoIP feature and have a direct connection to your computer.
- The hostname for OTA updates should be link-local (i.e. end in .local).

### Code to reproduce this issue
Some excerpts:

The hostname to resolve:
```cpp
		esp_http_client_config_t config = {
				.url = "https://imac-van-remco.local:8070/S000030.bin",
				.cert_pem = (char *)server_cert_pem_start,
				.event_handler = http_event_handler,
		};
```
## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5892
load:0x40078000,len:14184
load:0x40080400,len:3472
entry 0x40080644
I (27) boot: ESP-IDF v4.4-dev-744-g1cb31e509-dirty 2nd stage bootloader
I (27) boot: compile time 11:45:43
I (27) boot: chip revision: 1
I (29) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (35) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (66) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (72) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  3 ota_0            OTA app          00 10 00010000 00100000
I (85) boot:  4 ota_1            OTA app          00 11 00110000 00100000
I (92) boot:  5 ce_fw            unknown          40 00 00210000 00100000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 1, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1912ch (102700) map
I (154) esp_image: segment 1: paddr=00029154 vaddr=3ffb0000 size=022dch (  8924) load
I (158) esp_image: segment 2: paddr=0002b438 vaddr=40080000 size=04be0h ( 19424) load
I (166) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=6bf20h (442144) map
I (334) esp_image: segment 4: paddr=0009bf48 vaddr=40084be0 size=08384h ( 33668) load
I (348) esp_image: segment 5: paddr=000a42d4 vaddr=50000000 size=00010h (    16) load
I (348) esp_image: segment 6: paddr=000a42ec vaddr=00000000 size=0bc94h ( 48276) 
I (377) boot: Loaded app from partition at offset 0x10000
I (377) boot: Disabling RNG early entropy source...
I (387) cpu_start: Pro cpu up.
I (387) cpu_start: Single core mode
I (396) cpu_start: Pro cpu start user code
I (396) cpu_start: cpu freq: 160000000
I (396) cpu_start: Application information:
I (397) cpu_start: Project name:     S000030
I (401) cpu_start: App version:      0.1.18
I (405) cpu_start: Compile time:     Apr 15 2021 10:50:25
I (410) cpu_start: ELF file SHA256:  90569573fb701dec...
I (415) cpu_start: ESP-IDF:          v4.4-dev-744-g1cb31e509-dirty
I (421) heap_init: Initializing. RAM available for dynamic allocation:
I (428) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (433) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (438) heap_init: At 3FFB8660 len 000279A0 (158 KiB): DRAM
I (443) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (449) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (454) heap_init: At 4008CF64 len 0001309C (76 KiB): IRAM
I (460) spi_flash: detected chip: generic
I (463) spi_flash: flash io: dio
I (466) cpu_start: Starting scheduler on PRO CPU.
I (471) uart: queue free spaces: 10
I (471) system_api: Base MAC address is not set
I (471) system_api: read default base MAC address from EFUSE
I (491) esp_eth.netif.glue: 24:0a:c4:69:ff:7f
I (491) esp_eth.netif.glue: ethernet attached to netif
I (501) uart: queue free spaces: 10
I (501) esp_image: segment 0: paddr=00110020 vaddr=3f400020 size=1add4h (110036) map
I (521) LAN: Ethernet Started
I (551) esp_image: segment 1: paddr=0012adfc vaddr=3ffb0000 size=0239ch (  9116) 
I (571) esp_image: segment 2: paddr=0012d1a0 vaddr=40080000 size=02e78h ( 11896) 
I (571) esp_image: segment 3: paddr=00130020 vaddr=400d0020 size=6d154h (446804) map
I (821) esp_image: segment 4: paddr=0019d17c vaddr=40082e78 size=0a1a8h ( 41384) 
I (861) esp_image: segment 5: paddr=001a732c vaddr=50000000 size=00010h (    16) 
I (861) esp_image: segment 6: paddr=001a7344 vaddr=00000000 size=08c3ch ( 35900) 
I (881) esp_image: Verifying image signature...
I (4521) LAN: Ethernet Link Up
I (4521) LAN: Ethernet HW Addr 24:0a:c4:69:ff:7f
I (9771) esp_netif_handlers: eth ip: 169.254.128.255, mask: 255.255.0.0, gw: 0.0.0.0
I (9791) LAN: Socket created
I (9791) LAN: Socket bound
I (9821) LAN: Socket listening: 58904
I (22051) OTA: Starting new OTA check round
assertion "dns server out of array" failed: file "IDF/components/lwip/lwip/src/core/dns.c", line 791, function: dns_send

abort() was called at PC 0x40123e0f on core 0
Setting breakpoint at 0x400860aa and returning...
```

## Other items if possible

Alvin1Zhang, 2021-04-16:
Thanks for reporting, we will look into.

yuanjianmin, 2021-04-22:
@RJPoelstra Thanks for your reporting. We will fix this problem in the future version.

closed 2021-05-07 by igrr