espressif/esp-idf #6881: Download-Tool fails due to missing esspresif ressource

Duschgeler, 2021-04-15:
## Environment

- Operating System:     Windows

## Problem Description

Setup-Tool 2.7 cannot checkout master branch

### Expected Behavior

Download and install all necessary files including
https://dl.espressif.com/dl/idf-python-wheels/idf-python-wheels-3.8-2020-01-21-win64.zip

### Actual Behavior
Download ressources failure

### Steps to reproduce

1. Start Setup- Tool 2.7 
2. Select Version master (development branch)

## Debug Logs

```
                          Installing idf-python-wheels@3.8-2020-01-21
                          Downloading idf-python-wheels-3.8-2020-01-21-win64.zip to C:\Users\Berlin\.espressif\dist\idf-python-wheels-3.8-2020-01-21-win64.zip.tmp
                          WARNING: Download failure HTTP Error 404: Not Found
                          WARNING: Failed to download https://dl.espressif.com/dl/idf-python-wheels/idf-python-wheels-3.8-2020-01-21-win64.zip to C:\Users\Berlin\.espressif\dist\idf-python-wheels-3.8-2020-01-21-win64.zip.tmp
                          Downloading idf-python-wheels-3.8-2020-01-21-win64.zip to C:\Users\Berlin\.espressif\dist\idf-python-wheels-3.8-2020-01-21-win64.zip.tmp
                          WARNING: Download failure HTTP Error 404: Not Found
                          WARNING: Failed to download https://dl.espressif.com/dl/idf-python-wheels/idf-python-wheels-3.8-2020-01-21-win64.zip to C:\Users\Berlin\.espressif\dist\idf-python-wheels-3.8-2020-01-21-win64.zip.tmp
                          Downloading idf-python-wheels-3.8-2020-01-21-win64.zip to C:\Users\Berlin\.espressif\dist\idf-python-wheels-3.8-2020-01-21-win64.zip.tmp
                          WARNING: Download failure HTTP Error 404: Not Found
                          WARNING: Failed to download https://dl.espressif.com/dl/idf-python-wheels/idf-python-wheels-3.8-2020-01-21-win64.zip to C:\Users\Berlin\.espressif\dist\idf-python-wheels-3.8-2020-01-21-win64.zip.tmp
                          ERROR: Failed to download, and retry count has expired
                          Traceback (most recent call last):
                            File "C:\Users\Berlin\Desktop\esp-idf-2\\tools\idf_tools.py", line 1684, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\Berlin\Desktop\esp-idf-2\\tools\idf_tools.py", line 1680, in main
                              action_func(args)
                            File "C:\Users\Berlin\Desktop\esp-idf-2\\tools\idf_tools.py", line 1331, in action_install
                              tool_obj.download(tool_version)
                            File "C:\Users\Berlin\Desktop\esp-idf-2\\tools\idf_tools.py", line 717, in download
                              raise DownloadError()
                          __main__.DownloadError
```

## Other items if possible

Same occurs with install.bat after changing to master branch
1. git checkout master (v4.2 relaease)
2. git submodule update --init --recursive
3. install.bat

Alvin1Zhang, 2021-04-16:
Thanks for reporting, we will look into.

georgik, 2021-04-16:
@Duschgeler Thank you for reporting the issue.
The URL is pointing to 2020 instead of 2021 which is mistake.
We created symlink on the server so that even URL with 2020 works.
Please, retry the download now.

closed 2021-04-27 by espressif-bot

Sweetkubuni, 2022-08-05:
I think this is still a problem. I get 404 error

TimLohrer, 2023-08-08:
I also still get a 404 error