espressif/esp-idf #6883: wifi:shed is null, rcReachRetryLimit CRASH

aircable, 2021-04-16:
## Environment

- Module or chip used:  ESP32-U4WDH and ESP32-D0WD-V3
- IDF version v4.4-dev-930-ga6f33a9d3
- Build System:         idf.py
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

## Problem Description

Code example:
```
        for( int i = 0; i < numb_pkg; i++ ){
            // number of packet repeats
            err = esp_wifi_80211_tx( WIFI_IF_STA, (uint8_t*)&ppkt[i], transmit_length, false ); // we create the sequence no
            vTaskDelay( 50 / portTICK_PERIOD_MS );
        }
```
After about 2 to 20 packets transmitted, the processor throws this error:

```
E (73086) wifi:sched is null, if=0 tid=7 flags=6402
E (73086) wifi:trc.c rcReachRetryLimit 1736

E (78084) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (78084) task_wdt:  - IDLE (CPU 0)
E (78084) task_wdt: Tasks currently running:
E (78084) task_wdt: CPU 0: wifi
E (78084) task_wdt: Aborting.

abort() was called at PC 0x401348b4 on core 0
0x401348b4: task_wdt_isr at ESP/esp-idf/components/esp_system/task_wdt.c:184 (discriminator 1)

Backtrace:0x40090b66:0x3ffbe6e00x40090a91:0x3ffbe700 0x40097d9d:0x3ffbe720 0x401348b4:0x3ffbe7a0 0x40082f02:0x3ffbe7c0 0x40171211:0x3ffd32b0 0x4016c573:0x3ffd32e0 0x4016c94d:0x3ffd3310 0x4016cb67:0x3ffd3340 0x4016ccca:0x3ffd3370 0x40198fb7:0x3ffd33b0 
0x40090b66: panic_abort at ESP/esp-idf/components/esp_system/panic.c:365
0x40090a91: esp_system_abort at ESP/esp-idf/components/esp_system/esp_system.c:126
0x40097d9d: abort at ESP/esp-idf/components/newlib/abort.c:46
0x401348b4: task_wdt_isr at ESP/esp-idf/components/esp_system/task_wdt.c:184 (discriminator 1)
0x40082f02: _xt_lowint1 at ESP/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x40171211: rcReachRetryLimit at ??:?
0x4016c573: lmacProcessTxSuccess at ??:?
0x4016c94d: lmacProcessCollision at ??:?
0x4016cb67: lmacProcessTxError at ??:?
0x4016ccca: lmacProcessTxComplete at ??:?
0x40198fb7: ppTask at ??:?
```

### Expected Behavior

no error

### Actual Behavior

crash

### Steps to reproduce

see above, simple 802.11packet transmit, no connection required

Problem is, there is no reference, no way to debug, this code is all internal to Espressif
Please help

aircable, 2021-04-16:
It may have something to do with Bluetooth. Disabling the initialization of the Bluetooth subsystem, and I can't see that error anymore.
Any suggestions?

MaxwellAlan, 2021-04-20:
hi @aircable Thanks for report this issue.
Have you ever try esp_wifi_80211_tx( WIFI_IF_STA, (uint8_t*)&ppkt[i], transmit_length, **enable** ); when enable bluetooth?

aircable, 2021-04-20:
Yes, I did. Unfortunately it had no effect. 
It always happens after about 20-30 packets sent. Sniffer is disabled. Just transmitting.
Even with Wifi debug on VERBOSE, there is no further information about what is wrong.
Increasing the vTaskDelay or removing it does not change anything either.
```
E (8103) wifi:sched is null, if=0 tid=7 flags=6402
E (8103) wifi:trc.c rcReachRetryLimit 1736
```
Can you shed a light what this error actually means? There is nothing in the doc or anywhere.
Thanks

aircable, 2021-04-20:
I also verified that esp_wifi_80211_tx() did not return an error code.

MaxwellAlan, 2021-04-21:
hi @aircable , the errors means that the connections may disconnected and related sources has been releasesed, but still tx pkt may cause this issue. Can you provide a mini demo code for me to reproduce this issue and debug?

aircable, 2021-04-21:
But there is no connection. It is not running in station or ap mode. The implementation makes use of 802.11 frames, just like the ESPNOW implementation.
I will have to create a demo code for you, the system is pretty big. I setup Wifi for default empty STA mode:
```
    ESP_ERROR_CHECK( esp_wifi_set_config( ESP_IF_WIFI_STA, &wifi_sta_config ));
    ESP_ERROR_CHECK( esp_wifi_set_bandwidth( ESP_IF_WIFI_STA, WIFI_BW_HT20 )); // 20 MBit/s using 1 channels
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter( &filter));      // set MAC frame filter
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb( &wifi_sniffer_packet_handler ));
    ESP_ERROR_CHECK(esp_wifi_set_protocol( ESP_IF_WIFI_STA, WIFI_PROTOCOL_LR ));
```
The code only transmits 802.11 frames.

aircable, 2021-04-23:
I'm still trying to build an app that replicates the error. It's difficult. 
1: when the device only transmits (like example above) and a call to esp_wifi_set_promiscuous(true) does not lead to a wifi_sniffer_packet_handler() callback, I cannot reproduce the error. I cannot overload the Wifi system transmitting 802.11 frames.
But, when the receiver is sending back data and it leads to a callback to receive promiscuous packets, the Wifi system dies with that crash.

MaxwellAlan, 2021-04-25:
hi @aircable, thanks for doing that, can you try this lib  
[fix_esp_wifi_80211_tx_0425.tar.gz](https://github.com/espressif/esp-idf/files/6371151/fix_esp_wifi_80211_tx_0425.tar.gz)
Wating for you reply.

aircable, 2021-04-25:
Dear @MaxwellAlan, Thanks for the fix, I did try to run this library extensively.

1. transmit works now with coexisting Bluetooth, no crashes. I cannot overload it.
2. but receive stopped working, no 802.11 frames in promiscuous mode are received. Call back never happens.

Either the function call to esp_wifi_set_promiscuous_rx_cb( &wifi_sniffer_packet_handler ) did not fail but also did not install the call back handler. Or the call to esp_wifi_set_promiscuous(true) had no effect, can't tell.

A key original problem was that when the promiscuous mode is active, then the transmitter can fail. Even though, the main loop looks like this:

- enable promiscuous for 500ms
- stop promiscuous mode, and transmit 802.11 frames from the queue until empty
- loop

Now the receiver does not get anything and thus, a crash does not happen.
I can also double confirm that when Bluetooth is not started with this sequence the Wifi crash does not happen.
```
esp_bt_controller_init( &bt_cfg );
esp_bt_controller_enable( ESP_BT_MODE_BTDM ); 
esp_bluedroid_init();
esp_bluedroid_enable();
esp_bt_sleep_enable();

```

MaxwellAlan, 2021-04-26:
hi @aircable , notice that you are calling esp_wifi_set_protocol( ESP_IF_WIFI_STA, WIFI_PROTOCOL_LR ); which means you will use low rate, right? the default esp_wifi_80211_tx send pkts rate in 1Mbps, so you can't sniffer the pkts in Low rate phy mode. I add a new api for config esp_wifi_80211_tx rate, you can use like this:
`
extern esp_err_t esp_wifi_80211_tx_rate_config(wifi_interface_t ifx, wifi_phy_rate_t rate);
... // others init codes
ESP_ERROR_CHECK(esp_wifi_set_protocol( ESP_IF_WIFI_STA, WIFI_PROTOCOL_LR )); // after enable Low Rate Phy Mode
esp_wifi_80211_tx_rate_config(ESP_IF_WIFI_STA, WIFI_PHY_RATE_LORA_250K); // config 80211 tx rate
...// others codes
` 
[add_esp_wifi_80211_tx_rate_config.tar.gz](https://github.com/espressif/esp-idf/files/6374626/add_esp_wifi_80211_tx_rate_config.tar.gz)

MaxwellAlan, 2021-04-26:
@aircable the root cause is that:
usually when we send an unicast 80211 frame, we need an 80211 ack frame.
when not enable BT module, all the RF time is occupied by WiFi, so the ratio of successfully receive ack frame is high;
when enable BT module , the RF time is seprated by WiFi and BT, which cause the ratio of successfully receive ack frame a littlt lower, however there is a bug when not receive ack frame  exceeds retry times, which will cause crash.

negativekelvin, 2021-04-26:
>I add a new api for config esp_wifi_80211_tx rate

:thumbsup:

aircable, 2021-04-26:
Wow, thank you so much. Regarding the ack frame, can I disable that feature? I use the low rate to implement a broadcast feature and I don't need any acknowledgement or handshake for that. It is designed that packages can be lost.
I will test the Phy config today.

aircable, 2021-04-27:
My test showed a bit weird implication:

Because of this issue: https://github.com/espressif/esp-idf/issues/6930
I switched to version: git clone -b release/v4.2 https://github.com/espressif/esp-idf.git
I removed all your fixes and have the plain github version installed.
Observation: transmit and receive work just fine, and LEDs for transmit/receive Wifi events work too.
No crashes either.
Question: regarding WIFI_PROTOCOL_LR, WIFI_PHY_RATE_LORA_250K: does that mean that those calls were ignored?

The change that made the GPIOs not work and the crash were introduced sometime after version 4.3-1197.

MaxwellAlan, 2021-04-27:
@aircable The crash is introduced by we add new features to disable 11b rate in release/v4.3, not backport to release/v4.2. And the issue #6930 maybe also some new feature of gpio affect, it may takes some time to debug.

aircable, 2021-04-29:
Update: IDF_4.2 is working without problems.
I'm using the internal function
`esp_err_t esp_wifi_internal_set_fix_rate(wifi_interface_t ifx, bool en, wifi_phy_rate_t rate);`
for WIFI_PHY_RATE_LORA_250K  which works as expected. All transmit and receive is fine and issue: #6930 is good.

I will be waiting until 4.4 is ready and fixed.

MaxwellAlan, 2021-05-06:
@aircable thanks for your feedback, and we will merge related fix to v4.4 as soon as possiable.

AxelLin, 2021-05-11:
> @aircable The crash is introduced by we add new features to disable 11b rate in release/v4.3, not backport to release/v4.2. 
Hi @MaxwellAlan 

From the recent release/v4.3 update, I'm nost sure if this issue is fixed in v4.3 or not (since you mentioned it was introduced by the changes in release/v4.3).
Can you clarify if this is still an issue in v4.3 or it's already fixed? thanks.

AxelLin, 2021-07-15:
> @aircable thanks for your feedback, and we will merge related fix to v4.4 as soon as possiable.

Hi @MaxwellAlan 

From your reply it seems you know the root cause since https://github.com/espressif/esp-idf/issues/6883#issuecomment-827271598
The fix is not yet available in master tree so far, could you update the status?

BTW, you may also want to check this issue https://github.com/espressif/esp-idf/issues/7255

AxelLin, 2021-08-25:
> @AxelLin it still an issue in release/v4.3, we will fix it these days.

@MaxwellAlan 
Can you clarify if this is fixed by 1fd8fdcf7de7b25f9980e7cc973da2953784c273 or it needs other fixes?

Alvin1Zhang, 2021-09-09:
@aircable Thanks for reporting and sharing updates, feel free to reopen.

closed 2021-09-09 by Alvin1Zhang