espressif/esp-idf #6885: How to get new GCC compiler

0xFEEDC0DE64, 2021-04-16:
I noticed when sourcing the export.sh of esp-idf 4.4-dev I get a gcc compiler version 8.4.0:

```
$ xtensa-esp32-elf-g++ --version
xtensa-esp32-elf-g++ (crosstool-NG esp-2020r3) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

But then I noticed Arch Linux already has version 10. How can I get version 10 with esp-idf and without using distro provided compilers?

https://aur.archlinux.org/packages/xtensa-esp32-elf-gcc/

negativekelvin, 2021-04-16:
Those are unofficial builds and do not include all espressif patches.  Latest releases are here https://github.com/espressif/crosstool-NG/releases

0xFEEDC0DE64, 2021-04-17:
Ok got that, how can I stop the esp-idf export.sh from changing the PATH so that the outdated 8.4.0 compiler binaries are used and instead the distro provided ones from `/usr/bin/xtensa-esp32-elf-gcc` are used?

0xFEEDC0DE64, 2021-04-26:
Is there any esp-idf provided way to override the compiler?

igrr, 2021-04-26:
Hi @0xFEEDC0DE64, adding the toolchain `bin` directory as the first entry of PATH should be sufficient.

For example, you can run `. export.sh` and then do `export PATH=/path/to/my/toolchain/bin:$PATH`.

Or, don't run `. export.sh` at all, and add the necessary directories (Python virtual environment, cross-compiler toolchain, CMake, Ninja) to PATH yourself.

After doing this, remember to run `idf.py fullclean` or remove the build directory, as CMake normally caches the compiler path.

0xFEEDC0DE64, 2021-05-27:
the cmake step already fails because the arch rebased gcc isn't able to compile simple small test programs. I understand that the development on gcc from espressif may have been stopped, but is there anything going on in regards to newer toolsets? Is llvm in consideration?

igrr, 2021-05-27:
Yes we will definitely switch to a newer compiler version in one of the next IDF releases. I am sorry I can't give you more specific timeline. If you'd like to build GCC 10 yourself in the meantime, you can try the branches linked in https://github.com/espressif/crosstool-NG/issues/11#issuecomment-803864897.

b1ackviking, 2021-12-28:
According to [this comment](https://github.com/espressif/esp-idf/issues/6214#issuecomment-980114759) in #6214, it is planned to upgrade the toolchain to GCC 11.

closed 2022-06-11 by espressif-bot