espressif/esp-idf #6886: Secure boot v2 does not write BLOCK2 properly, causes device to be bricked

eriknorth, 2021-04-16:
Hi,

We have been using Secure Boot v2 + Flash Encrypiton succesfully, however after the latest uplift of ESP-IDF v4.2-414-g71e5b8757, devices started to get bricked becuase of BLOCK2 fuse not being written properly - all the bytes in the digest are 0x00.

I tried to compare changes in the bootloader since the old version, however I did not see any significant changes that might have broken the functionality.

Any guesses what could go wrong?

Here is menuconfig of the security settings:
[image]

Here is efuse summary:
```
Detecting chip type... ESP32
espefuse.py v3.0
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1121 R/- (0b00011)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 49 R/W (0x31)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 385 R/W (0x0181)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 1 R/- (0x1)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme                
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address                               
   = 7c:9e:bd:d1:55:e8 (CRC 0xed OK) R/W 
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 237 R/W (0xed)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 127 R/W (0b1111111)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = True R/W (0b1)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                         Flash encryption key                              
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/- 
BLOCK2 (BLOCK2):                         Secure boot key                                   
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/- 
BLOCK3 (BLOCK3):                         Variable Block 3                                  
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```
[... 98 lines cut ...]
I (2239) flash_encrypt: Setting FLASH_CRYPT_CNT for permanent encryption
I (2258) flash_encrypt: Flash encryption completed
I (2258) boot: Resetting with flash encryption enabled...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
invalid header: 0x573542f2
ets Jul 29 2019 12:21:46
```

Thanks,
/Erik

Alvin1Zhang, 2021-04-19:
Thanks for reporting. Would you please help provide more details of hardware and etc., which would help us debug further. Thanks.

eriknorth, 2021-04-19:
Hi, we are using ESP32-WROVER-E (M213EH6464PH 3QO)

eriknorth, 2021-04-19:
Here is the SDKCONFIG. As we use unmodified bootloader, this is the only thing that might affect the bootloader.

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="$(PWD)/../../3rdParty/tools/xtensa-esp32-elf_2020r3/bin/xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_APP_PROJECT_VER_FROM_CONFIG is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
# CONFIG_BOOTLOADER_APP_ANTI_ROLLBACK is not set
CONFIG_BOOTLOADER_RESERVE_RTC_SIZE=0
# CONFIG_BOOTLOADER_CUSTOM_RESERVE_RTC is not set
# end of Bootloader config

#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
[... 1583 lines cut ...]
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=2048
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

negativekelvin, 2021-04-19:
If you want to debug this you can try setting CONFIG_EFUSE_VIRTUAL and adding some debug prints to https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/esp32/secure_boot.c

projectgus, 2021-04-20:
Hi @eriknorth,

Thanks for all those details. The commit you have, 71e5b8757, isn't in the ESP-IDF repo. Are there some local patches applied? Could you please check the closest commit from the IDF repo and also share a summary about any patches that are applied in your copy?

>  try setting CONFIG_EFUSE_VIRTUAL

Unfortunately this feature isn't available in v4.2.x.

No need to risk any more devices being bricked, we can check from this side.

projectgus, 2021-04-20:
It's OK, we've confirmed the issue exists in latest release/v4.2 branch. Fix will be available ASAP.

KonstantinKondrashov, 2021-04-20:
@eriknorth Hi!
Sorry for this bug. The fix for v4.2 branch: [13243.zip](https://github.com/espressif/esp-idf/files/6341403/13243.zip).
It will be merged ASAP.
Thanks.

eriknorth, 2021-04-20:
Hi @projectgus,

Thanks for confirming the fault. 
Sorry, just now noticed that we had some additional patches which are related to the websocket, therefore the out of tree hash.

Please contact me if there any further questions. After my investigation I found that for some reason the register data from the WDATA for BLOCK2 memory does not get written to the Fuses. 

/Erik

eriknorth, 2021-04-20:
Had not refreshed the page, will check the bug fix. Thanks for so fast fix!

/Erik

eriknorth, 2021-04-20:
Just tested and the fix works! Thank you @projectgus

AxelLin, 2021-04-21:
@KonstantinKondrashov 
I'm wondering if v4.3 branch and master also need this fix? (v4.3-beta3 does **not** include this fix.)

KonstantinKondrashov, 2021-04-21:
Hi @AxelLin!
I had a look v4.3 as well, it does not have this bug.  For v4.3 there was a fix f1dae0d6e125b0001cd38de2e07475f592811f49.
The issue: there was a MIXED burn way using REGs and IDF efuse APIs. Something one should be used. IDF efuse APIs do clean the write regs before burning something (the efuse batch mode blocks this clean). 

v4.3-beta3 uses only IDF efuse APIs (with efuse batch mode) in the secure_boot file, which is good.

patrikhuss, 2021-04-23:
Hi, when is this patch expected to be merged with the release branch? Status is changed into Done but where is the fix :-) ?

KonstantinKondrashov, 2021-04-23:
Hi! the commit 4200af3e26bb736d8b168f11490fde8091eb1582 soon will be available in GitHub. The fix was merged in our internal repo.
This issue will be closed automatically when it will be published here.

closed 2021-05-05 by mahavirj

AxelLin, 2021-05-05:
@mahavirj 
Just curious how a commit in esp-afr-sdk can fix esp-idf issue?
Does esp-idf use esp-afr-sdk?

mahavirj, 2021-05-05:
@AxelLin 

Strange! Correct fix that should have linked with issue is https://github.com/espressif/esp-idf/commit/4200af3e26bb736d8b168f11490fde8091eb1582. Anyways, fix is available in official ESP-IDF v4.2 release branch, so marking issue as resolved is right thing.