espressif/esp-idf #6887: Typo

chegewara, 2021-04-16:
https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h#L873

Typo in struct name `esp_ble_gap_ext_adv_reprot_t` which is reproduced in whole API.

chegewara, 2021-04-16:
https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h#L207

`ESP_GAP_BLE_CHANNEL_SELETE_ALGORITHM_EVT`

chegewara, 2021-04-19:
https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h#L1145

`peroid_adv_set_params`

chegewara, 2021-04-28:
Another typo (copy/paste bug):
https://github.com/espressif/esp-idf/blob/master/components/usb/hcd.c#L1451
should be `USB_PRIV_XFER_TYPE_BULK`

Dazza0, 2021-06-08:
> Another typo (copy/paste bug):
> https://github.com/espressif/esp-idf/blob/master/components/usb/hcd.c#L1451
> should be `USB_PRIV_XFER_TYPE_BULK`

Closed by 7f421048936c6007186db00ed382cd989c4b905e

closed 2022-07-07 by espressif-bot