espressif/esp-idf #6888: Installation with Arduino-ESP32 as component in ESP-IDF project

lgLindstrom, 2021-04-16:
I have installed ESP-IDF version 4 using WS2 in Windows 10. 
It works well testing the blink example.

Later I would like to use Arduino-ESP32 components

On this page ..   there is  a instruction how to use Arduino-ESP32 as components in a ESP-IDF project. 
I tried to follow this instruction but it breaks on the two last lines.

```
mkdir -p components && \
cd components && \
git clone https://github.com/espressif/arduino-esp32.git arduino && \
cd arduino && \
git submodule update --init --recursive && \
cd ../.. && \
make menuconfig
```

I have asked this question on the Arduino-ESP32 page also,, but the recomendation is to ask here.
Can anyone help me out.

chegewara, 2021-04-16:
My suggestion to ask here was about this particular error:
```
CMake Error at C:/Users/lglin/OneDrive/experimentalProjects/esp-idf/blink/blink/components/arduino/CMakeLists.txt:171 (idf_component_register):
  Unknown CMake command "idf_component_register".
```
Maybe @projectgus can help

lgLindstrom, 2021-04-16:
Okay,, but the root cause to all problems futher on down in the process must be that 

`make menuconfig` 

never executed properly.

chegewara, 2021-04-16:
Did you try `idf.py` instead of `make`?

dobairoland, 2021-04-16:
I've used the suggested commands and `make menuconfig` works for me for both ESP-IDF v3.3 and v4.4 (master branch).

`idf.py menuconfig` isn't working because of `Failed to resolve component 'arduino_tinyusb'.` like it was reported [here](https://github.com/espressif/arduino-esp32/issues/5054). So this is a cmake arduino issue. All of these are working fine in ESP-IDF.

lgLindstrom, 2021-04-16:
> I've used the suggested commands and make menuconfig works for me for both ESP-IDF v3.3 and v4.4 (master branch).

As far as my knowledge about the make command is that it needs a make-file,, in this case menuconfig. Where did you find that file ?

dobairoland, 2021-04-16:
`menuconfig` is not a file but a target. It is defined [here](https://github.com/espressif/esp-idf/blob/master/make/project_config.mk#L141)

dobairoland, 2021-04-16:
The duplicate issue: https://github.com/espressif/arduino-esp32/issues/5064

lgLindstrom, 2021-04-16:
@dobairoland    I started that Issue.

Running : 
`make menuconfig `

will open a Makefile with this content:

```
#
# This is a project Makefile. It is assumed the directory this Makefile resides in is a
# project subdirectory.
#

PROJECT_NAME := blink

include $(IDF_PATH)/make/project.mk
```

`$(IDF_PATH)/make/project.mk`  

**Dont have any target named menuconfig**  

So,, how do I fix this ?

dobairoland, 2021-04-16:
> make menuconfig will open a Makefile with this content

What do you mean by that? It opens a text editor with that content?

lgLindstrom, 2021-04-16:
no,,,,
the command  "make"   will search for a file named "Makefile"   and  execute according to content in that file.

In this special case, content is a link to another file "$(IDF_PATH)/make/project.mk".

Then "make" uses that file as a target and tries to execute  using that file.

The parameter  "menuconfiguration"     tells "make" to execute the part of  "$(IDF_PATH)/make/project.mk" that contains that label.

The label "menuconfiguration" is missing in the file "$(IDF_PATH)/make/project.mk"

dobairoland, 2021-04-16:
Thanks the description but I understand how it is working. `project.mk` will include `project_config.mk` which contains the `menuconfig` target at the place I linked earlier.

I would like to see the output of `make menuconfig` so I could help you. Please run `make menuconfig` and post the output here.

lgLindstrom, 2021-04-16:
This is the outpur:

```
Toolchain path: /home/lg/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: esp-2019r2
Compiler version: 8.2.0
Python requirements from /home/lg/dev/esp/esp-idf/requirements.txt are satisfied.
make[1]: Entering directory '/home/lg/dev/esp/esp-idf/tools/kconfig'
cc -c  -DCURSES_LOC="<curses.h>" -DLOCALE -MMD -MP -I "." -I "/home/lg/dev/esp/esp-idf/tools/kconfig"  /home/lg/dev/esp/esp-idf/tools/kconfig/mconf.c -o mconf.o
<command-line>: fatal error: curses.h: No such file or directory
compilation terminated.
make[1]: *** [Makefile:173: mconf.o] Error 1
make[1]: Leaving directory '/home/lg/dev/esp/esp-idf/tools/kconfig'
make: *** No rule to make target '/home/lg/dev/esp/esp-idf/tools/kconfig/conf-idf', needed by '/home/lg/dev/esp/esp-idf/tools/kconfig/mconf-idf'.  Stop.
```

dobairoland, 2021-04-16:
So your menuconfig doesn't show up because you don't have the required [ncurses](https://en.wikipedia.org/wiki/Ncurses) library installed.

Are you in Ubuntu-based WSL2? If yes then I think `apt install libncurses-dev` should install the right library. I'm not a WSL user so I'm not sure, maybe you need `sudo` as well, i.e. `sudo apt install libncurses-dev`. Maybe you'll need `libncurses5-dev` instead if you have a too recent version.

lgLindstrom, 2021-04-16:
makes  no difference,,, tested both of them

dobairoland, 2021-04-16:
I'm sorry I'm not sure if this can be fixed in WSL. WSL was never officially supported.

Any chance that you could use these from Windows directly? Here is a recommendation what to use: https://github.com/espressif/arduino-esp32/issues/5064#issuecomment-821290262 

`idf.py` including `idf.py menuconfig` works pretty well on Windows from Windows command line or Powershell. These are best to be used with ESP-IDF v4.0 or newer. 

The MSYS-based environment downloadable from [here](https://docs.espressif.com/projects/esp-idf/en/v3.3.5/get-started/windows-setup.html#toolchain-setup) was the recommended system on Windows before ESP-IDF v4.0. The `make` build system is available from there.

lgLindstrom, 2021-04-17:
ahhh  I fell I getting contadictory advices in all directions :)  (no offence, greatful for help )

As I understand, latest EPS-IDF is python based. I decided to go back and try  " idf.py menuconfig ". This is the result.

```
idf.py menuconfig
Checking Python dependencies...
Python requirements from /home/lg/dev/esp/esp-idf/requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory /home/lg/dev/test/blink/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DWARN_UNINITIALIZED=1 -DCCACHE_ENABLE=0 /home/lg/dev/test/blink"...
Warn about uninitialized values.
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /home/lg/dev/esp/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lg/dev/esp/esp-idf/components/cmock does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lg/dev/esp/esp-idf/components/openthread does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/lg/dev/esp/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/lg/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/lg/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/lg/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/lg/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/lg/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
CMake Error at /home/lg/dev/esp/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'arduino_tinyusb'.
Call Stack (most recent call first):
  /home/lg/dev/esp/esp-idf/tools/cmake/build.cmake:216 (__build_resolve_and_add_req)
  /home/lg/dev/esp/esp-idf/tools/cmake/build.cmake:425 (__build_expand_requirements)
  /home/lg/dev/esp/esp-idf/tools/cmake/project.cmake:348 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/lg/dev/test/blink/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```
Please not this lines:

```
-- Project version: 1
-- Building ESP-IDF components for target esp32
CMake Error at /home/lg/dev/esp/esp-idf/tools/cmake/build.cmake:185 (message):
  Failed to resolve component 'arduino_tinyusb'.
```

Can i be that arduino-esp32 is faulty ??

chegewara, 2021-04-17:
Here is error you have now:
> Failed to resolve component 'arduino_tinyusb'.

Here is answer you explain about that error you got few posts above:
https://github.com/espressif/esp-idf/issues/6888#issuecomment-821177509

lgLindstrom, 2021-04-17:
The purpose of my working here was to convert my application from PlatformIo to ESP-IDF. 
If was not straight forward and have resulted in me writing 3 "issues" to search for help.

First one was this : https://github.com/espressif/vscode-esp-idf-extension/issues/371
Root cause was that I where using a bash shell and ESP-IDF only supported Power Shell and WSL

My project uses bash scripts to init stuff so I decided to use WSL for my next try.

When I installed ESP-IDF I unfourtnatly installed version 4 of the IDF. Installing Arduino-ESP32 I run into problems with "make menuconfig".  This thread pointed out that I should us "IDF.py menuconfig" instead wich caused another error more related to Arduino-ESP32 and have its solution here  https://github.com/espressif/arduino-esp32/issues/5054

Anyhow, someone pointed out that Arduino-ESP32 requires ESP-IDF version 3.5
So having all solutions made me go for a third attempt.

On a clean machine I installed ESP-IDF 3.5. Installed the blink example wich compiled ok. Installed Arduino-ESP32  and instead of using "make menuconfig" I used IDF.py menuconfig .....
Guess what :) I got a new error. I've seen the before but ignored it because I thought it was a secondary problem. 

```
CMake Error at C:/Users/lglin/OneDrive/experimentalProjects/esp-idf/blink/blink/components/arduino/CMakeLists.txt:171 (idf_component_register):
  Unknown CMake command "idf_component_register".
```

So what to do now?? I have correct version of ESP-IDF. Installed Arduino-ESP32 according to instructions with the exception that I changed "make" to "IDF.py"
Still it are not working...
:)

me-no-dev, 2021-04-17:
Current Arduino master is building with IDF master. There is a release/v1.0 branch that works with IDF 3.3.5. Instructions need updating. I would recommend you to use the lib builder: https://github.com/espressif/esp32-arduino-lib-builder
master branch builds with IDF master and release/v3.3 branch builds with IDF 3.3.5 (and Arduino release/v1.0)

dobairoland, 2021-04-18:
I think we can close this in favor of https://github.com/espressif/arduino-esp32/issues/5064. Please seek any further guidance regarding Arduino in that issue.

closed 2021-04-18 by dobairoland